Openvario/meta-openvario #2: Porting to yocto/warrior: making XCSoar run

kedder, 2020-02-01:
One of the major issues with XCSoar on current `warrior` branch is inability to rotate the screen to match the device orientation. XCSoar runs under linux framebuffer device and doesn't react to its `DisplayOrientation` configuration option, always running in landscape mode.

Possible solutions I can see: 

1. Switch back to old kernel-3.4 with proprietary mali drivers (known-to-work configuration)
2. Run xcsoar under xorg where it should be (is it?) possible to rotate the screen on Xorg level.

Is there any other option here?

kedder, 2020-02-01:
So far I have tried option 1:

Kernel 3.4 compiles and boots, but not completely - fails on systemd startup. Apparently systemd requires newer kernel (Linux kernel >= 3.13, according to https://github.com/systemd/systemd/blob/master/README) and will not start. This option would require either downgrading systemd (to some pre 2018 version) or patching it, or switching to systemv init.

None of these are simple/future proof, so I didn't try to continue with it. If completely stuck, we might try to revisit it.

kedder, 2020-02-02:
For option 2: I was trying to play with new "lima" driver and get it working under Xorg. What I was able to do so far is:

1. Make lima kernel module compile and load
2. Build new mesa (19.1.6) with lima driver
3. Start Xorg and start xcsoar on it.

So far xcsoar runs unsatisfactory slow under this configuration and I'm not sure any accelelartion is coming from lima (probably not). 

On the positive side: `eglgears` do spin under X.

kedder, 2020-02-02:
Another way to run XCSoar would be through DRM. XCSoar needs to be compiled with these flags:
```
ENABLE_MESA_KMS=y GEOTIFF=n OPENGL=y EGL=y
```

This requires at least kernel with lima driver and mesa with lima driver. Unfortunately so far I couldn't make it work (both kernel and mesa lima drivers present). XCSoar fails to start with this message:

```
Using DRI device /dev/dri/card0 (use environment variable DRI_DEVICE to override)
drmModeGetResources() failed
```

It seems like we need to make `kmscube` running first to make sure infrastructure actually works. It fails with similar error right now: 
```
drmModeGetResources failed: Operation not supported
failed to initialize legacy DRM
```

Looks like there is a thread dealing with similar issue here: https://gitlab.freedesktop.org/lima/linux/issues/22

kedder, 2020-02-03:
Trying to get some help from lima guys on this, started discussion here: https://gitlab.freedesktop.org/lima/mesa/issues/131.

kedder, 2020-02-03:
@linuxianer99, can you give me some info about these files in meta-openvario: [0008-cubiescreen_4.19.patch](https://github.com/Openvario/meta-openvario/blob/warrior/recipes-kernel/linux/linux-mainline/0008-cubiescreen_4.19.patch) and [0009-simple-display_4.19.patch](https://github.com/Openvario/meta-openvario/blob/warrior/recipes-kernel/linux/linux-mainline/0009-simple-display_4.19.patch). Where did they came from? Did they work (i.e. did they enable openvario LVDS panel)?

linuxianer99, 2020-02-03:
@kedder : These patches came from another developer who tried to bring Mali400 GPU to kernel 4.19. This was all for pyro. The patches where intended for X11 and for framebuffer EGL. I think we can delete them. 
The LVDS is setup via DTB file. There is a routing through the video units of A20 defined,

linuxianer99, 2020-02-03:
@kedder , did you have a look at "src/Hardware/RotateDisplay.cpp" ?
There are a some defines like "SOFTWARE_ROTATE_DISPLAY" and "KOBO" (I think KOBO uses also the framebuffer ...

kedder, 2020-02-03:
> did you have a look at "src/Hardware/RotateDisplay.cpp" 

I did look briefly, but it seems that the only place it is defined is in `Features.hpp`:

```c
#if defined(ENABLE_OPENGL) && !defined(ANDROID) && !defined(TARGET_OS_IPHONE)
/**
 * Support display rotation via glRotatef()?
 */
#define SOFTWARE_ROTATE_DISPLAY
#endif
```

It seems like it expects rotation to be done by means of opengl, but TBH I didn't dig very deeply into this.

> The LVDS is setup via DTB file

The trick is that we have to teach sun4i_drm to use the LCD panel so that the whole KMS/DRM system can output into it. Then I would expect we could get something analogous to old mali setup. XCSoar seems to have support for KMS, lima driver can indeed render off-screen on cubieboard. The display part is tricky, because I can't find any relevant example of device tree configs to set up the screen. There are things like `CONFIG_DRM_PANEL_LVDS` kernel module, but it is not clear how to make them work or whether they are relevant at all. Do you have any pointers / ideas?

linuxianer99, 2020-02-03:
Maybe you can find some hint here:
https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/Documentation/devicetree/bindings/display/sunxi/sun4i-drm.txt?h=v5.5

kedder, 2020-02-03:
Yeah, I saw that, but it doesn't make a lot of sense. I don't quite see how to map stuff from old .fex file into that.

> At the moment the framebuffer is initialized by uboot. I think there is no further init in kernel ...

Right, but it only seem to work for  simple framebuffer. I'm not completely sure how/why exactly that works, but if I remove simple_framebuffer from the kernel, screen turns all white once kernel takes over the console. So, my guess is that kernel needs to know something about the display.

Also, both XCSoar and kmscube now complain about `no connected connector!`.  I suspect this has something to do with devicetree, but not really sure.

kedder, 2020-02-03:
Ok, stuff from .fex seem to map nicely here: https://www.kernel.org/doc/Documentation/devicetree/bindings/display/panel/panel-lvds.txt

kedder, 2020-02-04:
Ok, I have some progress. I still have no output to the LCD panel, but I have kmscube (and XCSoar) running (but I don't see anything). Did that by constructing the right graphics pipeline in device tree. Now I need to figure out how to make the panel to show something...

kedder, 2020-02-06:
It seems like LVDS is explicitly not supported for A20 in the kernel. `sun7i_a20_quirks` in [sun4i_tcon.c, L1456](https://github.com/torvalds/linux/blob/e310396bb8d7db977a0e10ef7b5040e98b89c34c/drivers/gpu/drm/sun4i/sun4i_tcon.c#L1456) doesn't have `supports_lvds` property (compare to `sun8i_a83t_lcd_quirks`), so all LVDS setup is skipped in the tcon driver, no matter how you set up your device tree.

As a last ditch attempt, I'm going to submit my findings to [linux-sunxi mailing list](https://groups.google.com/d/forum/linux-sunxi). I hope I'm still missing something.

kedder, 2020-02-07:
Discussion topic is here: https://groups.google.com/forum/#!topic/linux-sunxi/yJyIitJMwNc

mihu-ov, 2020-02-07:
Don´t know if this helps or not but from the OV history:

>XCSoar runs under linux framebuffer device and doesn't react to its DisplayOrientation configuration option

AFAIK the XCSoar display orientation configuration was never used. From https://openvario.org/doku.php?id=projects:series_00:build:setupguide, "_Select System → Settings → Display_Rotation and select your preferred rotation. Reboot your Openvario. Note that this is the only rotation setting that is required, any setting in your XCSoar profile files is ignored!_". Display rotation was defined with  in /boot/config.uEnv ist with parameter rotation=0 or rotation=90, probably extended later to allow 180° and 270° rotation. Maybe linuxianer99 remembers the details or you could follow the scripts in ovmenu,

>It seems like LVDS is explicitly not supported for A20 in the kernel

Image 19271 provided by linuxianer99 has LVDS output on - if I remember correctly - kernel 4.19 (tested with PixelQi LCD version).

kedder, 2020-02-07:
> Image 19271 provided by linuxianer99 has LVDS output on - if I remember correctly - kernel 4.19 (tested with PixelQi LCD version).

Right, but that image uses old simple framebuffer device. Kernel is not participating in initializing/driving it in this case - initializing is done by u-boot and just passing memory address mapped to the screen to the kernel. In the image that I have built myself I couldn't make that framebuffer to rotate. Since that is showstopper for me, I went investigating other options. I didn't try 19271 though. Maybe there's something I still don't know.

Also, I'm under impression that running simple framebuffer prevents any possible hardware acceleration from lima driver, so I'm not looking further into that approach. I might be wrong on that, but I can't see yet how these two can work together.

linuxianer99, 2020-02-07:
So both statements from above are correct :-)

The images with sunxi kernel and MALI support used a kernel cmd line option to rotate the screen.

The newer images with mainline kernel are using simple FB which is inititalized by uboot and never touched by the kernel. So it is not sure that LVDS is working with kernel modules. On the other hand, the mainline effort matrix (https://linux-sunxi.org/Linux_mainlining_effort#Status_Matrix) states that it should work starting with kernel 4.16.

@kedder: Is it possible to rotate the simple fb from uboot ? Maybe that can be used a interim solution ? 
I am not sure if hardware acceleration is really needed for the map rendering .... My tests on the 7" display are very responsive even on the simple fb.

kedder, 2020-02-07:
> Is it possible to rotate the simple fb from uboot ? Maybe that can be used a interim solution ?

I couldn't figure out how to do this. I can rotate the linux console (you accepted the patches for that), but I can't find any knob to rotate the simple-framebuffer. 

> I am not sure if hardware acceleration is really needed for the map rendering .... My tests on the 7" display are very responsive even on the simple fb.

Yeah, that's the thing. I'm more or less fine with performance aspect too, so simple FB would be fine, even if it eats 40% of cpu. But the rotation aspect is a blocker (I've already cut the hole in my instrument panel :) ). As I understand, simple FB is considered an "old tech" and nobody cares about it too much.

OTOH, all I really tried is `echo 1 >  /sys/class/graphics/fb0/rotate`, which does nothing in my case (which is of course because kernel does not manage the screen in this case, as I know now). So it is either u-boot should set the screen up rotated (somehow, not sure if that's possible), or XCsoar should rotate the image it produces by itself (which, again, I'm not sure is easily achievable with fb output).

kedder, 2020-02-07:
So, the only reason I'm so deep into DRM/KMS/LVDS support is that it seems that screen rotation should be possible in that case. OpenGL acceleration is just a nice side-effect we can get from it :) 

But don't hold our breath for it: "plaes" from linux-sunxi ML says A20-LVDS support is not there (even though I suspect his data is a bit outdated).

kedder, 2020-02-07:
> Is it possible to rotate the simple fb from uboot

Now that I think about it, I have suspicion that it is not possible. I guess what is happening is the some hardware component on A20 has access to some memory area and it is generating pulses for the screen by reading bits directly from that memory area. Pixel by pixel, line by line, in panel's "natural" orientation. So it basically linearly scans the memory and sends data directly to the panel. "Rotating it from u-boot" would mean that hardware would need to be set up to do the translation, which I doubt is a feature, most likely not.

In DRM/KMS world, there is a ton of abstraction between buffers that software puts pixels on and actual screen hardware, so there is plenty of opportunities for rotation/translation/zooming and stuff like that.

kedder, 2020-02-08:
Who's the man?!

For the first time in history of mankind! Cubieboard 2 with LVDS screen shows a real lima-driver accelerated graphics by `kmscube`!

[image]

mihu-ov, 2020-02-08:
Woohoo! Good job kedder!

kedder, 2020-02-08:
No, xcsoar doesn't work (starts, but shows a black screen), but that doesn't feel horrible anymore :)

kedder, 2020-02-09:
Patch enabling LVDS on A20: https://github.com/torvalds/linux/compare/v5.5...kedder:a20-lvds-support

linuxianer99, 2020-02-10:
Nevertheless great job !!
Think this is a great starting point for futher development !!

kedder, 2020-02-11:
Status update: tried to compile xcsoar with different options: mesa/kms, wayland, sdl2, X11. Tried older xcsoar-6.8. Result is the same all the time: xcsoar starts, responds to input, but renders the black screen/window. Have no idea why though, will try to figure it out. 

It doesn't look there's a simple way to rotate emulated framebuffer, even when display is driven by the kernel. Didn't look too deeply into this, but tricks that I tried (setting `rotation` display property in device tree, writing to `/sys/class/graphics/fb0/rotate`) didn't work.

Made #4 to enable DRM/KMS/lima support for openvario, but it still needs some work (don't merge it yet). Feel free to try it out though. I'm going to spend some time trying to get my [LVDS patch](https://lkml.org/lkml/2020/2/11/40) into mainline kernel, then look into XCSoar graphics issues.

kedder, 2020-02-15:
Another update: still no much luck. I'm mostly busy getting my patch to mainline, so no much progress with openvario. 

One thing I tried though is to get proprietary mali driver back to mainline kernel. I did compile the kernel module and installed binary blobs (see https://github.com/Openvario/meta-openvario/commit/a01c6f39ba5a15044d4867e37cfcb10bf26e04b2). But no, it didn't work. XCSoar fails to start with `eglGetDisplay(EGL_DEFAULT_DISPLAY) failed` message.

kedder, 2020-02-16:
> AFAIK the XCSoar display orientation configuration was never used

Just discovered that DisplayOrientation xcsoar setting *was* actually the mechanism behind xcsoar rotation. We replaced that setting every time xcsoar was starting (see https://github.com/Openvario/meta-openvario/blob/warrior/recipes-apps/ovmenu-ng-skripts/files/xcsoar_config.sh). So, yes, in a sense the xcsoar config was ignored, because it will be replaced each time xcsoar starts. But it still was used to tell xcsoar how to render the screen.

That seem to only work when xcsoar uses opengl for rendering.

CazYokoyama, 2020-02-16:
So, on openvario, display orientation is transferred from /boot/config.uEnv to DisplayOrientation in /root/.xcsoar/openvario.prf. XCSoar displays according to DisplayOrientation. However, XCSoar which draws in frame buffer has no way to rotate drawing. Display rotation is only available on OpenGL.

Your warrior-kms patche enables OpenGL/GLES2, doesn't it? If so, when xcsoar is built with OpenGL=y GLES2=y EGL=Y, it can rotate drawing, correct? I'm confused.

kedder, 2020-02-16:
> Your warrior-kms patche enables OpenGL/GLES2, doesn't it? If so, when xcsoar is built with OpenGL=y GLES2=y EGL=Y, it can rotate drawing, correct? I'm confused.

Not really. On warrior-kms I only enable DRM drivers for the screen, no changes to the xcsoar yet.

I tried switching xcsoar to opengl (with lima drivers), but it just draws a black screen. It might be a rotated black screen though, but I cannot really confirm that :) Anyways, I'm going to try that again and dig deeper into it.

kedder, 2020-02-16:
I've started `warrior-lima` branch with opengl-enabled xcsoar: https://github.com/kedder/meta-openvario/tree/warrior-lima

It adds fairly recent mesa with lima drivers, lima kernel module and xcsoar with `ENABLE_MESA_KMS=y` build parameter enabled. This should be as close as possible to previous mali-based setup, but again, xcsoar draws a black screen at the moment.

kedder, 2020-02-20:
Ladies and gentlemen, victory! I've made xcsoar to run under lima, drm, kms, opengl and bunch of other cryptic buzzwords I had no clue about just a month ago. There is still some issue with the font on the front screen, but at least xcsoar seem to show something. Mouse cursor works, keyboard input also works.

[image]

[image]

This was actually a compilation issue, but I had to learn OpenGL in order to track it down. Apparently the "black screen" issue is caused by `-ffast-math` compiler flag. Duh!

I'm going to check how screen rotation work and submit a patch tomorrow.

Meanwhile, my kernel patches got accepted, so hopefully next stable kernel will have them.

mihu-ov, 2020-02-20:
What an achievement, thank you! Could you inform the XCSoar devs about the compiler flag issue just in case this shows up on other platforms as well?

kedder, 2020-02-20:
@mihu-ov good idea, I'll post a question to XCSoar ML.

linuxianer99, 2020-02-21:
@kedder: Great work !! 
After testing the rotation i think we are ready for a public alpha release  !!

We see the font issue also with FB. There was a fix by Caz
https://github.com/Openvario/meta-openvario/blob/warrior/recipes-apps/xcsoar/files/0001-avoid-tail-cut.patch

linuxianer99, 2020-02-21:
Tested on 7" display:
Lima working !
Also the font error is gone (other resolution then 5,7").

@kedder : May you prepare a pull request ? ;-)

kedder, 2020-02-21:
> Tested on 7" display:
> Lima working !

That's great to hear! I actually applied timings blindly without any testing. There is still an issue with 4" screen, we need someone who owns it to help with it. It is likely not LVDS one, so I suspect the same pattern I used for 7" will not work.  

> @kedder : May you prepare a pull request ? ;-)

Of course! I will post a bunch this evening and over the week-end.

linuxianer99, 2020-02-21:
The 4,3" is RGB. It is my own implementation of the Openvario for gliders with smaller instrument panel.
I think i can do this adaption.

kedder, 2020-02-21:
It is not all rainbows and unicorns just yet :)

Screen rotation still doesn't work yet (argh!). Screen with `DisplayOrientation=1` looks like this:

[image]

It kinda tries, but not there yet :/

Also, there are some stability issues. From time to time xcsoar freezes with message from lima in system log:

```
lima 1c40000.gpu: mmu page fault at 0x1434000 from bus id 0 of type read on gpmmu
```
I think we need to try newer kernels to address that. 5.2-rc2 was one of the first with lima driver integrated, it is likely some issues got fixed in subsequent releases.

I'll start posting MRs with what I have, I think we are on the right path and we can always revert to software rendering if we really want to.

kedder, 2020-02-21:
> The 4,3" is RGB. It is my own implementation of the Openvario for gliders with smaller instrument panel.
> I think i can do this adaption.

@linuxianer99, I think it is a matter of making dts file similar to what I did for 5.7" and 7". However panel definitely should *not* have `compatible="panel-lvds"` in it. FWICT, driver will revert to "rgb" mode if it is not "panel-lvds".

kedder, 2020-02-21:
> From time to time xcsoar freezes

Symptoms look similar to this https://gitlab.freedesktop.org/lima/mesa/issues/105. It was fixed 4 months ago, so good chances that newer kernel will fix our problem as well.

CazYokoyama, 2020-02-22:
Your intention is "run xcsoar with OpenGL instead of FB", correct? You create 3 branches. Please combine those 3 branches into multiple patches in a single PR. Otherwise, we can't find what you are doing.

kedder, 2020-02-22:
@CazYokoyama yes, that is my intention (and the theme of this whole thread). I've split the work into 3 self-contained PRs to make reviewing easier. Wasn't "There are so many files which make reviewer difficult" your complaint in #4?

I'm fine with combining them into one PR if that helps in any way.

kedder, 2020-02-22:
Apparently the rotation issue is an XCSoar regression introduced in `v7.0_preview13`. `v7.0_preview12` doesn't have the problem - screen is rotated as expected.

Do you think we can ship `v7.0_preview12` before we figure out the issue upstream?

kedder, 2020-02-23:
Newer kernel (5.5.3) doesn't solve the stability issues. Sigh :(

Now with different error in syslog though.

```
[drm:lima_sched_timedout_job [lima]] *ERROR* lima job timeout
lima 1c40000.gpu: gp task error int_state=0 status=8a
```

kedder, 2020-02-23:
With mesa-20.0.0, that was released just 4 days ago, I can not reproduce freezes (a.k.a. stability issues) anymore. Also, rendering artifacts on map are gone too. XCsoar runs for 20 minutes without any visible issues.

This all is so bleeding edge though...

Anyways, how do you prefer your PRs cooked? All in one piece or split into semi-independent features?

kedder, 2020-02-23:
After I downgraded XCSoar to 7.0-preview12, I have a fully functional openvario with rotated screen:

[image]

linuxianer99, 2020-02-23:
Great !! 
I merged your PRs just a minute ago.

But it is an easy task to create a xcsoar recipe for 7.0-preview12.
Think we can build an alpha release image with this version, can't we ?

kedder, 2020-02-23:
@linuxianer99 thanks! 

> Think we can build an alpha release image with this version, can't we ?

Sure, I'll make a PR with 7.0-preview12 in a moment.

kedder, 2020-02-23:
@linuxianer99, I just posted #9. I do not intend to upgrade the kernel just yet: it seems preview12 runs on 5.2-rc2 just fine. Also, the upgraded kernel I made lacks networking support for some reason, so it is not fully ready for release yet.

kedder, 2020-02-23:
Well, there's one more problem - default xcsoar profile formatted with DOS line endings and xcsoar start script fails to replace orientation with selected one :) Let's fix this too before release (will do).

kedder, 2020-02-23:
@linuxianer99 ok, you can merge #9 (I've added few more fixes there) and do a release! Feel free to test it too :)

hpmax, 2020-02-24:
@kedder I apologize in advance for the ignorance of this comment.

I was running the openvario build environment published on openvario.org (old Angstrom release) and wanted to make some updates to sensord, and found it hopelessly broken, to the point that I couldn't get any working executables.  The networking functionality on Angstrom also seemed... flaky.  At someone's urging I attempted to download and build this perhaps 1-2 weeks ago.  And found out the hard way about the rotation issues.

Could you, pretty please, tell me how to update the build and/or reconfigure to make the display work properly (I' using the 7" version of Stefan's hardware).  If I should wait a day or two for a new release that's okay too...

kedder, 2020-02-24:
@hpmax, you're right, the old build is not working anymore. We are trying to revive it here. 

If you really in hurry, you can try to build the new image we are working on (you'll need to merge #9 manually to make it work ATM, but I really hope @linuxianer99 will merge it soon so you don't have to). We are working on `warrior` branch. Follow the README.md of this repo and after 3-4 hours of compiling you should get the image. You'll need around 50GB of free space for that.

Also, new alpha images should be published soon, so it might be worth just waiting couple of days.

Please note, that until now we were focused mainly on fundamental existential things - making graphics subsystem and xcsoar work, so I can't promise any improvement in networking or sensord just yet (in fact there may be even regressions - I didn't test these areas very thoroughly yet). Any extra hands on this is would be really really helpful, so I'm willing to guide you through the building process if you would like to improve something. Let me know if you get stuck anywhere (feel free to open a new github issue for that).

hpmax, 2020-02-24:
@kedder Sounds like I should wait for @linuxianer99.  I'm glad there is active development going on.  Obviously, the priority should be a stable build environment where things that are supposed to work, do, before additional development occurs.  I was able to build whatever was published here a couple weeks ago, and when I tried it, I experienced what I can only describe as weird screen behavior (as if the screen were both rotated and improperly sized, and attempts to rotate it, did not seem to help matters).

My goal was to get more stable networking (might have been my hardware, but I'm suspecting Angstrom doesn't play well with IPv6 and/or my wireless device)  as well as to improve sensord to get proper voltage (the multiplier will never give good results without an offset), temperature readings, and a flap sensor, and possibly AHRS data from the 9150 (via the jfwells code).  

My skillset may not be ideal for providing assistance with working this stuff out.  I'm an electrical engineer, with pretty expert level C skill, but more limited C++ and Linux configuration -- I know enough to be dangerous.

kedder, 2020-02-24:
> I experienced what I can only describe as weird screen behavior (as if the screen were both rotated and improperly sized

I figured that out literally just yesterday. So building `openvario-image` (with #9 applied) should give a properly rotated UI. But yeah, probably the most sensible thing now is to wait for that to get merged in.

> My goal was to get more stable networking

That's probably my next goal too. I have a Ralink RT5370 wifi usb dongle and I want to make sure it works. I have high stakes on wifi connectivity in Openvario. BTW, what was the problem with "stable networking" in your case?

> improve sensord to get proper voltage (the multiplier will never give good results without an offset), temperature readings, and a flap sensor, and possibly AHRS data from the 9150 (via the jfwells code).

Ah, that's cool! I don't have sensors on my device and my only complaint is that sensord eats ~100% cpu when no hardware is connected. But that's pretty low on my todo list, because I can simply turn the daemon off.

> My skillset may not be ideal for providing assistance with working this stuff out. I'm an electrical engineer,

Oh trust me :) I'm a python webdev, I had no clue about all that graphics stuff just a month ago, but then found myself contributing to the linux kernel in the process of figuring it all out :) Lack of time or motivation (mainly) might be an obstacle, not the skillset.

linuxianer99, 2020-02-24:
@hpmax So the build environment is updated:
https://github.com/Openvario/meta-openvario

It now uses a docker container as well as the repo tool for setting up the sources.
You an give it a try ...

linuxianer99, 2020-02-24:
As XCSOAR works now, think we can close this issue ...
If there are further things please open another issue.

In my opinion it makes no sense to discuss improvements for sensord in the meta-openvario repo

closed 2020-02-24 by linuxianer99

kedder, 2020-02-24:
Yeah, this was tough!