Openvario/meta-openvario #11: Release a preview image

kedder, 2020-03-01:
Image built from current `warrior` branch seems to be stable enough for public testing. Please publish new images and make a public announcement.

@linuxianer99, do you need any assistance with that?

linuxianer99, 2020-03-01:
@kedder : It is just a problem with time ;-)
Let me have a look this week in the evenings.

We have also discuss with @mihu-ov  and Andi where to put these images ..
Either on the "old location" openvario-ftp or here in the github releases.
What do you think is the best location ?

kedder, 2020-03-01:
@linuxianer99 understandable :)

WRT location - there is one consideration that comes to mind: it would be nice to keep opkg repository available for users, so they can download additional packages / upgrade without reflashing the SD card. That will probably still require to keep FTP server around. For the images themselves, I don't really have an opinion :) whatever seems easier.

linuxianer99, 2020-03-01:
ok.
So i tried to build an image a few minutes ago. There is a problem with XCSOAR downloading boost at the moment (403). Maybe it works again tomorrow..

mihu-ov, 2020-03-01:
Why not have the images in both locations? On the Openvario FTP (our homebase) and github as backup? 

Its too early for the Openvario preview image but as I am thinking about it:
XCSoar 7.0 preview 12 is no official release and 1.5yrs old, would it be possible to have the latest official 6.8.x as an alternative?
Also what about Tophat and LX8000, Caz and Bruno had these integrated (at least experimental). And there was an issue with variod audio output and XCSoar airspace warnings (I think Caz got them working) at the same time.

kedder, 2020-03-01:
BTW, it seems like there is only 500 MB of storage available on guthub: https://github.com/pricing. That's about 2 sets of openvario images.

kedder, 2020-03-01:
> XCSoar 7.0 preview 12 is no official release and 1.5yrs old, would it be possible to have the latest official 6.8.x as an alternative? Also what about Tophat and LX8000

I'd suggest to have these and other forks as downloadable .ipk packages, but afaik these are not building as part of openvario warrior build right now. I've built 6.8 with couple of custom patches while I was investigating graphics problems, so I can get it in, if we want to (I'm personally just fine with preview 12). Didn't try the other forks, but I guess it shouldn't be super-hard.

kedder, 2020-03-01:
@linuxianer99 lol, the 403 is not going away tomorrow :) The boost archive was removed from the hosting, see https://fossies.org/linux/misc/boost_1_70_0.tar.bz2

It seems we need to patch it to download https://fossies.org/linux/misc/boost_1_72_0.tar.bz2 instead.

kedder, 2020-03-01:
> lol, the 403 is not going away tomorrow 

Ok, spoke too soon.  https://fossies.org/ is just a mirror, so maybe the [primary location](https://dl.bintray.com/boostorg/release/1.70.0/source/boost_1_70_0.tar.bz2) will be fixed (the file is still seem to be there)

CazYokoyama, 2020-03-01:
> Tophat and LX8000, Caz and Bruno had these integrated (at least experimental). And there was an issue with variod audio output and XCSoar airspace

My code for LK8000 and Tophat is in branch. I'll push the branch.

For variod audio and XCSoar airspace warning,
1. Kill variod
2. vario data from 4353 instead of 4352, i.e. from sensord.

In addition, I want to have
1. build config.ko which creates /proc/config.gz. It is kernel .config file.
2. build pl2303.ko which is serial-USB converter.

You may use my home PC, www.caztech.com as a download site.

CazYokoyama, 2020-03-02:
LK8000 and Tophat: tophat_lk8000 branch in git@github.com:CazYokoyama/meta-openvario.git
config.ko and pl2303.ko: IKCONFIG branch in git@github.com:CazYokoyama/meta-openvario.git

linuxianer99, 2020-03-02:
@kedder : The 500MB are just for private repos. We are complete public and open-source. Think this limit will not apply ..

BTW XCSoar build works again ...

linuxianer99, 2020-03-03:
@kedder: I built two images yesterday evening. Works well. I also had a look for the opkg feed. Implementation is almost finished ... I need some test before release. .. so i am working on it ..

kedder, 2020-03-06:
Btw, I'm more or less ready to announce this little app I was working on for last couple of weeks: https://github.com/kedder/openvario-compman. The only thing missing is the image to install it on :) So, really looking forward for it!

mihu-ov, 2020-03-07:
@kedder: could this be extended to work with Dropbox or similar for the non-competition pilots?

linuxianer99, 2020-03-07:
@kedder: Image is release !
@mihu-ov : I also uploaded to ftp.openvario.org. Can you please write the news ?

@kedder: Also the opkg feed are online. Which packages do you need for your tool ?
I can compile and upload to the feed as well ...

closed 2020-03-07 by linuxianer99

kedder, 2020-03-07:
@linuxianer99 That's awesome!

kedder, 2020-03-07:
> could this be extended to work with Dropbox or similar for the non-competition pilots?

@mihu-ov yes! If not in scope of that particular app, then the other. I'd really like to make programming for openvario more accessible to people, so that it would be easy to implement dropbox integration or syncthing integration or automating flight track delivery or whatever makes your life easier. 

There are so many opportunities there! It takes some work to make it simple / easy to make and publish your own apps, but now with new images we can move there.

(sorry for off-topic @linuxianer99, we'll continue on xcsoar forums, once I make an official announcement for compman thingie)

mihu-ov, 2020-03-08:
That´s great, I am travelling but looking forward to testing the image tomorrow evening!

mihu-ov, 2020-03-08:
I just could not wait until tomorrow ;-)
This works really well already! My summary (from my notes added below):

- The image boots into XCSoar, OVmenu settings work, rotation works, upload / download works, WiFi dongles work (RT5370 and RTL8188CUS tested)
- USB thumb drive automount missing (there is a pull request from kedder already)
- touch works, but calibration / rotation matrix not activated?
- variod sound output broken
- OVmenu font size should be larger in my opinion
- no connman-ncurses (but WiFi setup with connmanctl works fine)
- Alps Hi Res map is included in image, could be removed to reduce image download size
- seems no passwd is set? Maybe set one ("openvario" ?) to avoid issues with SSH?
- Can´t open SFTP channel (“SFTP channel closed by server”, used to work in old image)
- Can´t use USB tethering from my cell phone (used to work in old image)
- Processor load (and power consumption) is higher than in previous image (~20-25%, was ~10%) but zoom in/out is quick

Notes:
Image 20067
Flashed with Balena Etcher on 32GB Samsung µSD
Boots into XCSoar, sensor data available, touchscreen works!
Sound output does not work (noise from speaker, no tone)
Openvario.prf and map loadad, openvario.xci available, all OK
OVmenu with small font size, should be larger (yes, may cause issues in shell)
RT5370 detected
No connman-ncurses
Set up Wifi according to https://www.ev3dev.org/docs/tutorials/setting-up-wifi-using-the-command-line/
Calibrate sensors ->OK
Languages available are English, German, France, Italian, Hungary, Poland, Czech, Slowak
Touch Cal works
Set rotation 90° and reboot
Start up rotated, great! 
Touch rotation matrix incorrect!
Set rotation 0° and reboot
Touch calibration not correct
Processor load ~20-25%, higher than previous image, also higher current consumption
Map zoom is snappy!
Alps hi-res map is included in image? Rather large file, remove and save on image size?
Upload files: /dev/sda1 not mounted
Mkdir /usb
Mkdir /usb/usbstick
Mount /dev/sda1 /usb/usbstick
Upload USB stick to XCSoar works (after manual mounting of USB stick), upload is from /openvario/upload/xcsoar
passwd -> change password
SSH login with bitvise SSH works
Cannot open SFTP window (“SFTP channel closed by server”)
IGC download works (after manual mounting)
Boot time 10s until OV logo shows, 25s to OVmenu, 30s into XCSoar
RTL8188CUS detected
USB tethering does not work

mihu-ov, 2020-03-08:
@kedder / @linuxianer99 
You have done the work so the honor of announcing the new preview image belongs to you! 
If you don´t want that I will announce it in the XCSoar forum tomorrow afternoon.

kedder, 2020-03-08:
> seems no passwd is set? Maybe set one ("openvario" ?) to avoid issues with SSH?
> Can´t open SFTP channel (“SFTP channel closed by server”, used to work in old image)

Hm, I have no issues with SSH / SFTP - works out-of-the-box. I do `ssh root@<ip>`.

kedder, 2020-03-09:
> You have done the work so the honor of announcing the new preview image belongs to you!
If you don´t want that I will announce it in the XCSoar forum tomorrow afternoon

@mihu-ov, please do! The more people involved into the project in one way or another the better :)

mihu-ov, 2020-03-09:
Done, see https://forum.xcsoar.org/viewtopic.php?f=30&t=3642