Openvario/meta-openvario #12: Menu not finding USB drive

iglesiasmarianod, 2020-03-03:
Hi guys, I've fixed a misnomer in the Menu scripts to allow file download from OV to USB and added an option to delete all igc stored files.
Also added a Spanish language option to the menu.
Works OK with previous OV buildsystem, now it seems it does not mount USB drives.
If you'd like to have a look to the scripts a may add a pull request. Perhaps it can merged.
Let me see if I can figure out why USB seems not to mount.

Regards.

linuxianer99, 2020-03-03:
The reason for the missing USB drive is a missing auto-mounter in the image. It was not ported up to now to the warrior based image.
We can do this in the future, but i suggest, we will release an preview image now.

In the next release, we can fix the missing things discovered during tests with the image ...

iglesiasmarianod, 2020-03-03:
Hi Timo! It's OK for me.
I agree these mods can be included in the future.

PR #15 (2020-03-10): Enable USB drive automount
This includes autofs with respective kernel module to support USB stick automounting.

Fixes #12
closed 2020-03-10 by linuxianer99