Openvario/meta-openvario #16: Texim 7" display doesn't work

kedder, 2020-03-09:
Users are reporting that 7" Texim display doesn't work with the recent (20067) image.

>  Here's a video of non-working image: https://drive.google.com/open?id=1qQbzZW6G60sZAgtfYW7ZHiwC8yZA5LGY

Apparently 7" Texim has different resolution (800x480) than PixelQi one (1024x600). So devicetree files we provide are obviously wrong. U-boot configuration is also wrong. 

We need to pick the right timings from old fex file.

See `recipes-kernel/linux/linux-mainline/openvario-7-CH070.dts`, `recipes-bsp/u-boot/files/openvario-7-CH070/openvario_defconfig`.

kedder, 2020-03-09:
Testing image to try for 7" Texim users is here: https://www.dropbox.com/s/ahj0s0oel3qqo0z/OpenVario-linux-openvario-image-glibc-ipk-20069-openvario-7-CH070.rootfs.img.gz?dl=0

Please try and report back!

Made from my fork at https://github.com/kedder/meta-openvario/tree/warrior-ch070.

closed 2020-03-11 by linuxianer99

bomilkar, 2020-03-21:
Hi kedder, just tested your image OpenVario-linux-openvario-image-glibc-ipk-20069-openvario-7-CH070.rootfs.img.gz.
Scaling seems to be off, at least for me. See attached pictures.
I verified the H/W with an old image (a copy of SDcard from my glider). This works as expected.
[image]
[image]
[image]
[image]

kedder, 2020-03-21:
@bomilkar can you double-check the image you have used? My image comes with xcsoar 7.0-preview12, not preview14 on your screenshots.

bomilkar, 2020-03-21:
You are right. I must have screwed up writing the SDcard. The 2nd partition must have survived from a previous test. Your hint with "preview14" made me think ...
Now it works as expected. Thanks!

PR #22 (2020-03-11): Support Texim 7" screens
Timings for Texim 7" was copied from PQ display and wasn't right. I took the new settings from old .fex file and adjusted boot images for Texim resolution (800x480).

Fixes #16 

@linuxianer99, please make sure we provide images for `7-CH070` too.