Openvario/meta-openvario #17: Touch screen calibration / rotation

kedder, 2020-03-09:
According to @mihu-ov, touch screen calibration and rotation doesn't work in 20067.

> touch works, but calibration / rotation matrix not activated?

kedder, 2020-03-11:
The `caltool` that is intended to calibrate touchscreen seems to be broken. It segfaults when started with any `-r` parameter (intended for rotation). 

It doesn't run on laptop either: requires root to run and completely hangs, doesn't react to Ctrl-C and requires laptop reboot.

Fixing this requires access to hardware (which I don't posses, unfortunately). @linuxianer99, can you look into this?

linuxianer99, 2020-03-11:
@kedder : Yes, i can do so. But nit before the weekend ...

SashLi, 2020-05-03:
Is there any new information to this issue ?

linuxianer99, 2020-05-12:
@kedder : the caltool used before is very fragile.
Maybe we can use tslib https://github.com/libts/tslib/blob/master/README.md#setup-and-configure-tslib instead ?
There is a possibility to emulate a "calibrated touch" via uinput in linux kernel.
Do you know if it is possible to set the input device XCSoar is using ?

kedder, 2020-05-13:
@linuxianer99, I've never dealt with touchscreens before, so it is hard to say.  I'm not even sure what mechanisms/apis are involved in reading the touchscreen input. I can see tslib is included in `meta-oe`, so it should be easy to build it. But again, since I don't have the touchscreen hardware, I can't test it.

linuxianer99, 2020-05-13:
I compiled tslib yesterday evening.
Installation worked fine but i need uinput kernel module. So i have to compile a new kernel for testing.

The only thing is, don't know if i can tell XCSoar to use the /dev/uinput instead of /dev/input/eventx ... We will see ;-)

marulidu, 2020-05-13:
Just completed installation of new touchscreen unit and come across this issue since screen is installed at portrait, 270 orientation. Cursor stays at upper edge despite of touch area. Extremely disappointing. Touch feature became totally useless. I could be more then happy to contribute to software testing.

linuxianer99, 2020-05-22:
@kedder : Think i have a solution for the touch:

Using tslib with tslib_uinput:
tslib_uinput make a virtual event device with calibration and rotation information.

Things to do:
- disable real touch device for libinput using udev ENV.
`ACTION=="add|change", KERNEL=="event[0-9]*", ENV{ID_PATH_TAG}=="platform-1c25000_rtp", ENV{LIBINPUT_IGNORE_DEVICE}="1"`

- set Env variable `TSLIB_TSDEVICE=/dev/input/event5` to tell tslib which device to use.
- start tslib_uinput: This uses event5 and emulates event7 with all overlay features tslib can do (calibration, rotation, sensitivity, ...)

Do you think this is a useable approach ?

kedder, 2020-05-22:
@linuxianer99, This won't require any change to XCSoar to support touch events via tslib, right? if it works, it looks great!

linuxianer99, 2020-05-22:
@kedder : Yes, no changes to XCSoar.
Just udev things ... and a bit of configuration ...

Blaubart, 2020-05-28:
I compiled the new image last night, but touch screen dosen't work. I can't calibrate touch screen. If I start the calibration tool I get the message touch screen is calibrated. If I start XCSoar I have no reaction about the touch screen.

marulidu, 2020-05-28:
Same with mine. Actually touch moves cursor along upper edge only

linuxianer99, 2020-05-28:
@Blaubart 
@marulidu 
Which type of image did you build ? openvario-testing ?
Which type of hardware do you have ?

Blaubart, 2020-05-28:
I have a OV with Pixel Qi 7.0 and tried openvario-testing

linuxianer99, 2020-05-29:
Can you please do the following:

- on the Openvario linux change to` /dev/input/by-path`
- Do a `ls -alh` and post this output here

Blaubart, 2020-05-29:
[image]

linuxianer99, 2020-05-29:
@Blaubart Do you have the possibility to change files config files in the linux system ?
Or do you need a full image for testing a fix ?

Blaubart, 2020-05-29:
I can change files, no problem

linuxianer99, 2020-05-29:
Try to change the following:

in `/home/root/.profile`
from
`export TSLIB_TSDEVICE=/dev/input/event5`
to
`export TSLIB_TSDEVICE=/dev/input/by-path/platform-1c25000.rtp-event`

and
in `/etc/ts.env`
from
`TSLIB_TSDEVICE=/dev/input/event5`
to
`TSLIB_TSDEVICE=/dev/input/by-path/platform-1c25000.rtp-event`

Blaubart, 2020-05-30:
Now I can calibrate the touch screen, but it dosen't react to my finger when XCSoar is running

linuxianer99, 2020-05-30:
Did you do a reboot after the calibration ?
The thing is that there is a systemd service (ts_uinput) which needs to be restarted to apply the calibration. So the easiest way is to reboot.

Blaubart, 2020-05-30:
Yes I did.

I don't know why, but now the arrow moves with touch screen. But at X-axis arrow moves just half way. At Y-axis arrow reaches the end of display but my finger moves just have way.

kedder, 2020-06-08:
Thanks to Stephan Langer, who generously donated me the hardware for development, I have a 7" device with the touchscreen (and sensors too) and can play with it now!

I've checked out the image with the tslib support and found couple of issues:

1. The touchscreen is not always registering as `/dev/input/event5` - the number actually depends on how many other input devices (keyboards, mice, etc) are connected. If it is not on `event5`, ts_uinput will not start at all (therefore tochschreen won't work). We can use `/dev/input/by-path/platform-1c25000.rtp-event` instead, as @linuxianer99 suggested.

2. The touchscreen appears to work only for the first time XCSoar is started - if you exit and start the XCSoar again, touchscreen doesn't work until you restart ts_uinput service. Also, when it works, it does in somewhat weird way - the touch screen appears to be always "pushed", I can draw red lines on XCSoar map, and it will not process the gesture even when I'm not touching the screen anymore. Taps (clicks) are not registered - I can't select anything or bring up menu - just draw with the red line.
The `ts_uinput`, if run with `-v` switch would log the pressed coordinate even when XCSoar would not react to touchscreen.

3. The `TSLIB_TSDEVICE` env variable doesn't seem to have any effect on XCSoar, so there's no point to put it to `.profile` (unless I'm missing something)

I'll play with it a little bit more, maybe I'll be able to come up with something working.

kedder, 2020-06-09:
I think I made tslib work, including screen rotation, but it is a bit hacky  (using `LIBINPUT_CALIBRATION_MATRIX` to force the rotation), so I'm not too happy with it just yet.

Without ``LIBINPUT_CALIBRATION_MATRIX` hack I've observed the same behaviour on rotated screen as @Blaubart, and this seems like a bug (or just missing feature) in tslib version we use right now (1.16). It seems like [1.18](https://github.com/libts/tslib/releases/tag/1.18) has some better support for screen rotation and I'll try to upgrade and see if it improves things.

kedder, 2020-06-12:
Ok, I think I figured it out. @Blaubart, can you try this new image: [20164-openvario-7-CH070](https://www.dropbox.com/s/yg24eis0imqsvdh/OpenVario-linux-openvario-image-testing-glibc-ipk-20164-openvario-7-CH070.rootfs.img.gz?dl=0)?

The touchscreen should work in all 4 directions after calibration. First, pick your screen orientation, then reboot, then calibrate the touch screen, then start xcsoar.

Blaubart, 2020-06-12:
Thats sounds good!! Thanks! I use a Pixel Qi 7, so can you send me the image for this display, please?

kedder, 2020-06-13:
@Blaubart here you go: [20165-openvario-7-PQ070](https://www.dropbox.com/s/qjn74cevxexmynf/OpenVario-linux-openvario-image-testing-glibc-ipk-20165-openvario-7-PQ070.rootfs.img.gz?dl=0).

Blaubart, 2020-06-14:
Hi Kedder,

it almost works ;-)
Touchscreen works really good, but calibration is not saved. If I restart OpenVario I have to calibrate again because touchscreen dosen't respond to the my finger. After calibrate again touchscreen works until I powered off the device.

Regards and thanks a lot!
Blaubart

kedder, 2020-06-15:
@Blaubart yeah, apparently I didn't test it well enough. Please try the new image: [20167-openvario-7-PQ070](https://www.dropbox.com/s/771vn48kmbc2hm3/OpenVario-linux-openvario-image-testing-glibc-ipk-20167-openvario-7-PQ070.rootfs.img.gz?dl=0), it should have the problem fixed.

Blaubart, 2020-06-15:
Thanks a lot!!!! It works perfectly!

Blaubart, 2020-06-15:
How can I see, that it is merged to warrior?

kedder, 2020-06-15:
I'll make a pull request, as soon as it merges, this issue will close, you'll get a notification (probably).

Blaubart, 2020-07-24:
I compiled stable and testing version at the same day with the same sources. Touchscreen calibration with testing-image works, with stable not. It dosen't make sense, right??

kedder, 2020-07-24:
@Blaubart, no, it does not. Does the officially released stable version have the same issue?

Blaubart, 2020-07-25:
I will test next days.
Thanks for helping!!

Blaubart, 2020-07-26:
It is the same with officially released stable version for PixelQi 7.0

linuxianer99, 2020-07-26:
Checked content of image a few minutes ago.
So the PR is in the official release image. Checked the udev rules file.

Just to be sure:
The official testing image works
The official stable not ?

Blaubart, 2020-07-26:
I tried self compiled testing image (works), self compiled stable image (dosen't work) and officially released stable image (dosen't work)

kedder, 2020-07-27:
I'll check as soon as I get to my testing device, hopefully this evening.

kedder, 2020-07-27:
Ok, I can confirm the problem. It seems like XCSoar-6.8 is interpreting the touch input somehow differently from 7. It should be fixable, but I can't do it till the next week.

kedder, 2020-08-05:
@Blaubart, I figured it out. Touchscreen on xcsoar-6.8.x will work after #102 is merged.

Blaubart, 2020-08-05:
great! Thanks for your help!!

PR #86 (2020-06-16): Make touchscreen work
This fixes support for touchscreens.

The basic approach here is to use libinput to handle screen rotation, because tslib own rotation support doesn't work well for some reason. We force tslib rotation to be 0 (newtral) for `ts_uinput` service, and rotate the screen using `LIBINPUT_CALIBRATION_MATRIX` variable for udev devices.

This also gives touchscreen a permanent device path: `/dev/input/touchscreen0` for easier discovery.

Touchscreen works in all 4 rotation directions (AFAIU, it used to work only in 2 on old images). 

Fixes #17.
PR #102 (2020-08-06): Explicitly compile xcsoar-6.8 with libinput support
XCSoar-6.8.x (differently from 7.x) does not compile with `libinput` by default. `libinput` is needed for touchscreen support.

Fixes #17
closed 2020-08-06 by linuxianer99