Openvario/meta-openvario #18: Vario sound is broken

kedder, 2020-03-09:
Vario sound is reported to be broken by several users:

> variod sound output broken

> The audio was odd.  It may have been that things were just incorrectly configured, but my original image, the audio seems very normal compared to a vario.  This  seemed like it was modulated with a low frequency square wave.  I.e. it would just cut in and out constantly.

This gives an idea how it sounds like: https://drive.google.com/file/d/1qQbzZW6G60sZAgtfYW7ZHiwC8yZA5LGY/view

CazYokoyama, 2020-03-09:
I already reported how to fix, didn't I?
-caz

kedder, 2020-03-09:
> I already reported how to fix, didn't I?

Um, really?  Where can I find it?

kedder, 2020-03-09:
Some hints from @CazYokoyama (taken from #11):

> For variod audio and XCSoar airspace warning,
> 1. Kill variod
> 2. vario data from 4353 instead of 4352, i.e. from sensord.

mihu-ov, 2020-03-09:
> 1. Kill variod
> 2. vario data from 4353 instead of 4352, i.e. from sensord.

This is the configuration we had before variod with XCSoar providing audio vario. Back then the update rate was 2Hz which was too slow and really annoying. It is much better now but leaves us without volume control over stick remote / .xci file and without STF (speed to fly) audio.

kedder, 2020-03-11:
The noise is indeed coming from `variod`. OTOH xcsoar own audio vario gives nice clear sound. I'll experiment with @CazYokoyama approach, maybe we can increase the update rate to acceptable level (if that's the only issue with it).

CazYokoyama, 2020-03-11:
MS5611 -> sensord -> tcp 4353 -> variod +-> tcp 4352 -> xcsoar +
                                                                   |                                    |
                                                                  +----------------------------+-> speaker
As mihu said, the path, tcp 4352 -> xcsoar -> speaker is slow. I analyzed sensord, but not variod. To analyze/debug variod, there is no good debug environment. I wish I can run variod on raspbian.

CazYokoyama, 2020-03-12:
Is anyone working for variod? I built debug environment on raspbian. If no one, I will.

kedder, 2020-03-12:
@CazYokoyama please do! 

BTW, I've found that recent gcc (9.2 in my case) cannot build variod anymore, but it is fixable with this patch:

```patch
diff --git a/Makefile b/Makefile
index 4a017e7..591d422 100644
--- a/Makefile
+++ b/Makefile
@@ -22,7 +22,7 @@ doc:
        doxygen
        
 variod: $(OBJ)
-       $(CXX) $(LIBS) -g -o $@ $^
+       $(CXX) -g -o $@ $^ $(LIBS)
        
 install: variod
        install -D variod $(BINDIR)/$(EXECUTABLE)
```

linuxianer99, 2020-03-12:
variod was invented by Andreas Wasserbauer. A friend of Michael Huber.
But he is also very short on time.
@CazYokoyama : If you can debug, feel free
@mihu-ov : Think this is ok, isn't it ?

linuxianer99, 2020-03-12:
One other thing:
variod was invented, as XCSoar vario was not very responsive in the past.
Has this changed in the current releases ??

@mihu-ov : If so, we maybe can remove variod as XCSoar vario is good enough ?

mihu-ov, 2020-03-12:
Andi has no github account yet but answered on the Redmine, I think he is OK with me copying his reply below. My personal preference (and I am no software developer) would be to keep audio vario in variod even if this causes an additional interface to XCSoar because I think

- vario quality is the first priority
- vario information (sensor data to audio) should be as quick as possible
- in flight the audio vario from variod worked really very well (STF not yet) 
- variod could be useful with LK8000, Tophat or other software as well
- further improvements are probably easier implemented and flight tested in a small app like variod than in a package like XCSoar
****

Hi caz, kedder

I don't have a github account yet, so let me just give you an update concerning variod this way.

First of all: It's of course ok to work on XCSoar's vario output, but it is very limited compared to variod. So it's probably quite some work to overcome it's current limitations.

Concerning the broken variod: I know about the issue and I am working on it by replacing the low-level alsa code with pulseaudio. I think that won't be too much work, and should also allow us to have XCSoar and variod output sound in parallel.
From my perspective this is the way to go.

cheers

Andi

kedder, 2020-03-12:
That's awesome news, @mihu-ov!

kedder, 2020-03-22:
Status update for anyone following this: Andreas Wasserbauer started working on porting variod to pulse audio. This will allow us to fix the sound output and also support mixing vario output with XCSoar own events (e.g. airspace warnings).

 Andreas doesn't have github account and was posting updates on openvario redmine server (http://www.openvario.org/redmine/boards/3/topics/1773). Unfortunately redmine doesn't work anymore (Internal error) and we lost the only communication link with him. @linuxianer99, can we get redmine back up at least until we figure this issue out?

mihu-ov, 2020-03-22:
Andi is the Redmine admin, I have contacted him by mail as it is too late to call him.

kedder, 2020-03-24:
I've included pulseaudio server into the image in my branch here: https://github.com/kedder/meta-openvario/tree/warrior-pulseaudio

We can use it as soon as variod is ported to pulseaudio.

kedder, 2020-04-01:
So I was playing with variod, ported to pulsaudio, that Andreas sent me (pushed it to [my fork](https://github.com/kedder/variod/tree/pulseaudio), if anyone is interested).

And it seems like we were barking at the wrong tree this whole time.

Jittering sound is still there.

However, after closing investigation (I've stopped variod service and ran it in foreground with `/opt/bin/variod -d -f`), it became apparent that the jittering sound comes from this:

```
!! DEBUG LEVEL 1 !!
!! STAY in foreground !!
Socket created ...
=========================================================================
Runtime Configuration:
----------------------
Vario:
  Deadband Low:			-0.000000
  Deadband High:		0.000000
  Pulse Pause Length:		48000
  Pulse Pause Length Gain:	1.000000
  Base Frequency Positive:	350
  Base Frequency Negative:	350
Speed to fly:
  Deadband Low:			-2.500000
  Deadband High:		2.500000
  Pulse Pause Length:		48000
  Pulse Pause Length Gain:	0.200000
  Base Frequency Positive:	350
  Base Frequency Negative:	350
=========================================================================
Waiting for incoming connections...
Connection accepted
Connection dropped
Waiting for incoming connections...
Connection accepted
Connection dropped
Waiting for incoming connections...
Connection accepted
Connection dropped
Waiting for incoming connections...
Connection accepted
Connection dropped
Waiting for incoming connections...
Connection accepted
Connection dropped
Waiting for incoming connections...
Connection accepted
Connection dropped
Waiting for incoming connections...
Connection accepted
Connection dropped
Waiting for incoming connections...
Connection accepted
Connection dropped
...
```

My impression is that `sensord` is connecting and disconnecting from variod socket all the time and variod just making a short "beep" each time it does it.

Does it ring any bells?

kedder, 2020-04-01:
Here's the output of sensord that is run in foreground:

```
# /opt/bin/sensord -f -d
!! STAY in g_foreground !!
!! DEBUG LEVEL 1 !!
Opened 24C16 on 0x50 failed !!!
No EEPROM found !!
=========================================================================
Runtime Configuration:
----------------------
Vario:
  Kalman Accel:	0.000000
Sensor TEK:
  Offset: 	0.000000
  Linearity: 	1.000000
Sensor STATIC:
  Offset: 	0.000000
  Linearity: 	1.000000
Sensor TOTAL:
  Offset: 	0.000000
  Linearity: 	1.000000
=========================================================================
Opened MS5611 on 0x76
Error writing to i2c slave (ms5611_reset)
Error writing to i2c slave (write cal reg)
Opened MS5611 on 0x77
Error writing to i2c slave (ms5611_reset)
Error writing to i2c slave (write cal reg)
Opened AMS5915 on 0x28
Opened ADS1110 on 0x48
Error writing to i2c slave (ms5611_start_temp)
Error writing to i2c slave(ms5611_read_temp)
Error writing to i2c slave: start conv: adr 76
Error writing to i2c slave: write Read result(ms5611_read_pressure)
Error writing to i2c slave (ms5611_start_temp)
Error writing to i2c slave(ms5611_read_temp)
Error writing to i2c slave: start conv: adr 76
Error writing to i2c slave: start conv: adr 77
Error writing to i2c slave: write Read result(ms5611_read_pressure)
Error writing to i2c slave: write Read result(ms5611_read_pressure)
Unable to read from slave
ams5915_calculate @ 0x28: Pressure: -6.248569 Temp: -50.000000
ads1110_calculate @ 0x48: Voltage: infV
Error writing to i2c slave (ms5611_start_temp)
Error writing to i2c slave (ms5611_start_temp)
Error writing to i2c slave(ms5611_read_temp)
Error writing to i2c slave(ms5611_read_temp)
Error writing to i2c slave: start conv: adr 76
Error writing to i2c slave: start conv: adr 77
Error writing to i2c slave: write Read result(ms5611_read_pressure)
Error writing to i2c slave: write Read result(ms5611_read_pressure)
Unable to read from slave
ams5915_calculate @ 0x28: Pressure: -6.248569 Temp: -50.000000
ads1110_calculate @ 0x48: Voltage: infV
Error writing to i2c slave: start conv: adr 76
Error writing to i2c slave: start conv: adr 77
Error writing to i2c slave: write Read result(ms5611_read_pressure)
Error writing to i2c slave: write Read result(ms5611_read_pressure)
Unable to read from slave
```

These errors are probably from the fact that I don't have any sensors connected.

linuxianer99, 2020-04-02:
Hmmm ... This is a good point !
This maybe the reason, not everybody has this problem ...
My openvario variod works fine here on my desk .. but i have sensors connected ..

I will hav a look at sensord tomorrow. Adding some detection at the start or something like this ...

mihu-ov, 2020-04-02:
Unless for testing using variod without pressure sensors is pretty pointless, isn´t it?
If we have packages for pulseaudio and variod or an image for PQ070 display I can test (with sensorboard) in the evening.

kedder, 2020-04-02:
I've uploaded packages to my PPA. If anyone wants to try it out, you can do this:

```sh
echo src kedder http://openvario.lebedev.lt/opkg/armv7vet2hf-neon/ >> /etc/opkg/customfeeds.conf
opkg install pulseaudio-server
opkg install variod-testing
```

and restart the openvario. Next time it loads, you can run `alsamixer`, that should show the single volume bar - that's how you make sure pulseaudio is running.

I'll play with it a bit more today evening too.

mihu-ov, 2020-04-02:
```
root@openvario-7-PQ070:~# alsamixer
ALSA lib ../../alsa-plugins-1.1.8/pulse/pulse.c:243:(pulse_connect) PulseAudio: Unable to connect: Access denied

cannot open mixer: Connection refused
root@openvario-7-PQ070:~#
```

kedder, 2020-04-02:
@mihu-ov that means pulseaudio server is not running

kedder, 2020-04-02:
@mihu-ov Oh, I remembered now. You have to do this:

```
newgrp - audio
```

and restart the device. This will be incorporated into the image with pulseaudio, but if you install server manually, you have to do that.

mihu-ov, 2020-04-02:
Thank you kedder, variod is beeping on my desk now and alsamixer shows the single volume bar. I got a bit confused as there were actually two beep tones coming from the speaker but I had XCSoar audio vario turned on as well, so variod and XCSoar audio really go together now!
I need to start variod manually but this is something I may have broken on my OV.

Looks good! (And I need to go back to work but this was too exciting ;-)

kedder, 2020-04-02:
Oh, that's a great news! I'll make a PR this evening.

> And I need to go back to work but this was too exciting

Yeah, same here :)

mihu-ov, 2020-04-02:
OK, after newgrp - audio autostart works again. Great news indeed!

kedder, 2020-04-02:
Ok, so I learned something about the issue:

If I replace sensord with simple `telnet localhost 4353`, I get clear tone of about 700Hz out of my speaker. As soon as real `sensord` starts, sound gets corrupted. "Monitor" view in XCSoar devices screen shows a wall of repeated messages: 

```
$POV,E,+99.00*09
$POV,V,+0.00*2A
$POV,P,+000.00.Q,+0.00*48
```

Again, I don't have any sensor hardware. I need someone with the hardware to check if it really works. I'll create couple of images for testing on real hardware.

kedder, 2020-04-02:
@mihu-ov, please test this image with real sensors:

https://www.dropbox.com/s/emw5qi63ve73mfw/OpenVario-linux-openvario-image-testing-glibc-ipk-20093-openvario-7-PQ070.rootfs.pulseaudio.img.gz?dl=0

That's an image for PQ070 display (I'm assuming you've got that one) and pulseaudio-enabled variod. 

You should hear the nice vario once xcsoar starts, that should react to pressure changes on the sensors.

kedder, 2020-04-02:
Wow :)

Please also check the responsiveness. How quickly does it react to pressure changes? That's something I'm not very confident about. Would it be possible to compare this aspect with the old image?

mihu-ov, 2020-04-02:
I don´t think we can test responsiveness on pressure sensor data easily (even with an avionics test set). As we have changed variod only maybe we could generate a data file, replay it with sensord, record the speaker output and compare the two files?
Do we have to expect any additional delays due to pulseaudio or the new variod update method?

kedder, 2020-04-02:
On old variod the vario tone was recalculated each 100ms. On the new variod the tone is calculated when pulseaudio sound buffer is exhausted. I'm not sure how big that buffer is and whether it will make any difference in terms of responsiveness. I don't expect hard numbers here, only if new version "feels laggy" or not.

PR #45 (2020-04-10): Enable pulseaudio-based variod on testing image
This adds pulseaudio on testing image and uses pulseaudio-enabled variod. Switching to pulseaudio-server based sound will also allow to mix XCSoar own sounds with vario.

Requires https://github.com/Openvario/variod/pull/1.

This is supposed to fix #18.
closed 2020-04-10 by linuxianer99