Openvario/meta-openvario #20: Tophat and LK8000 not included

kedder, 2020-03-09:
Apparently there was an option to run Tophat or LK8000 instead of XCSoar. We only ship XCSoar right now. We might want to ship these, but I'm not sure in which form.

kedder, 2020-03-09:
@CazYokoyama mentioned there is a branch with Tophat and LK8000: 

>  tophat_lk8000 branch in git@github.com:CazYokoyama/meta-openvario.git

linuxianer99, 2020-03-10:
@CazYokoyama : I am not sure if this builds with the new image. Can you please verify if the recipes still work ??

linuxianer99, 2022-05-30:
@kedder Do we still have the strategy of shipping the other apps ?
Think as of the lean firmware strategy, we should offer them as installable packages ??

MaxKellermann, 2022-05-30:
Since nobody has been working on this for at least two years (= no developer is interested), the decision on how to integrate other applications is premature, and should be postponed until there is a working and stable recipe.

closed 2022-05-30 by linuxianer99