Openvario/meta-openvario #23: Next preview Release

linuxianer99, 2020-03-11:
@kedder : When should we build the next preview release ?
What do you think ??

I will also build an image for CH-070 in this cycle ..

kedder, 2020-03-11:
To my understanding, the biggest blocker right now is the audio vario problem (#18), I think we should figure it out before offering new image.

When we solve the touch screen issue (#17), we should be more or less feature-complete, compared to the last stable image (correct me if I'm missing something important).

mihu-ov, 2020-03-12:
It would be good to have automount (already done?), nano editor and SFTP access in the next image to make it better usable for non-IT savvy users. I have opened issues for these.

kedder, 2020-03-12:
Yeah, automount is there already. Thanks for filing those!

kedder, 2020-04-09:
@linuxianer99 #45 is ready, I think we can cut a release once we merge it.

linuxianer99, 2020-04-10:
@kedder : OK. #45 is merged.
I will trigger a release build this weekend.

At the moment, only testing image builds correctly. the stable image has a problem with xcsoar 6.8. think this is a compiler thing as gcc is to new for xcsoar 6.8.

kedder, 2020-04-10:
@linuxianer99 Sounds good! I'll check out xcsoar 6.8. 

Btw, Max had to revert our screen rotation fix, so -testing image will be broken right now as well. I have submitted PR with the updated fix (https://github.com/XCSoar/XCSoar/pull/373), but it is not merged yet. That's something that would block the release too. We might need to pin the xcsoar-testing to last known good commit, or apply the patch ourselves for our build.

kedder, 2020-04-10:
@linuxianer99 what compilation errors for xcsoar did you get? I have tried right now and it compiled just fine.

Nevertheless I've upgraded it to the latest version (6.8.13b) and also applied the rotation fix, see #49.

linuxianer99, 2020-04-16:
@kedder : XCSoar error is

`|   HOSTCXX output/host/tools/GenerateSineTables.o
| g++: error: unrecognized command line option ‘-fmacro-prefix-map=/workdir/poky/build/tmp/work/armv7vet2hf-neon-ovlinux-linux-gnueabi/xcsoar/6.8.13b-r0=/usr/src/debug/xcsoar/6.8.13b-r0’
| build/host.mk:23: recipe for target 'output/host/tools/GenerateSineTables.o' failed
|`  

Seems a problem with the host g++ compiler within the docker container.
g++ version is:
`g++ (Ubuntu 7.4.0-1ubuntu1~18.04.1) 7.4.0
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
`
Testing image with new XCSoar works well. Which g++ host version are you using ??

kedder, 2020-04-16:
Oh, I remember someone had the same error and it was resolved by upgrading debian one version up. For some reason xcsoar uses HOSTCXX, which points to native compiler of the host operating system, and it needs uses `-fmacro-prefix-map` argument which is implemented first in GCC-8.

I have these versions on my environment: 

```
$ $CXX --version
arm-ovlinux-linux-gnueabi-g++ (GCC) 8.3.0
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

$ g++ --version
g++ (Ubuntu 9.3.0-10ubuntu1) 9.3.0
Copyright (C) 2019 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

`$CXX` is the compiler built by yocto, and `g++` is compiler from my Ubuntu.

CazYokoyama, 2020-04-16:
It seems you are not using the latest LTS, 18.04 bionic beaver. Please show
your /etc/os-release.

No, I can't build openvario anymore since a couple of month ago.
 -caz

kedder, 2020-04-16:
> It seems you are not using the latest LTS, 18.04 bionic beaver. Please show your /etc/os-release.

I'm on 20.04 right now, but was on 19.10 a week before.

linuxianer99, 2020-04-17:
@CazYokoyama 👍 
/etc/os-release:

pokyuser@2d19c9ec6a87:/workdir$ more /etc/os-release
NAME="Ubuntu"
VERSION="18.04.3 LTS (Bionic Beaver)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 18.04.3 LTS"
VERSION_ID="18.04"
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
VERSION_CODENAME=bionic
UBUNTU_CODENAME=bionic

I am using the docker build container. I will try another container based on 19.10. Let's see if it is working ...

CazYokoyama, 2020-04-17:
Tim,
I am using similar one.
VERSION="18.04.4 LTS (Bionic Beaver)"

Its default g++ are 7.4.0 as follow.
caz@ov-ubuntu18:~$ apt list | grep ^g++ | grep installed
g++/bionic-updates,bionic-security,now 4:7.4.0-1ubuntu2.3 amd64 [installed]
g++-7/bionic-updates,bionic-security,now 7.5.0-3ubuntu1~18.04 amd64
[installed,automatic]
g++-8/bionic-updates,bionic-security,now 8.4.0-1ubuntu1~18.04 amd64
[installed]

Please aware that 19.10 is not LTS.
-caz

kedder, 2020-04-17:
The problem is that 'warrior' yocto prepares compiler flags in environment variables for gcc it provides (8.3.0). These env vars gets used by `$HOSTCXX` while building xcsoar. If host gcc does not understand compiler flags provided by those yocto env vars, build fails.

It is not ideal that xcsoar build uses host gcc (even though the use is limited to few files), but upgrading the docker image seems easier solution than trying to change the build. Especially as 20.04 is going to be LTS release and it is around the corner.

linuxianer99, 2020-04-17:
so ... build is working...

CROPS (Container for Yocto build) uses only LTS versions. So 18.04 is current at the moment. 20.04 is not available for CROPS right now. But nevertheless i installed gcc 8.4 in our build container. 

Everything seems to build now, but not finished yet ;-)
If both builds will pass (testing and stable) i will tag the repo and start the release process ...

linuxianer99, 2020-04-17:
@kedder  should we merge pull request #37  for the image ??

kedder, 2020-04-17:
@linuxianer99 sounds good!

>  should we merge pull request #37 for the image ??

Yes, let's do that. That seems like a good feature.

linuxianer99, 2020-04-24:
Done ! Released ...

closed 2020-04-24 by linuxianer99

kedder, 2020-04-25:
@mihu-ov, can you do the announcement on the forum?

mihu-ov, 2020-04-26:
Done, https://forum.xcsoar.org/viewtopic.php?f=30&t=3669
Thank you guys!