Openvario/meta-openvario #25: SFTP access not working in image 20067

mihu-ov, 2020-03-12:
SSH access works (tested with Bitvise SSH and Windows Power Shell) but I get no SFTP access. 
Error message is 

root@192.168.1.206's password:
sh: /usr/libexec/sftp-server: No such file or directory
Received message too long 2037998963

kedder, 2020-03-12:
I wonder what the advantages of `sftp` over `scp` for us?

mihu-ov, 2020-03-12:
I have no technical answer but think anything that has a graphical front end that looks like Windows Explorer will do ;-)
Probably some users use https://www.bitvise.com/ssh-client because it was mentioned in the setup instructions so it would be good (but not required) if this could still be used with the new image.

mihu-ov, 2020-03-12:
Thank you kedder!

PR #26 (2020-03-13): Include nano editor and sftp server
Fixes #24
Fixes #25
closed 2020-03-13 by linuxianer99