Openvario/meta-openvario #27: Default password for root

kedder, 2020-03-12:
Right now, as @mihu-ov pointed out, there is no password for root user in default image.

I expect many users will enable wifi on their openvarios. Having internet-facing device with unprotected root access is probably not the brightest of ideas (if we don't want openvarios to accidentally start mining bitcoins or something like that).

Should we set it to something hardcoded, like `openvario` for simple protection against bots and recommend users changing it immediately? Something more sophisticated?

linuxianer99, 2020-03-13:
Hmmm ... but the WLAN should just be connected for updates, right ?
I think WLAN should be disabled by default ...

mihu-ov, 2020-03-13:
WLAN is the first thing I enable when testing an image and with kedders new Competition Manager app (maybe extended for non competition syncing with Dropbox or similar in the future) it will be used by many pilots.

kedder, 2020-03-13:
Well, WLAN is not enabled until you connect your own USB dongle and configure it with `connmanctl`. So it is of course not the issue for plain out-of-the-box installations. But I suspect most of the users would not realize they become vulnerable as soon as they connect to the Internet, so we can just make sure they are not completely unprotected in that case.

lordfolken, 2022-01-28:
The problem here is that dropbear is started with -B
That allows to login with blank passwords. 

The proper operation should be:
you set a root password via alt-f2 (or exit to shell) to whatever you want. 
If the password is not set, no login is possible.

Please do not set a default password.

closed 2022-02-04 by linuxianer99