Openvario/meta-openvario #28: "re-run" bitbake failed

brunotl, 2020-03-13:
using linuxianer99/ovbuild docker container to build image, each time i try to re-run build i got a the same error : 
`ERROR: ExpansionError during parsing /workdir/poky/meta-openvario/recipes-apps/variod/variod-testing_git.bb`

full error Traceback -> [build-error.txt](https://github.com/Openvario/meta-openvario/files/4329236/build-error.txt)

linuxianer99, 2020-03-13:
How did you start the container ?

As mentioned in the guide ?
Can you please provide a "ip addr" output of the host system and from inside the container?
Seems there is a DNS problem with your docker ...

kedder, 2020-03-13:
Yeah, the last line of the seem to be the culprit:
```
fatal: Unable to look up github.com (port 9418) (Temporary failure in name resolution)
```

Failure is dure to `SRCPV` variable substitution. I found this on the internets:
>> Off the top of my head, I know SRCPV is a combination of the VCS type
>> and revision used in the actual build

So, it couldn't get revision from github -> could not generate value of `SRCPV`  -> could not make a substitutions in a recipe.

Check if github works for you now.

brunotl, 2020-03-13:
looking the error the first check i did is to ping 'github.com'  -> no problem

if i remove build directory i can build image a second time, but it's to long i can't do that all the time ...

### all about my work env :  

host version
```console
bruno@debian:~$ uname -a
Linux debian 4.19.0-8-amd64 #1 SMP Debian 4.19.98-1 (2020-01-26) x86_64 GNU/Linux
```
docker version
``` console
bruno@debian:~$ docker -v
Docker version 19.03.7, build 7141c199a2
```
host ip 
``` console
bruno@debian:~$ hostname -I
192.168.1.14 172.17.0.1
```
cmd line to run docker container 
``` console
bruno@debian:~/devel/openvario$ docker run -it --rm -v $(pwd):/workdir linuxianer99/ovbuild --workdir=/workdir
pokyuser@71752b54d4ff:/workdir$
``` 
container version
``` console
pokyuser@71752b54d4ff:/workdir$ uname -a
Linux 71752b54d4ff 4.19.0-8-amd64 #1 SMP Debian 4.19.98-1 (2020-01-26) x86_64 x86_64 x86_64 GNU/Linux
``` 
container ip 
``` console
pokyuser@71752b54d4ff:/workdir$ hostname -I
172.17.0.2
``` 
within container i'm able to ping github ( name resolution work ...)
``` console
pokyuser@71752b54d4ff:/workdir$ ping github.com
PING github.com (140.82.118.3) 56(84) bytes of data.
64 bytes from lb-140-82-118-3-ams.github.com (140.82.118.3): icmp_seq=1 ttl=49 time=62.9 ms
64 bytes from lb-140-82-118-3-ams.github.com (140.82.118.3): icmp_seq=2 ttl=49 time=66.2 ms
```

brunotl, 2020-03-13:
I've also checked the container git config, nothing wrong inside :

``` console
pokyuser@71752b54d4ff:/workdir/poky/build$ git config --list --show-origin
file:.git/config        core.repositoryformatversion=0
file:.git/config        core.filemode=true
file:.git/config        filter.lfs.smudge=git-lfs smudge --skip -- %f
file:.git/config        filter.lfs.process=git-lfs filter-process --skip
file:.git/config        remote.yocto.url=git://git.yoctoproject.org/poky
file:.git/config        remote.yocto.projectname=poky
file:.git/config        remote.yocto.fetch=+refs/heads/*:refs/remotes/yocto/*
```

brunotl, 2020-03-13:
same problem if i run build directly on the host ...

kedder, 2020-03-13:
So, let's get to the basics: can you do this in your container: 

```
git clone git://github.com/Openvario/variod.git
```

brunotl, 2020-03-13:
no surprise that work perfectly ...  otherwise, a clean build would failed too ...
 
```console
pokyuser@c63f156bba14:/workdir$ git clone git://github.com/Openvario/variod.git
Cloning into 'variod'...
remote: Enumerating objects: 210, done.
remote: Counting objects: 100% (210/210), done.
remote: Compressing objects: 100% (82/82), done.
remote: Total 210 (delta 127), reused 210 (delta 127), pack-reused 0
Receiving objects: 100% (210/210), 94.52 KiB | 203.00 KiB/s, done.
Resolving deltas: 100% (127/127), done.
pokyuser@c63f156bba14:/workdir$
```

kedder, 2020-03-13:
Okay. So you still getting this name resolution error every time you run `bitbake variod-testing`?

Here's few things to try:

```
$ bitbake variod-testing -c clean
$ bitbake variod-testing
```

if doesn't help, try changing git url in `SRC_URI` in `meta-openvario/recipes-apps/variod/variod-testing_git.bb` to use https protocol: `git://github.com/Openvario/variod.git;protocol=https`

kedder, 2020-03-15:
@brunotl did you figure it out?

brunotl, 2020-03-16:
need to be confirmed, but it seem's that happen only when it run inside ssd session....

brunotl, 2020-03-17:
confirmed, that happen only inside ssh session. 

i've found workaround :  runing inside tmux ...

closed 2020-03-17 by brunotl