Openvario/meta-openvario #29: No ttyUSBx devices created for USB to serial converters but many useless tty* devices existent

mihu-ov, 2020-03-13:
I have tested four different USB to serial converters with image 20067 and found some issues. 

1) The first USB2serial that is plugged in an USB hub is shown by lsusb command, a second one is not shown until the first one is removed. 
2) Dmesg shows that the devices are enumerated but no device drivers are loaded and no ttyUSBx devices are created (even for the first device listed by lsusb).
3) ls /dev/tty* (and hence the selection list in XCSoar) shows a lot of tty* devices but probably only ttyS0 to ttyS3 (the four serial ports) are real while the rest clutter up the selection lisk. I think we had the same issue in the old image until it was solved.

kedder, 2020-03-13:
@mihu-ov can you find out what drivers are needed for USB2Serial devices you have? You can check that by inserting these devices to some linux laptop (if you have one). The device type should be visible in syslog.

I suspect some drivers are missing in kernel. I also received complaints about very similar issue by email.

mihu-ov, 2020-03-13:
@kedder: I have tested FT232, CP2102, PL2303, HL-340

kedder, 2020-03-13:
Or: provide `lsusb` output on the device.

Anyway, I can repro this myself. At least my CH340G device is not recognized properly (due to missing kernel module). Let's see, maybe I can just enable all of them...

CazYokoyama, 2020-03-13:
I pointed the branch which enables PL2303. I am successfully feeding GPS
information through PL2303.
-caz

DanD222, 2020-03-14:
@mihu-ov Please also note that ttys4 can come in handy if you want to modify the script.bin file in order to be able to connect four hardware devices to OpenVario, see: https://forum.xcsoar.org/viewtopic.php?f=30&t=3437

PR #30 (2020-03-14): Include USB-serial kernel drivers
Includes all modules available for 5.2 kernel.

Fixes #29
closed 2020-03-14 by linuxianer99