Openvario/meta-openvario #31: Idea: file synchronization

kedder, 2020-03-13:
Everybody knows dropbox and it is great for file synchronization across devices. However it has several drawbacks:

* It is a commercial project, and will probably not allow us to redistribute its binaries
* These days they only allow 3 synchronized devices per account on a free plan
* Only allow to sync one folder: `~/Dropbox`

There's a better alternative though: [syncthing](https://syncthing.net/), which is a peer-to-peer file synchronization program.

* It is fully open-source
* No limitation on number of devices
* Has clients for android, linux, windows and macos
* Works without internet (local network is enough)
* It allows to synchronize any number of folders on your system

By a lucky coincidence, XCSoar runs on android phones too. If we share `~/.xcsoar` directory between openvario and the phone, this effectively would allow us to implement this workflow:

* While at home, you configure your xcsoar on your phone the way you like it - pick navboxes, screens, settings, etc; download maps, airspaces, waypoints.
* You arrive at the airfield, go to the briefing and enter today's task to your phone
* You turn on your Openvario, tether internet from your phone and all your configuration and current task gets transferred to Openvario (in a matter of seconds).
* You just launch xcsoar and fly today's task
* After you land, your flight log automatically gets synced back to your phone (and/or a folder in a club' computer, if you want), so you can just go home and analyze it, without dealing with usb sticks.

I'm not aware of such a feature for any of commercial flight computers, so we might even get some envy from their proud owners :)

And I think it should not be super difficult to get it on the openvario image. What do you think?

CazYokoyama, 2020-03-14:
That is a nice idea. At morning meeting, we heard today's task. I many
times enter the task into my Android phone and again enter to my glide
computer, Tophat. So
Who will implement it?
Which release of openvario will it be included?
-caz

Scumi, 2020-03-14:
What about NextCloud? I would prefer this one, since it is also open source and would therefor fit nicely for the openvario. I made some tries with that 2 years ago, but my initial approach failed probably due to the missing battery powered RTC in the openvario.

kedder, 2020-03-14:
> So Who will implement it? Which release of openvario will it be included?

I can make an attempt.There will obviously be some challenges to "productize" it (like not syncing and running xcsoar at the same time, because xcsoar would not pick up config/task changes while running). But idea of not having to enter tasks with knobs or remote stick sounds very seductive to me.

> What about NextCloud

TBH I haven't heard of it before. I briefly looked at it and it looks like a client-server app and can be self-hosted. [server](https://github.com/nextcloud/server) is written is PHP though, so it might be  difficult to run it on Openvario itself. Or we would need to set up and maintain some central infrastructure which is also difficult. Or we have to rely on some 3rd party provider for the infrastructure, which also  might not be too future proof. But again, I have very superficial knowledge about it, so I might miss something.

mihu-ov, 2020-03-14:
Sounds all good, would this work in a club environment with different users?

kedder, 2020-03-14:
> would this work in a club environment with different users?

Well, in club environment, where gliders are shared, the use case where openvario is synched with users' phones would probably not be very useful. In that case it would probably be better to set up a club computer to synchronize with all the openvarios on club gliders to automatically receive flight logs.

kedder, 2020-03-15:
Ok, I've made #33 that would create installable package for `syncthing` (but not include it into the image itself). It is installable with

```
opkg udpate
opkg install syncthing
```

Syncthing comes with web UI, available under `http://<openvarioip:8384`. We can experiment with it and find the best way to use it.

kedder, 2020-03-15:
It works darn well I must tell you. The only gotcha I came across so far is you don't want to run xcsoar on both synchronized devices simultaneously, because one of them will override the profile of the other eventually.

Otherwise I can enter task on my phone and it gets installed in xcsoar on openvario :)

Scumi, 2020-03-16:
> TBH I haven't heard of it before. I briefly looked at it and it looks like a client-server app and can be self-hosted. [server](https://github.com/nextcloud/server) is written is PHP though, so it [...]

For my part, I'm not familiar at all with Dropbox. However, if I understand it correctly, both work roughly the same. Both are a client/server application, meaning you need only the client on the Openvario for them to work. For Nextcloud, even only webdav (i.e. the linux application _dav2fs_) should work. 
I might have a look into your syncthing package, maybe it can be extended to work with Nextcloud as well.
Question: How does Dropbox handle the different timestamps? Due to the Openvario not having a battery powered RTC, files created never have the correct timestamp. This caused some issues with syncing applications for me (rsync, nextcloud, ...).

kedder, 2020-03-16:
AFAIU, as soon as you get internet up, the time gets synchronized with NTP. But while you are not, I suspect you'lld get the wrong time everywhere. How these sync systems behave in these conditions is to be determined. I just don't know if it will be sane or not.

Scumi, 2020-03-16:
To be tested, agreed. I did some research and syncthing is not working with nextcloud. However, nextcloud-client package offers a commandline interface. Will try to add a recipe for this, but I might need some help with that. Also I hope it doesn't install all graphic components as well, then its way bigger than required for our needs.

Scumi, 2020-03-19:
I am not able to compile the recipe. It needs QtKeychain and Qt 5.x.x, which are not available in the build environment as it seems. I don't know if it would be easy or not to add them.

kedder, 2020-03-19:
I think it won't be able to run on linux without graphical environment if it needs QT to compile

Scumi, 2020-03-19:
It contains a command line client (https://docs.nextcloud.com/desktop/2.6/advancedusage.html#nextcloud-command-line-client), so it should be able to run. It just seems the command line client cannot be compiled separately.

PR #33 (2020-03-20): Recipe for Syncthing
This adds a recipe for [syncthing](https://syncthing.net/) - file synchronization program. This does not include syncthing into the image, but adds a package to opkg repository so users can install it and play with it. 

It is just a stock installation, without any Openvario-specific features. We can experiment with it and figure out the best way to use it.

After installation, syncthing will start its UI on port 8384, it should be accessible under url `http://<openvarioip>:8384`

Fixes #31.
closed 2020-03-20 by linuxianer99