Openvario/meta-openvario #35: Drop support for XCSoar 7.0-preview12

kedder, 2020-03-16:
Since https://github.com/XCSoar/XCSoar/pull/363 with the fix by @roeles got merged into XCSoar mainline, there is no reason to ship XCSoar 7.0-preview12 anymore. Screen rotation should work on most recent XCSoar master. 

This basically means that `openvario-image` and `openvario-image-testing` becoming identical.

With that in mind, what do you think would be the most sensible way to deal with it:

1. Just drop `openvario-image-testing` and have one image variant to ship with the latest XCSoar from master
2. Ship latest stable 6.8x with  `openvario-image`  and bleeding edge with `openvario-image-testing`
3. Ship latest stable 6.8x with  `openvario-image` and drop `openvario-image-testing`, but allow to `opkg install` latest version of XCSoar for those who wish the latest features.
4. Any other option?

Thoughts? Opinions?

mihu-ov, 2020-03-16:
OV is making it into more and more cockpits and gliding clubs so in the long run I think it would be good to have a stable openvario-image for the users and bleeding edge openvario-image-testing for developers with newest syncthings, XCSoar mainline etc.

CazYokoyama, 2020-03-16:
No need to consider Skysight integration. Its PR is dropped.
-caz

kedder, 2020-03-16:
@mihu-ov I agree, though I'm leaning towards 3 - it should be easy enough to upgrade from stable image to testing/bleeding edge software with `opkg` and we would have to ship 4 images instead of 8.

kedder, 2020-03-16:
@CazYokoyama ah, that's too bad :(

linuxianer99, 2020-03-17:
there is a lot more difference from openvario-image to openvario-image-testing:

sensord and variod are also in testing version.

So i would prefer to stay with the stable and testing variant.
I am working on an automatic CI with github. 
Releasing 4 or 8 images is just a more build time ...

kedder, 2020-03-17:
Ok, let's settle on option 2 then.

mihu-ov, 2020-03-17:
Can we merge the different LCD variants into one image and select device tree, resolution etc. with a config file? Years ago we thought about having a permanent data partition with user data and settings and a system partition which is updated with a new image when available.

kedder, 2020-03-17:
So, right now, AFAIU, the difference between images is this:

* U-boot is compiled with screen-specific configuration to init screen early
* Set of resolution-specific splash-screen images for u-boot
* Device-tree file with screen-specific resolution and timings

All of it is stored on /boot partition. The rest is absolutely identical among all variants.

We could probably ship common and screen-specific parts separately, but I'm not sure how to make it in user-friendly way. User would need to partition their SD card manually for that, I guess.

Personally I'd prefer to upgrade using `opkg` via the internet.

PR #36 (2020-03-18): Switch openvario-image to XCSoar 6.8.12
Make `openvario-image` ship with latest stable XCSoar-6.8.12 and `openvario-image-testing` the most recent version from XCSoar master branch.

Also, make sure `xcsoar` package conflicts with `xcsoar-testing`, so that users would need to remove one in order to install the other.

Closes #35.
closed 2020-03-18 by linuxianer99