Openvario/meta-openvario #38: openvario-image fails to build

iglesiasmarianod, 2020-03-18:
I've built openvario-image-testing and the previous version of the openvario-image.
Now openvario-image-testing builds but the latest version of XCSoar fails to compile.
Error states:

```
g++: error: unrecognized command line option ‘-fmacro-prefix-map=/home/mariano/ov-image/poky/build/tmp/work/armv7vet2hf-neon-ovlinux-linux-gnueabi/xcsoar/6.8.12-r0=/usr/src/debug/xcsoar/6.8.12-r0’
build/host.mk:23: recipe for target 'output/host/tools/GenerateSineTables.o' failed
| make: *** [output/host/tools/GenerateSineTables.o] Error 1 

```
Using Debian Stretch 9.12.

kedder, 2020-03-19:
Hm. I cannot repro this. Does it still fails when you clean and rebuild xcsoar:

```
$ bitbake xcsoar -c clean
$ bitbake xcsoar
```

iglesiasmarianod, 2020-03-19:
No, keeps failing.
I've set up my buildsystem as Timo suggested in redmine.
Perhaps my mistake is there.

```
  git clone git://git.yoctoproject.org/poky
  cd poky
  git fetch --tags
  git checkout tags/yocto-2.7.1 -b my-ovlinux

  git clone -b warrior https://github.com/openembedded/meta-openembedded.git
  git clone -b warrior https://github.com/linux-sunxi/meta-sunxi.git
  git clone -b warrior https://github.com/Openvario/meta-openvario
  
  export TEMPLATECONF=meta-openvario/conf
  . oe-init-build-env

 bitbake openvario-image
```
Are you using the same versions?

kedder, 2020-03-19:
The new "official" build system is described in the README here: https://github.com/Openvario/meta-openvario/blob/warrior/README.md. But I don't think this is the issue.

Can you do this:

```
bitbake xcsoar -c devshell
```

It will open new terminal. In that terminal, show the output of this:
```
echo $CC
$CC --version
which $CC
```

iglesiasmarianod, 2020-03-19:
Output for echo$CC

```
arm-ovlinux-linux-gnueabi-gcc -march=armv7ve -mthumb -mfpu=neon -mfloat-abi=hard -fstack-protector-strong -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security --sysroot=/home/mariano/ov-image/poky/build/tmp/work/armv7vet2hf-neon-ovlinux-linux-gnueabi/xcsoar/6.8.12-r0/recipe-sysroot
```
Output for $CC --version

```
arm-ovlinux-linux-gnueabi-gcc (GCC) 8.3.0
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

```
output for which $CC
```
/home/mariano/ov-image/poky/build/tmp/work/armv7vet2hf-neon-ovlinux-linux-gnueabi/xcsoar/6.8.12-r0/recipe-sysroot-native/usr/bin/arm-ovlinux-linux-gnueabi/arm-ovlinux-linux-gnueabi-gcc

```

kedder, 2020-03-19:
Hm. Everything seem to be right. 

I suspect that xcsoar build uses wrong compiler somehow, but can't really check it myself. Unfortunately I accidentally updated poky warrior branch and bitbake now wants to rebuild the world. So I'll only have working environment tomorrow.

iglesiasmarianod, 2020-03-19:
There's no hurry @kedder.
I'll keep looking in case I can figure out what's wrong.

kedder, 2020-03-20:
@iglesiasmarianod TBH, I'm almost out of ideas. `-fmacro-prefix-map` is a GCC8 flag. However bitbake builds its own compiler (gcc-8.3). I suspect earlier gcc is used in your case but why and how is it even possible? I don't know. 

You can try opening devshell (`bitbake xcsoar -c devshell`) and building it manually with verbose mode:

```
make DEBUG=n DEBUG_GLIBCXX=n ENABLE_MESA_KMS=y GEOTIFF=n Q=
```

It will show you exactly what command (and what compiler) it is using to compile the source. Hopefully it will give an idea what is wrong. It should use compiler from `/home/mariano/ov-image/poky/build/tmp/`.

iglesiasmarianod, 2020-03-20:
@kedder I think my host compiler might have someting to do with it.
I was thinking about a fresh start with Debian 10.
Here's the log from the commands you sent me. It's GCC-6.

[log.txt](https://github.com/Openvario/meta-openvario/files/4362092/log.txt)

kedder, 2020-03-21:
Ok, this is useful. apparently `HOSTCXX` is set to your OS g++, not the compiled one. I guess this is wrong. 

BTW, you don't have to upgrade your OS in order to build the Openvario. You can use docker setup described in the Readme.md.

iglesiasmarianod, 2020-03-21:
Problem Solved.
Updated GCC and G++ to 8.4 in my host system and XCSoar compiled.
Thanks @kedder

closed 2020-03-21 by iglesiasmarianod