Openvario/meta-openvario #39: scale font size for InfoBoxes

bomilkar, 2020-03-23:
The font in the InfoBoxes, particularly the label and the comment are awfully small.
I've been looking around in the source but can't find anything.
The font scaling on the map window works as expected.
Does anybody have an idea??
[image]

BTW: a screenshot function would be so nice. ;-)

kedder, 2020-03-23:
I've found that "DPI" setting in XCSoar system configuration affects the size of these labels. It is on the same configuration page as font size setting, but only available in XCSoar-7.

mihu-ov, 2020-03-23:
On the old image I coped the framebuffer content to a file (over ssh) and converted that file to some standard graphics format to take screenshots. Maybe kedder can advise on how to take screenshots for the new image?

bomilkar, 2020-03-23:
The DPI setting has no effect on the font sizes in the InfoBoxes. It affects the map window, but that's not what I'm looking for.
Unfortunately.

kedder, 2020-03-23:
@bomilkar you're right, DPI setting doesn't affect infobox labels. So I'm not sure there's even a knob for that. It looks like the size of the font depends on the size of infobox - the more space infobox has the larger the font is. So if you have 4 infoboxes in a row, the text will be bigger than if you have 6 infoboxes.

WRT to screenshots - it is tricky as well - framebuffer is not used to draw xcsoar, so you can't grab pixels from there. So, no advice at the moment, I'll keep looking.

bomilkar, 2020-03-24:
Yes @kedder you are right the font inside the InfoBox scales with the available space. I suspect the function AutoSizeFont in Look/AutoFont.cpp being used in Look/InfoBoxLook.cpp does exactly this for several fonts being used. But I can't figure out, where it gets the fonts from. Can you please look into this? You've been dealing with the graphics stuff. Right?
Maybe we can pick a different font type which is easier to read, like on other platforms.

DanD222, 2020-04-05:
@bomilkar I’m probably not the right person to answer this, but I see some differences in the master branch of _src/Look/InfoBoxLook.cpp_, compared to the v6.8 branch (see _void
InfoBoxLook::ReinitialiseLayout_)

https://github.com/XCSoar/XCSoar/blob/master/src/Look/InfoBoxLook.cpp

https://github.com/XCSoar/XCSoar/blob/v6.8.x/src/Look/InfoBoxLook.cpp

Could be worth investigating?

bomilkar, 2020-04-06:
Thanks @DanD222 , I'll look into this.

mihu-ov, 2020-04-06:
This is not OV specific and was discussed many times in XCSoar forum, e.g.
https://forum.xcsoar.org/viewtopic.php?f=7&t=2140
https://forum.xcsoar.org/viewtopic.php?f=19&t=2899
https://forum.xcsoar.org/viewtopic.php?f=7&t=3206

DanD222, 2020-04-06:
Reading the forum, it seems that the problem was with the font size selection system, yet it also seems that users are unhappy with the infobox title font size being too small in v7. 

How about increasing the infobox title font size in v7 so that is in line with what it currently is in v6.8, without having a font size selection system? 

It might also be worthwhile to test whether the 15 character width affects the title font size, and if so, think about whether we really need to fit 15 characters in the infobox title, or whether we could live with say 10.

For reference, here's a side-by side comparison between v7 and v6.8.

[image]

bomilkar, 2020-04-06:
The effect is less annoying in landscape orientation. This has to do with the aspect ratio of the InfoBoxes. I tried to change that (https://github.com/XCSoar/XCSoar/pull/327) but it didn't fly so I pulled back. I introduced the 10-InfoBox layouts (PR 324) which alleviates the issue by a bit because it produces wider info boxes.
The culprit is probably in InfoBoxLook::ReinitialiseLayout(unsigned width). There are some subtle differences between 6.8 and 7.0. I'll try to find the commits for these changes to understand the rational behind it.
One question in my mind: why do we use the same font size for titles like "Kreisdurchmesser" and titles like "Wind". Leave alone the fact that "Kreisdurchmesser" is a bad choice if it has to fit into limited space.
In any case: looking at the many forum posts (thanks Michael for pointing this out) on this topic, it is clearly a sour spot which should be addressed.

kedder, 2020-04-06:
This seems to be the reason why we have it this way: https://forum.xcsoar.org/viewtopic.php?f=7&t=2140&start=15#p11450

We probably can increase the size for for openvario specifically (with the custom patch), but then I guess we'll start getting bug reports that some text doesn't fit...

mihu-ov, 2020-04-07:
Frankly, I think this discussion is XCSoar specific, was started in XCSoar forum and on XCSoar github a few times and should be continued there. 
As Ronald wrote, this is an issue for many users of XCSoar 7.0 previews and should be addressed. But I really believe addressing this in XCSoar (not in Openvario) is the only solution in the long run or we will end up with two different XCSoar flavors which would be a bad idea in my opinion. 
And XCSoar 6.9.x is still the official release and has the old font scheme.

bomilkar, 2020-04-07:
I agree with Michael: XCSoar is the place to discuss the issue and ultimately fix it.
However, I was trying to build a lobby to put more weight behind this issue. My assumption is, that 7.0 is mostly used by the (small) openvario user community. The majority flies with 6.8 and for them it's a mute point. I'm sure the issue will escalate as soon as 7.0 becomes the main release. We can wait until then or we can make an effort now. But I'm not going to make the effort all by myself.

kedder, 2020-04-07:
It looks like there is no objections from anyone for having an option in XCSoar to scale infobox label size. And it doesn't look like a rocket science to multiply a value, calculated by `AutoSizeFont`, by some coefficient, read from xcsoar profile. 

It just takes a volunteer to actually do this.

bomilkar, 2020-04-08:
I can pick it up, unless somebody feels strongly about it. Please note: I've failed with something similar: https://github.com/XCSoar/XCSoar/pull/327 . So I recommend you read Max' comments in this PR before we dive in here. And I really request your support on it since I don't want to fail (again) alone.

A fully automated solution (without yet another parameter in the setup) would be best. I guess it would have a good chance to pass the gatekeeper. The second best solution would be to introduce a setup parameter which is "intuitive" to use. (I guess my previous attempt failed on this.)

Here is part of the problem: even long Waypoint names should fit into the IB. Analyzing the waypoints from OpenAIP for Cerntal Europe I've seen this:
The list has 3099 waypoints. 963 names are longer than 15 char. 83 names longer than 25. One as long as 39 char. (The smaller the village, the longer the name.)

The "fully automated" solution could prepare a couple of fonts: normal with 10 char, long with 15 char, long long with 25 char. The IB would pick up the appropriate font depending on the length of the text at hand.

One alternative to kedder's "intuitive" setup parameter could be to let the user pick the length of the field: normal (10), long (15), long long (25). Then AutoSizeFont would use the appropriate string.

A simple scale factor for "InfoBox Title and Comment size" like "Text size" might be rejected, I guess. Even though implementing this would be a piece of cake.

Any thoughts?

bomilkar, 2020-04-08:
Here is my first shot at it.
It scales title and comment depending on the length of the text to be printed. It has 2 options: a small font which fits 15 chars and a "normal" font which fits 10 chars.
You can see the effect side by side. On the left is the scaled font, on the right is the original 7.0 font size.
[image]

bomilkar, 2020-04-08:
Here is another screenshot with 480x800 format, which is what at least the 7" Texim display uses.
[image]

kedder, 2020-04-08:
Can I offer you some critique? :) (not meant to discourage you in any way)

From the pure aesthetic point of view, I'd prefer the right version. Because of inconsistent font sizes the left one looks less tidy to me.

The readability benefits are also not so apparent: most of labels that we can make bigger and more readable are static, meaning that you don't really read them all the time. You learn that top right is your AGL height, same way you learn where the altimeter is on your instrument panel. OTOH, the labels that you might want to read on the top left stay small (yes, I see counter examples too).

In any case, I think we should file an issue on XCSoar, stating the problem and discuss possible solutions there. This will not likely to be easy (because everyone has opinions on how things should look like) but at least this way it will more likely find some kind of resolution.

bomilkar, 2020-04-08:
No worries, I appreciate the discussion and critique.
I agree: it is not pretty, but what other option do you see. This approach uses bigger font where possible. Maybe the difference between small and normal font is too big?
And if you look closely: the values use different font sizes already.
Keep in mind: the more info boxes you have the less readable is the small font. For me 8 info boxes is not enough. 12 is too many and above all unreadable. That's why I introduced the 10 info box layouts.

I don't think filing an issue on XCSoar without a clear vision of how to address it will not get us anywhere. So what do you envision?

kedder, 2020-04-08:
I'd personally would probably prefer a setting in a profile to scale label font size 100%-150% and would probably pick 110 or 105% myself.

BTW, I like 10 box layout and use it myself in my OV (because 8 is too little and 12 is too many).

bomilkar, 2020-04-09:
Yet another approach to improve readability of IB title and comment: Make the IBs wider. And in order to have enough IBs create a layout with 3 rows of 4 IBs. That would be beneficial for tall slim screens, like 7" Texim in portrait orientation. The font would increase by 5/4 (125%) over a 2 rows by 5 layout. And on a tall screen (480x800) a third row might be an acceptable trade off.
Thoughts?

kedder, 2020-04-09:
> create a layout with 3 rows of 4 IBs

I agree, this might be another possible solution.

bomilkar, 2020-04-12:
I've just created 2 PRs in XCSoar:

https://github.com/XCSoar/XCSoar/pull/375
12 box layout 4 boxes by 3 rows

https://github.com/XCSoar/XCSoar/pull/376
scalable InfoBox title and comment

I think this is the most flexible solution for this issue.
The "adapting InfoBox title font" has the advantage of not needing a setup parameter, but it's not as flexible. That's why I ruled it out.

bomilkar, 2020-04-12:
[image]
This is the 3x4 layout with 100% InfoBox title size.

DanD222, 2021-03-26:
This has now been addresded in https://github.com/XCSoar/XCSoar/issues/552 via d3ea258340d49b9cdf3be2bce34229321401cd61

I for one am happy with the solution.

(v7 has also become the main release and v6.8 is being abandoned)

kedder, 2021-03-26:
Awesome! Preparing a PR to switch openvario-stable to 7.0.

PR #127 (2021-04-06): Upgrade stable xcsoar to 7.2
XCSoar 7.0 was just released. Upgrade stable image to 7.0.

Fixes #39 
Fixes #120
closed 2021-04-06 by linuxianer99