Openvario/meta-openvario #41: polar string being sent twice per second

bomilkar, 2020-03-24:
Now that I have a GPS source connected to the test H/W I notice that polar strings are being sent out twice per second. This is how it looks like when I tap the TX line:
`<15:28:01> $POV,C,RPO,0.002916,-0.176400,3.422314*51
<15:28:01> $POV,C,IPO,0.003168,-0.176400,3.150000*4C
<15:28:01> $POV,C,RPO,0.002916,-0.176400,3.422314*51
<15:28:01> $POV,C,IPO,0.003168,-0.176400,3.150000*4C
<15:28:02> $POV,C,RPO,0.002916,-0.176400,3.422314*51
<15:28:02> $POV,C,IPO,0.003168,-0.176400,3.150000*4C
<15:28:02> $POV,C,RPO,0.002916,-0.176400,3.422314*51
<15:28:02> $POV,C,IPO,0.003168,-0.176400,3.150000*4C
<15:28:03> $POV,C,RPO,0.002916,-0.176400,3.422314*51
<15:28:03> $POV,C,IPO,0.003168,-0.176400,3.150000*4C
<15:28:03> $POV,C,RPO,0.002916,-0.176400,3.422314*51
<15:28:03> $POV,C,IPO,0.003168,-0.176400,3.150000*4C
<15:28:04> $POV,C,RPO,0.002916,-0.176400,3.422314*51
<15:28:04> $POV,C,IPO,0.003168,-0.176400,3.150000*4C
<15:28:04> $POV,C,RPO,0.002916,-0.176400,3.422314*51
<15:28:04> $POV,C,IPO,0.003168,-0.176400,3.150000*4C
<15:28:05> $POV,C,RPO,0.002916,-0.176400,3.422314*51
<15:28:05> $POV,C,IPO,0.003168,-0.176400,3.150000*4C             
`
The timestamp at the beginning of each line is generated by the terminal tool.
What you see are polar strings for Real (RPO) and Ideal (IPO) polar.

Now when looking at the code which generates this:
Device/Driver/OpenVario.cpp line 53 and on it becomes obvious why this happens.
The function is called OpenVarioDevice::OnCalculatedUpdate
And it seems as if really does what it says it does: being called OnCalculatedUpdate.
So every time a GPS fix is received the position slightly changes. And every change in position seem to trigger a call to OnCalculatedUpdate .
But obviously the polar doesn't change every time the geo position changes.

I'd like to fix this and that's why I'm writing this message. I see 2 ways to fix it and I'd like to get your thoughts on it:
1. quick and dirty: remember the last polar and send the string only if one of the coefficients has changed
2. register a function which is more appropriate for the purpose

I can do 1 easily. I'd prefer to do 2 but I don't know how.
Any idea?
Any thoughts??

kedder, 2020-03-24:
Can you give an idea what this polar NMEA string is useful for? Why is it being sent at all?

linuxianer99, 2020-03-24:
The polar is used for calculation in the variod.
@mihu-ov : What exactly is done in variod ??

What exactly is the problem if the NMEA string is send twice a second ?

bomilkar, 2020-03-24:
@kedder  The connected vario needs it to calculate speed to fly.
The (real) polar changes with the ballast, the bugs, an the glider. So it does change, but rarely.

@linuxianer99 it creates unnecessary traffic between processes or over the serial port.

linuxianer99, 2020-03-24:
so, the polar should be sent every time the bugs as well as the ballast changes.
Also it should be send out when the system boots, so variod has a change to get the polar data.

As far as i know, there is no other way than this "OnCalculatedUpdate" function. (Thats the reason i implemented it in that way)
Maybe it can be limited be sent out every 10th update or so, but than there is a delay in polar update for variod.

Of course it is not necessary to send it twice a second, but is there a negative impact on the whole system by doing so ??

kedder, 2020-03-24:
Hm, probably the more robust and economical approach would be for variod to request the polar via some special NMEA message when it needs it, and XCSoar to send it when it changes (e.g. when bugs / ballast / MC is changed).

Not sure how involving this change would be though.

linuxianer99, 2020-03-24:
You are right, there is of course room for improvement ...
But i think we should focus on the core problems (screen cal, vario sound, etc.) for the moment, shouldn't we ?

We can file this as improvement ...

kedder, 2020-03-24:
@linuxianer99 I agree

linuxianer99, 2020-03-24:
👍

bomilkar, 2020-03-24:
I suggest I fix it in XCSoar. And since nobody here seem to know a better trigger for the update, I do it the quick and dirty way:
inside OnCalculatedUpdate I check if the polar coefficients have changed. If so it generates the output.
Maybe Max knows how else this should be done. Otherwise he'll merges it as is.
Agreed?

kedder, 2020-03-24:
@bomilkar then the risk is that variod will never get the first message (e.g. if it gets restarted while in flight).  In general that approach will only work if variod is listening when this initial polar string is sent.

linuxianer99, 2020-03-24:
@bomilkar : I also suggest not to change it at the moment. 
There is no known error at the moment. I see it as an improvement, but not as an urgent bug.

bomilkar, 2020-03-24:
@kedder you are right: there is an issue if variod isn't listening while XCSoar starts.
As you suggested earlier it would need a query command which variod sends when it starts (or otherwise needs a refresh).
I'll put the issue aside.

mihu-ov, 2020-03-24:
@bomilkar Can you please check how this is done in other drivers. The Vaulter driver should be state of the art and probably syncs any changes in bugs, water etc. in XCSoar to the external vario. Let´s not re-invent the wheel and do it the same way?

bomilkar, 2020-03-25:
@mihu-ov will do. Good idea!
We call it design re-use. ;-)

bomilkar, 2020-03-28:
Who knows the ins and outs of variod ? I'd like to discuss possible solutions to the issue at hand and I want to make sure variod is in the loop.

kedder, 2020-03-28:
AFAIK the author of variod is Andreas Wasserbauer, he can be contacted via the redmine (apparently he doesn't have github account).

He is working on porting variod to pulseaudio to fix the audio output, but I haven't heard from him for the past 2 weeks: https://www.openvario.org/redmine/boards/3/topics/1773

bomilkar, 2020-04-06:
To address this enhancement I have posted a Pull Request one week ago. Nothing seem to happen:
https://github.com/XCSoar/XCSoar/pull/370

I see a few ways forward:
1. I keep waiting
2. I close the XCSoar PR and this Issue here, delete the branch and keep it as my private add on.
3. you guys chime in on the discussion on the XCSoar PR, maybe Max is waiting for your OK.

In essence, the most important new feature is this:
if variod sends a sentence
$POV,?,RPO,IPO,MC,BU,WL*xx
XCsoar will respond with this:
$POV,C,WL,1.367*10
$POV,C,BU,0.94*22
$POV,C,MC,0.80*3E
$POV,C,IPO,8.177143e-04,-9.728571e-03,2.357143e-01*00
$POV,C,RPO,8.097897e-04,-1.089426e-02,2.984780e-01*1F

XCSoar's response reflects the most recent values of WL,BU,MC,IPO,RPO. 
variod should send this NMEA-sentence upon (re)start as soon as it's ready to receive the requested "$POV,C," sentences. variod is free to send this request at any time, or with a sub-set of the requested settings.

kedder, 2020-04-06:
@bomilkar I just posted a code review. It looks good to me in general. 

The frequency in which Max returns to reviewing PRs is unknown to me, so we just have to wait. I waited for 8 months for mine to be merged, but I guess it is not indicative :). I think we can get it in faster :)

bomilkar, 2020-04-13:
I re-implemented the changes in a single commit in this PR:
https://github.com/XCSoar/XCSoar/pull/378
Hoping this will be approved and merged soon.

kedder, 2020-04-13:
@bomilkar, cool! Usually you just force-push to old branch instead of starting new PR though. Otherwise you lose all the comments and reviews of old PR.  That's fine though, just keep it the way it is now.

condoratberlin, 2020-05-26:
Is this issue still open? I think the pull request is closed on xcsoar.

kedder, 2020-05-26:
Yes, variod needs to be updated to request the data from xcsoar

bomilkar, 2020-05-26:
The XCSoar side is done. I can confirm this.
As far as I know the current version of variod doesn't even understand POV,C,ROP sentences when OpenVario driver in XCSoar sends it.