Openvario/meta-openvario #51: STF and Vario mode change not reliable.

iglesiasmarianod, 2020-04-20:
Hi guys, I've compiled the latest versions of Warrior and tested them in my OV.
Both images compiled.
OV image still has the sound issue in my case.
Don't know what may I be doing wrong. Perhaps it's version of Variod is not updated to pulseaudio?
OV testing sound is working OK with a few quirks I've noticed.
Changing between STF and Vario mode seems rather random.  You need to press several times the S or the V key to make it change.  The STF mode and Vario mode labels show timely on screen.
It looks like a communication issue between Variod and XCSoar. 
I have sensors in my OV so I can help Mihu with the testing. If some kind of logging is helpful you only need to ask. I have a 5.7 texim unit built.

Thanks a lot for all the hard work and kind regards.

kedder, 2020-04-20:
`openvario-image` was not updated to use pulseaudio yet, because I considered pulseaudio as something untested. But since soundsimply doesn't work under "stable" variod, I guess we should just enable pulseaudio on stable image too.

WRT STF/Vario switch, I also experienced some quirks, namely sometimes sound just disappeared when I pressed S/V repeatedly several times. But now I'm thinking maybe I just switched to STF mode that way? I don't have sensors though, so it is hard to tell. I'm not sure how STF mode is supposed to sound like without sensor input.

Something is off there for sure and needs debugging. Ideally we should implement some kind of test bench for variod to simulate both sensor input and xcsoar NMEA input so that we can actually run it locally on our laptops and test. But probably a bit more logging would help revealing the problem in short term too.

iglesiasmarianod, 2020-04-20:
Hi @kedder !
OK with variod in ov-image. Just thought I was doing something wrong.
If you don't have sensors connected (or uncalibrated board) as IAS is not there you get a high pitched fast pulsed beeping sound just like you were circling a 10 m/s thermal.
The same happens to the vario sound, you get 99 m/s climb so beeping is fast and high pitched.
Sometimes the IAS calibration gets lost from sensorboard loosing IAS data. You get a working vario sound but high pitched STF sound. This issue is easily fixed re calibrating sensors.
I'll log the communication between XCSoar and Variod with the device debug button inside XCSoar.
Can you think of another log that might be useful?

kedder, 2020-04-20:
We need to ensure two things:

1. NMEA sentence for STF switch is actually being sent by XCSoar
2. It is received and properly parsed by variod

We can probably use netcat (`nc`) to mock xcsoar and monitor communications.

iglesiasmarianod, 2020-04-21:
I logged the communication but can't see commands being sent.
They are sent directly by .xci file and monitor only reproduces incoming messages.
Don't know if this will be a problem when event files support ends to be replaced by LUA scripting.
I found a communication error in the xcsoar log file.

[xcsoar.log](https://github.com/Openvario/meta-openvario/files/4510697/xcsoar.log)

> [2020-02-11T05:03:43Z] Error on device TCP port 4352: End of file

It repeats constantly until shutdown.

kedder, 2020-04-21:
I have also seen these constant re-connections. This might be the root of the problem.

iglesiasmarianod, 2020-04-22:
Found variod.log. It shows the same disconnections and reconnections as XCSoar log.
I suspect variod is ending the connection somehow. I started looking inside variod code.
Here's the log I extracted from my device. Variod debug level is 1. 

[variod.log](https://github.com/Openvario/meta-openvario/files/4518992/variod.log)

iglesiasmarianod, 2020-04-23:
Think something fishy is going on with this message send.
Audio does not stop even with XCSoar offline and this "break" terminates the while loop leading to the connection dropped message. 
I'll try to log the send result to see if this is the the reason for disconnection. 

```
if (send(xcsoar_sock, client_message, strlen(client_message), 0) < 0)
			{	
				if (errno==EPIPE){
					fprintf(stderr,"XCSoar went offline, waiting\n");

					//reset socket
					close(xcsoar_sock);
					vario_mute();
					xcsoar_sock = socket(AF_INET, SOCK_STREAM, 0);
					if (xcsoar_sock == -1)
						fprintf(stderr, "could not create socket\n");
					
					wait_for_XCSoar(xcsoar_sock,(struct sockaddr*)&s_xcsoar);
					
					// make socket to XCsoar non-blocking
					fcntl(xcsoar_sock, F_SETFL, O_NONBLOCK);
					break;

				} else {
					fprintf(stderr, "send failed\n");
				}
```

bomilkar, 2020-04-24:
Did anybody find the issue behind the continuous connection drop?

```
Waiting for incoming connections...
Connection accepted
Connection dropped
Waiting for incoming connections...
Connection accepted
Connection dropped
Waiting for incoming connections...
```
I don't have any sensors in my test setup. Maybe this is the reason why it looses connection?

kedder, 2020-04-24:
If I'm not mistaking @iglesiasmarianod is seeing the same behavior with sensors.

bomilkar, 2020-04-24:
Instead of variod I use nc:

```
# nc 127.0.0.1 4352
$POV,C,WL,nan*5C
$POV,C,BU,0.99*2F
$POV,C,BU,0.98*2E
$POV,C,BU,0.97*21
$POV,C,RPO,0.002128,-0.101447,1.945879*59
$POV,C,IPO,0.000818,-0.009729,0.235714*41
$POV,C,RPO,0.000968,-0.011144,0.261150*5F
$POV,C,WL,1.000*12
$POV,C,WL,1.014*17
$POV,C,WL,1.027*17
$POV,C,WL,1.041*17
$POV,C,WL,1.054*13
$POV,C,BU,0.96*20
$POV,C,BU,0.95*23
$POV,C,RPO,0.000963,-0.011378,0.273798*58
```
This is what XCSoar says when I change wing load, bugs, etc. That looks very reasonable. Exactly what I expected.

Now I want to see how the other direction works:
Stop XCSoar and start "nc -l 127.0.0.1 4352" instead.
Unfortunately the busybox version of nc doesn't support "-l" option.
Hmmmm.

However, there is something interesting:
I start variod and it says:
`Waiting for incoming connections...`

As soon as I start "nc 127.0.0.1 4352", which of course exits immediately, but variod enters this continuous loop:

```
failed to connect, trying again
failed to connect, trying again
failed to connect, trying again

```

kedder, 2020-04-24:
I have a suspecion that it is xcsoar that is responsible for these disconnections. I see the same behavior if I connect xcsoar directly to sensord (TCP port 4353 instead of 4352).

iglesiasmarianod, 2020-04-24:
Saw the same behaviuor with sensord.
If it where XCSoar, going back to preview 10 would solve the problem. ¿Don't you think?

kedder, 2020-04-24:
Hm, you mean preview 10 doesn't have this issue?  Going back that far would probably not be an option - there are bunch of features in newer xcsoar that people would like to have. 

It should be possible to narrow it down and fix it in xcsoar (if that is really xcsoar at fault).

iglesiasmarianod, 2020-04-24:
Sorry if I expressed myself the wrong way.
On the old build systen OV7 preview 10 was used. Sensord and Variod were the same as before pulseaudio and everything worked as a charm.
Just to confirm or rule out XCSoar as the source of the problem I was thinking on trying my old preview 10 .ipk. If it worked, the problem was a change in xcsoar. If it did not work, the problem is either variod/sensord or the operating system.
I agree going back to preview 10 is out of the question. Just thought about a quick test to prove the theory.

kedder, 2020-04-24:
@iglesiasmarianod, ah that makes sense. Another thing to try would be the stable xcsoar (6.8.13) from the `openvario-image`  image.

bomilkar, 2020-04-24:
Port 4353 is used between sensord and variod, sensord being Client.
Port 4352 is used between XCSoar and variod, XCSoar being Server.

As you can see on my previous comment the communication on 4352 looks OK.

I wrote a dummy for sensord to supply some meaningful data on port 4353 and everything suddenly seem to work. I'm working on a debian 10 environment which makes it easier to debug. It may not produce the same effect though. But the continuous loop `failed to connect, trying again` is the same here if there is no meaningful data on port 4353.

Unfortunately I don't have any sensors on my OV test environment. I'm starting to think that sensord is causing the trouble. Maybe I will create a sensord dummy for the OV system to see what happens. .... Tomorrow.

iglesiasmarianod, 2020-04-24:
Just tested the stable release in my OV (6.8.13).
Same disconnection/connection sequence in Variod.log
As a curious thing, this reconnects make the debug button in the devices screen flicker from grey to black.
In the old versions sound muted when XCSoar ended.
The fact that sound does not mute now leads me to believe the problem is in the Variod side or the OS side given the fact that XCSoar is not present anymore and the behaviuor is still wrong.
Even before pulseuadio implementation sound kept coming out of the speaker when xcsoar wasn't running.
If variod is unable to send data to xcsoar, it should mute, reset socket, and wait for xcsoar to get a message. If the error message from the send command is not correctly interpreted by Variod I think this kind of odd things might happen.
Execution never reaches this part of the code.

```
if (send(xcsoar_sock, client_message, strlen(client_message), 0) < 0)
			{	
				if (errno==EPIPE){
					fprintf(stderr,"XCSoar went offline, waiting\n");

					//reset socket
					close(xcsoar_sock);
					vario_mute();
					xcsoar_sock = socket(AF_INET, SOCK_STREAM, 0);
					if (xcsoar_sock == -1)
						fprintf(stderr, "could not create socket\n");
					
					wait_for_XCSoar(xcsoar_sock,(struct sockaddr*)&s_xcsoar);
					
					// make socket to XCsoar non-blocking
					fcntl(xcsoar_sock, F_SETFL, O_NONBLOCK);
					break;
```

I'll try to add a few fprintfs to log these messages. The two fist ifs are the ones that catch my eye.

DanD222, 2020-04-25:
Not sure if this helps in any way, but I found this post on the forum: https://forum.xcsoar.org/viewtopic.php?f=30&t=3642&start=45#p20860

_"That´s what variod is supposed to do. It starts beeping when XCSoar starts but keeps beeping even if you quit XCSoar. In case of a XCSoar crash you would still have audio vario, keep climbing and restart later."_

bomilkar, 2020-04-25:
I have it running on my test bench on Debian 10:
Latest XCSoar with OpenVario driver on TCP port 4352
variod with some fixes (wip I'll commit and push later today)
sensord mock up (attached). Just a few lines to produce POV sentences on the socket.
variod must run before sensord_mock

[sensord_mock.c.txt](https://github.com/Openvario/meta-openvario/files/4532482/sensord_mock.c.txt)
gcc -o sensord_mock sensord_mock.c

I will compile sensord_mock.c for my OV test setup and run it there with latest variod and XCSoar.

bomilkar, 2020-04-25:
Created https://github.com/Openvario/variod/pull/4
This is probably not the final version, but at least I didn't get Segmentation Faults any more. :-)
And it is compliant with what it gets from XCSoar (and the documentation).

iglesiasmarianod, 2020-04-26:
Tested XCsoar preview 12 with variod-testing in warrior.
This preview works OK in Armstrong and is the one I'm currently using in my OV.
Same disconnections occur in Warrior with this version.

iglesiasmarianod, 2020-04-28:
Merged fix in Variod

closed 2020-04-28 by iglesiasmarianod