Openvario/meta-openvario #53: variod not doing what the documentatio says

bomilkar, 2020-04-23:
The documentation
https://www.openvario.org/doku.php?id=projects:series_00:software:nmea
says POV,C,RPO and POV,C,IPO are used to send polar information from XCSoar to variod.
Looking at the C-code, nmea_parser.c in particular it is looking for POV,C,POL
This is inconsistent with the documentation and - even worse - XCSoar never sent a POV,C,POL sentence. XCSoar always sent POV,C,RPO and POV,C,IPO sentence, which were never understood by variod.

How do we resolve it?

BTW: I'm implementing the counterpart to https://github.com/XCSoar/XCSoar/pull/378
In particular the to get the relevant current settings of XCSoar (MacCready, wing load,
polar, polar degradation). variod can do this at any time, especially upon restart to sync up with XCSoar’s settings.

bomilkar, 2020-04-24:
After having thought about it for a while I'll go ahead and prepare some changes for a PR:
1. make variod's protocol consistent wit XCSoar's
2. use values which XCSoar can readily supply
3. keep the POV,C,POL (as much as possible) as undocumented feature to keep backward compatibility

kedder, 2020-04-24:
@bomilkar I agree, we should change the variod to support `IPO` and/or `RPO`, whatever is relevant to it.

linuxianer99, 2020-05-07:
@bomilkar : Should we extend the POV,C command with the "?" command ?
Like POV,C,?,BU,MC,WL ??

Think this makes more sense that open another command structure beside Value pair and commands ??

bomilkar, 2020-05-07:
@linuxianer99 : all POV,C, sentences go from XCSoar to variod. POV,C,?, would be the only exception to go the opposite direction. I think this might confuse thing more than it helps.

linuxianer99, 2020-05-07:
OK. That's also a valid argument.
@mihu-ov : What's your opinion ? see above

hpmax, 2021-02-19:
How was this resolved (or was it?)  I can update variod if needed.

bomilkar, 2021-02-19:
I implemented the changes on the XCSoar side: https://github.com/XCSoar/XCSoar/pull/378
I guess nothing happened on the variod side. A quick look in the code confirms: the memory leak is still in there today. Probably the other weak stuff, too.
After all I gave up pushing the rope....

hpmax, 2021-02-19:
@bomilkar Could you clarify about the memory leak, if it's in variod or sensord I'll fix it if you can point it out to me.

bomilkar, 2021-02-19:
Count the number of malloc() vs. the number of free() in nmea_parser.c .
Then see what happens it it receives the "beginning of line" sign "$". Nothing ... How can it recover from getting out of sync with the sender? By luck, not by design.
If I recall right it can happen, that it receives multiple sentences from sensord even to the extent that the last sentence it cut off at the end of the buffer and continued in the next round of receive. It only deals with the first sentence in the buffer.
Etc...

hpmax, 2021-02-19:
@bomilkar That is bizarre.  I don't even understand why they are doing any mallocs.  I'll fix it up this weekend.

bomilkar, 2021-02-20:
@hpmax indeed bizarre. No need to malloc() at all here.

On the XCSoar side is implemented what is documented in here: 
https://www.openvario.org/doku.php?id=projects:series_00:software:nmea

In addition, I implemented an NMEA sentence to XCSoar which triggers XCSoar to send the requested information. Documented here:
[Extension of NMEA Protocol.pdf](https://github.com/Openvario/meta-openvario/files/6014496/Extension.of.NMEA.Protocol.pdf)
Unfortunately the OV-documentation was never updated.

Good luck!