Openvario/meta-openvario #55: Packages file is missing for openvario_57_lvds opkg feed

kedder, 2020-04-24:
When I do `opkg update`, I get error:

```
root@openvario-57-lvds:~# opkg update
Downloading http://ftp.openvario.org/feed-warrior/all/Packages.gz.
Updated source 'remote-all'.
Downloading http://ftp.openvario.org/feed-warrior/armv7vet2hf-neon/Packages.gz.
Updated source 'remote-armv7vet2hf-neon'.
Downloading http://ftp.openvario.org/feed-warrior/openvario_57_lvds/Packages.gz.
Collected errors:
 * opkg_download_backend: Failed to download http://ftp.openvario.org/feed-warrior/openvario_57_lvds/Packages.gz, wget returned 8.
```

This happens because `Packages.gz` file is missing from http://ftp.openvario.org/feed-warrior/openvario_57_lvds.

Please provide that file.

linuxianer99, 2020-04-25:
upload done ..
@kedder : please check again

kedder, 2020-04-25:
Same problem as in #56, no `Packages` package index in  http://ftp.openvario.org/feed-warrior/openvario_57_lvds. Do this before upload:

```
MACHINE=openvario-57-lvds bitbake package-index
```

kedder, 2020-04-25:
It works now, closing, thanks!

closed 2020-04-25 by kedder