Openvario/meta-openvario #56: No opkg feed for openvario_7_CH070 variant

kedder, 2020-04-24:
opkg feed for `openvario_7_CH070` variant is missing https://ftp.openvario.org/feed-warrior/. This will result in errors when trying to use opkg on this openvario flavor. 

Please upload respective packages to the server.

linuxianer99, 2020-04-25:
upload done ..
@kedder : please check again

kedder, 2020-04-25:
@linuxianer99 the dir is on ftp in the right location, but it is missing `Packages` and `Packages.gz` files, so `opkg` cannot use them. These files can be generated with bitbake:

```sh
MACHINE=openvario-7-CH070 bitbake package-index
```

Then the `Packages` files will be generated in `tmp/deploy/ipk/openvario_7_CH070/`

linuxianer99, 2020-04-25:
sorry. i did the index only for PQ070 ...
give me a few minutes ...

linuxianer99, 2020-04-25:
done.
Please try again ..

kedder, 2020-04-25:
Awesome! It works now!

closed 2020-04-25 by kedder