Openvario/meta-openvario #57: Vario not un/muteable

OBrown92, 2020-04-26:
Hello,
first of all, big respect to the development done in the last few weeks. I'am really happy to see new progress for the openvario and can hopefully get in to it soon. Also I like that all of this is now happening on Github. I installed the new image to a 7" stefly OV yesterday and it's working fine, but I get some problems with the vario sound. 
We mute the vario and then it wasn't possible to unmute it again, only a restarted solved this issue. But now we aren't able to mute the sound again. Is this happened to anyone before?
I will try to reproduce this with a fresh install tomorrow, but maybe someone has a idea so far.
Also it wasn't possible in the past to get the sounds of xcsoar (like warnings and other stuff), is this now possible with the new sound library? Would be awesome.
So, big thanks to everyone and stay healthy.

bomilkar, 2020-04-27:
It might be an issue in nmea__parser.c  of variod. Can you please look into the code to see if that is the case? There are several commands all starting with the letter 'V'.
I'm working at it. Should be done in a few days.

OBrown92, 2020-04-27:
Thanks, I will try this evening. But I'm really new in the ov development, so I will need some time to get in to it. I just saw, the issue is already in the 'known issues' of the 20115 release. So if this ticket is unnecessary feel free to close it.

kedder, 2020-04-27:
It is known, but wasn't registered on github yet, so let's close it when the issue is fixed.

bomilkar, 2020-04-27:
@OBrown92, which
> 'known issues' of the 20115 release
are you referring to?

OBrown92, 2020-04-27:
That the unmute function isn't working. In the releases here on Github is this not mentioned but I saw later that it is in the forum.

kedder, 2020-04-27:
@bomilkar it is from the release announcement, https://forum.xcsoar.org/viewtopic.php?f=30&t=3669

bomilkar, 2020-04-27:
It's still not clear to me which way you are trying to mute/unmute ?
Do you mean the "VOLUME ON/OFF" button from the openvario.xci (the F1 menue) ?
XCSoar sends the  "POV,C,VM" command to variod, which it decodes and calls "toggle_mute();".
That works in the new variod . I can see it in /variod.log . (Sorry, have no sound on this machine.)

OBrown92, 2020-04-27:
Exactly this isn't working for me.

iglesiasmarianod, 2020-04-27:
Hi, there's a problem with Variod receiving messages from Xcsoar. If you insist repeatedly you may fund sometimes it works.
Has to do with the way variod creates and destroys sockets leading to command loss.
I'm working on a proposed solution in the Variod repo. Openvario/Variod#6.
Same happens with STF and Vario Commands. With this change things seem to improve but it's not fully tested yet.

bomilkar, 2020-04-27:
@OBrown92 can you send us the log file "variod.log", It's probably at / .

iglesiasmarianod, 2020-04-28:
Tested the modified variod in my OpenVario.
Communications works OK except the volume toggle function.
STF, VAR, Volume Up, Volume Down, MC up, MC dn, work OK from XCSoar.
Checked .xci file and seems OK. SendNMEAPort1 sentences work except for mute/unmute.
I conclude from this observation that communication works.
Just took a quick look at audiovario.c. 
Mute and Unmute work OK from variod. Sound starts and stops as it should.
So the mute() and unmute() functions in audiovario.c are working.
I wonder if toggle_mute() working correctly. The line mute=!mute is really inverting mute value?
Has anyone logged that variable?

bomilkar, 2020-04-28:
@iglesiasmarianod do you see this in the log file?
debug_print("Toggle Mute\n");
And is this consistent with what you see from:
ddebug_print("Message from XCSoar: %s\n",client_message);

I've see issues with this, which I believe I have fixed in https://github.com/Openvario/variod/pull/4 : 1. split what is received from XCSoar into single NMEA sentences, and make nmea_parser more robust.

bomilkar, 2020-04-28:
I can confirm: mute, unmute, volume up, volume down, work with the changes in https://github.com/Openvario/variod/pull/4
Not only I see it in variod.log, but also hear it from the connected speaker.
Case closed.

OBrown92, 2020-04-28:
Hi again,
great work. 
@bomilkar, I don't have the logs, cause I read this too late. But seems to work now anyway or? If I should try to get them, feel free to get back to me.

bomilkar, 2020-04-28:
It works for me on my test system. This is a Texim 7" with the 20115 image with the fixed variod and a sensord mock-up to provide sensor data w/o the need for sensors.
No need for logs anymore. Thanks!

condoratberlin, 2020-05-26:
I just updated my stefly OpenVario with 7"Texim display and sensorbox and i can confirm, that the variod will be muted and unmuted with the latest sensord-testing, variod-testing and xcsoar-testing.

linuxianer99, 2020-07-30:
@OBrown92 : So we close this issue ?

kedder, 2020-08-07:
Closing this. Feel free to open if it is still an issue.

closed 2020-08-07 by kedder

OBrown92, 2020-08-07:
Yes, sorry busy times. I will test it next week in my holidays.