Openvario/meta-openvario #64: Simplify creation of USB thumb drive

phorward, 2020-05-10:
Hi there!

Thank you for creating this great hard- and software combination! A friend bought an openvario from SteFly and I helped him to get it working properly, especially with the software, and also that was the reason why I also bought a used one for myself now ;-) 

Anyways, the creation of the USB thumb drive for file up-/download as described [here](https://www.openvario.org/doku.php?id=projects:series_00:build:setupguide#transfer_data_to_your_openvario) could be done easier: Why not provide a way directly from the openvario menu to format the drive properly, when a drive is inserted and not mountable? I've already found the script files [here](https://github.com/Openvario/meta-openvario/tree/warrior/recipes-apps/ovmenu-ng-skripts/files) and think this could be a nice place to put such a drive formatting tool.

Do you agree with this idea, or are there already other developments active in this way? Would it be OK for you when I do a pull request in some time, when I'm ready to experiment with my own OV device?

linuxianer99, 2020-05-10:
So what are the difficulties with formatting an USB Stick with FAT 32 and create some folders ?

The advantage of "manual" creation is, that the stick can used also for other things, as it will not be formatted every time ;-) That was the reason for this approach.

Feel free to provide some code for that ... but please follow the [Contrib Rules](https://github.com/Openvario/meta-openvario/wiki/Code-Contribution-Rules)

phorward, 2020-05-11:
> So what are the difficulties with formatting an USB Stick with FAT 32 and create some folders ?

Hi, its only for the usability, especially for users who are not hackers or experienced computer users (even with Windows). 

> The advantage of "manual" creation is, that the stick can used also for other things, as it will not be formatted every time ;-) That was the reason for this approach.

That's exactly what this issue is meant to be. The idea is to provide a tool directly from the device that is capable to format a USB device in the way it is required by OpenVario to be used.

> Feel free to provide some code for that ... but please follow the [Contrib Rules](https://github.com/Openvario/meta-openvario/wiki/Code-Contribution-Rules)

Wilco :+1:

schonstudent, 2021-03-11:
Would be also cool to have a "Norton Commander" in standard available - maybe accessing from the starting menu
-> to delete easily flights, old maps etc

phorward, 2021-03-11:
Hello @schonstudent thank you for your comment. I try to respect this, maybe in a separate issue.
I'll soon have my OpenVario device running, then I will finally start to work on this issue.

linuxianer99, 2022-03-08:
@phorward : Is this issue still valid ??

schonstudent, 2022-03-08:
Hi Timo,

would be lovely to have this open.

Maybe even with blank files inside the USB folder(s) to know where to put what….

 

Cheers,

Sascha

phorward, 2022-03-08:
> @phorward : Is this issue still valid ??

Hello @linuxianer99, sure I would participate, right now I'm in lack of time on this. You can close the issue if wanted, but I might reopen it somewhere in future.

linuxianer99, 2022-03-08:
ok ... lets keep ti open ...
Just want to check, if there is still activity ;-)