Openvario/meta-openvario #65: Vario Audio

hpmax, 2020-05-11:
I'm hearing an intermittent warble in my audio.  In fact, there appears to be a few different types of warble.  I'm running kedder's test release, the problem is not in my "original" version.

https://drive.google.com/file/d/1FsxFNGcUYk_g-gh0v_DCaI--quW0yQlE/view?usp=sharing

condoratberlin, 2020-05-22:
Hi @hpmax, do you have checked if the XCSOAR audio vario is enabled? On my Openvario with sensorbox there was an overlay of the XCSoar VArio audio and the variod audio, which sounds terrible. Turning off the XCSoar audio solved it.

kedder, 2020-05-22:
We have reverted bunch of changes we've made to variod after latest release, that we believe introduced this regression. If 20115 image sounds good to you, it should sound as well on the next image.

linuxianer99, 2020-07-30:
Is there any improvement with the actual image ??
@kedder mentioned, the changes in variod have been reverted ...
If this is no longer an issue, i will close this ...

kedder, 2020-07-31:
I agree, this should be addressed, let's close this. Feel free to reopen if the issue is still there.

closed 2020-07-31 by kedder