Openvario/meta-openvario #71: OpenVario freezes in flight

bomilkar, 2020-05-18:
In 2 of 3 flights with the new image (version 20130 for 7" Texim) I experienced a system freeze which I had never seen with the old image in 300+ hours of flight time. I tried 2 different SD-Cards (with the same image): no difference.

My configuration:
SteFly system with adapterboard, without sensorboard
FLARM on ttyS1, e-vario on ttyS2 ($POV,E,0.58,P,944.33,Q,570.83*1E sentences 2 per second)
sensord and variod disabled with systemctl disable
SteFly stick control, no touch screen

The symptoms:
Acouple of hours into the flight I see that cursor movements (stick control in mouse mode) are getting sluggish (slow reaction). Especially on busy maps, like zoom out to 50km or more.
When moving the mouse on a zoomed-out map the system freezes, meaning no more (visible) map updates, many InfoBoxes still update, but not all. Barometric altitude freezes, for instance.
I leave it alone for ~10 minutes, keeps frozen. Power cycle is the only way to get it going again.
When I end XCSoar after the flight I see a lot of stuff on the screen for a split second before the OpenVario menu takes over the screen again. (I'll change start_xcsoar to capture this "stuff", hoping it reviles something.)
One other thing, which may be unrelated: I find it impossible to "point&click" on the map to get to the details of an airport. That was never a problem with the old image. To some extend it might be related to the "sluggishness" of the cursor movement, but even when it's spot on it doesn't want to pick the airport it is pointing to. I wonder if the "internal pointer on the map" points to the same place as on the screen.
I have the CPU-Load InfoBox open. It seems to update even if the map is frozen. CPU load is reported between 20% and 30% even in freeze. (But the InfoBox may only report user CPU, not system.)

@linuxianer99 , @kedder I think this requires attention. Whoever works/worked on the OS should be alerted. In my humble opinion the system is not usable for the "normal" pilot unless this is fixed.
I have logs from XCSoar both IGC and NMEA which I can make available if that helps. As soon as I have changed ovmenu-ng.sh to capture stderr and stdout from xcsoar I'll supply these, too. Next flight tomorrw :-) :-)

kedder, 2020-05-18:
Hm, this looks exactly like my configuration (except display and peripherials), but I've flown around 21 hours with it already without single issue. I'm using 20115-testing image with xcsoar7-preview.

Anyway, what version of XCSoar are you using? Can you try the other (testing or stable) flavor of official image (i.e. not self-built one you are using now)? Can you repro this problem on the ground?

linuxianer99, 2020-05-18:
So, what is 20130 ?
It has never be released here. 

As kedder said, i would only doing tests with released images. The content of 20130 is not reproducable ...

Scumi, 2020-05-18:
I have observed similar behaviour.

Setup:
OV 7" with Sensorboard
Connected Units:
- Air Control Display
- PowerFlarm
- LXNAV S10
- Stick Remote

I even disabled Openvario in XCSoar -> Devices.

My screen only froze for ~30 seconds completely, but lagged for good 5 minutes. Was better again for some time then. CPU loads were as high as 55% (normal up to 30%).

kedder, 2020-05-18:
@Scumi, did you disable Openvario device before you observed that or after? Did you observe this behavior after you disabled the device? What version of XCSoar are you running? Can you still repro this? Can you repro with the other XCSoar version?

linuxianer99, 2020-05-18:
Guys,
it makes no sense to chase errors in images where content is undefined ... Even if we know the date of build (20130) it is not sure which version are included, as the build date (os-release) is not updated on every build.

Lets see if we can reproduce on a release image ....

Scumi, 2020-05-18:
> did you disable Openvario device before you observed that or after? 
Disabled after Flight on Saturday where I observed laggy behaviour first.

> Did you observe this behavior after you disabled the device? 
Yes

> What version of XCSoar are you running? 
Testing (7.x)

> Can you still repro this? 
Only inflight, so earliest next week (maybe, since it doesn't occur all the time).

> Can you repro with the other XCSoar version?
I can try, but then I miss many options from XCSoar 7.x I regularly use (including my ACD driver).

Scumi, 2020-05-18:
> Guys,
> it makes no sense to chase errors in images where content is undefined ... Even if we know the date of build (20130) it is not sure which version are included, as the build date (os-release) is not updated on every build.
> 
> Lets see if we can reproduce on a release image ....

I don't fly with 20130 (I use release with git added) and still have this behaviour. Was about to open an issue, but Ronald beat me to it. :-)

linuxianer99, 2020-05-18:
@Scumi : What do you mean with "I use release with git added" ??

So it will be helpfull to have the exact version here. 
Is it testing image or stable ?
Have you changed something on the image ?? (Updated some files after burning to SDCard, etc ...)

kedder, 2020-05-18:
It is of course would be much better to repro it on the ground, because you cannot do much to debug stuff while flying. The general strategy is to narrow down the problem - find configuration that works, then change things one by one until it stops working.

Assuming we all have the same hardware and it is not at fault, there must be something in software that is causing that. Mine seem to work (20115 - testing, XCSoar 7, two drivers enabled - LX with FLARM secondary device on ttyS1 and "Generic" - my OGN tracker - on ttyS2).

Scumi, 2020-05-18:
@linuxianer99 
Testing image downloaded here with git additionally installed as well as the shell provided by @kedder (https://github.com/kedder/openvario-shell). 
No other changes have been made (apart from using my XCSoar config from the pre-warrior image).

@kedder
I agree, unfortunately I cannot predict when the lags and freezes are coming (might as well flight when I sit in front of it). Will try anyway next week on-ground, maybe I can find out something. One option for me might be to install an Android phone with XCSoar in parallel while flying to at least have a backup.

kedder, 2020-05-18:
Also, it might be worth noting the time of these "lags" and then looking at system log at that time - maybe the system was busy doing something specific at that time (for example, having problems writing to sd card). Note that system time might not be in sync with real time, so take it into account.

bomilkar, 2020-05-19:
@linuxianer99 : which "system log" are you talking about? "/var/log/messages" ??

kedder, 2020-05-19:
@bomilkar Logs are managed by `journald`, you can see/filter them with `journalctl` command, e.g. 

```
journalctl --since='2020-05-19 08:00'
```

bomilkar, 2020-05-19:
I used the plain vanilla 20115 release with XCSoar 6.8.13b from here:
https://github.com/Openvario/meta-openvario/releases
on a 4:30 flight today: exact same issue: it froze twice. I have the complete SD card here. So let me know which files you need to analyze what happened.
I can reproduce it, at least in the glider and will try tomorrow on the test system: zoom out to 250km or more, then flip pages and/or move mouse. It doesn't take long until it freezes.

I redirected stdout and stderr of xcsoar to a log file: nothing reviling. The garbage characters are still visible on the display during the split second when xcsoar stops and the Main Menu overwrites the screen. Something must be spewing to the console. As far as I can tell these are non-printable characters. And I can see them on the test system, too. The amount of charactes is proportional to the time xcsoar was running. It can be half the screed after a couple of hours.

bomilkar, 2020-05-19:
Apparently nothing happened in the 15 days since 20115 to change the behavior in any way.

kedder, 2020-05-19:
@bomilkar can you zip contents of your `/root/.xcsoar` dir?

The garbage characters you see are irrleevant. It is just your keyboard input printed into the terminal.

bomilkar, 2020-05-19:
It's in here:
https://www.dropbox.com/sh/83p53l0pgfh408v/AAD3dTRNZuUV7vOCopIqmyp6a?dl=0

Anything else?

In .xcsoar/logs are the logs from that flight including IGC file from FLARM. The gaps in the NMEA logs are the reboots.

kedder, 2020-05-19:
One thing that stands out in that data dir is map of a huge size. I suggest trying smaller map of the area to see if it makes any difference.

bomilkar, 2020-05-20:
I'm using this map since years. No issue whatsoever. Not on Android, not on the PC at home, not on the OV system in the glider with the previous OS.
I can certainly try a smaller map to narrow down the problem, but it is certainly not an option to say "on OV your map must be less than XX MB" when there is no real limitation in XCSoar and all other platforms can deal with it.

linuxianer99, 2020-05-20:
It is not about saying "you can only have small maps on OV" but to find the root cause of the issue ...
The main thing change in the warrior release is kernel and lima driver ... So graphics issue are not that unlikely...No boday is talking about have just small maps on OV ...

linuxianer99, 2020-05-20:
@bomilkar : is you are talking about vanilla release and XCSoar 6.8 you have installed stable build ??
Not testing, right ?

bomilkar, 2020-05-20:
@linuxianer99 I downloaded and unzipped OpenVario-linux-openvario-image-glibc-ipk-20115-openvario-7-CH070.rootfs.img.gz  (md5sum: 3318f6828f0615e691f89be184833158) burned it to SD, resized partition 2 to 4000MB (parted, fsck, resize2fs) copied my .xcsoar directory on the SD (the tgz file is available from my drop box, see earlier comment).
Then on the OV box: orientation to portrait and `systemctl disable sensord variod` . XCSoar 6.8.13b-Linux, April 17, 2020.
Takeoff :-)
Everything up until the freeze was OK normal.
Power cycle seem to be the only way out of the freeze.

bomilkar, 2020-05-20:
Good news: I can reproduce the freeze on my test system at home:
zoom out to 500km and try to flip pages.
The configuration is exactly like the one described above.

bomilkar, 2020-05-20:
@kedder : I tried GER_HighRes.xcm instead of the large CE.xcm: exact same behavior.
I just downloaded GER_HighRes.xcm from here: https://www.xcsoar.org/download/maps/
Zoom out to 250% or 500% and flip pages -> freezes.

One other observation: on my test system I have a e-vario connected via ttyS1 and a FLARM connected to ttyS2. As soon as I unplug the FLARM XCSoar recovers.

We probably have to skip rendering the map if the previous rendering hasn't finished.

One other thing which becomes apparent: when I want to flip pages while the system is in overload, then it keeps flipping the pages (if it can) without having to hit any keys. So it seems as if the input events from the keyboard are getting messed up when the system is in overload. ?!?!

bomilkar, 2020-05-20:
Please note: it's not the size of the map but more how busy the screen is.
If I don't load any airspace, it'll behave much more gracefully. I guess XCSoar is smart enough to not let the whole map be rendered if it only needs a portion of it. I bet if I don't load any waypoints, then the CPU load reduces, too.
In essence: as soon as the CPU goes into saturation, it'll freeze. Something smart must be done when this happens.

kedder, 2020-05-20:
>  As soon as I unplug the FLARM XCSoar recovers

What if you don't plug in FLARM at all? Can you still repro the problem?

bomilkar, 2020-05-20:
The FLARM provides GPS fixes one every second (unless it's "GPS waiting for fix").
Each fix being a little different compared to the previous fix, even if it sits on my desk not moving.
As far as I know XCSoar triggers a "refresh everything" if the position has changed. (If you have no GPS fixes, then the vario-bar on the right side doesn't change at all, unless you flip the screens.)

So "don't plug in FLARM at all" causes "refresh everything" only upon flipping the screens. So the problem doesn't exist. So, why do we have a GPS source at all??? ;-)

kedder, 2020-05-20:
> So, why do we have a GPS source at all??? ;-)

Do you have a better idea how to isolate the problem?

bomilkar, 2020-05-20:
Sorry, that was just a lame joke.
If you are looking for a way to test it w/o a GPS source:
replay a IGC or NMEA file in XCSoar
use sensord_mock -i nmea_log_with_GPRMC_sentences

kedder, 2020-05-20:
I'll try to repro, but I need to take my device out of the glider, so next week at best.

moser, 2020-05-20:
Hi,
I guess I have reproduced it. I was using 20115 testing (xcsoar 7.0 preview 15) on Stefly hardware. Map was Alps HiRes, I supplied GPS NMEA from gpsfeed+ over the network.

Here is what journalctl says:
```
May 20 19:42:51 openvario-7-CH070 kernel: [[0;1;31m[[0;1;39m[[0;1;31mlima 1c40000.gpu: gp error irq state=4 status=2b[[0m
May 20 19:42:51 openvario-7-CH070 kernel: [[0;1;31m[[0;1;39m[[0;1;31mlima 1c40000.gpu: gp task error int_state=0 status=aa[[0m
May 20 19:42:51 openvario-7-CH070 kernel[163]: [[0;1;31m[[0;1;39m[[0;1;31m[  623.625781] lima 1c40000.gpu: gp error irq state=4 status=2b[[0m
May 20 19:42:51 openvario-7-CH070 kernel[163]: [[0;1;31m[[0;1;39m[[0;1;31m[  623.632831] lima 1c40000.gpu: gp task error int_state=0 status=aa[[0m
```

Here is a video of the frozen device: https://nx9458.your-storageshare.de/s/6Ziz79zTWGcH6gY
Note that there is still audio output (some airspace warning). Zoom was pretty high (500km). It seems that only two frames are painted in (slightly random) alternation. CPU load was ~50%. XCSoar could be killed by `kill <PID>` and after that I could start it again.

kedder, 2020-05-20:
@moser thanks a lot! So this indeed looks like a lima driver issue :( Last time I saw logs resembling this, the prolem was solved by uprgrading mesa to the newest version. It was 2 months ago (https://gitlab.freedesktop.org/lima/linux/-/issues/28). We can try this trick again, but I didn't see anything lima related in recent changelog for mesa. Anyways, this narrows the number of targets to attack quite a bit.

moser, 2020-05-20:
Btw, I have been running the device for an hour now and it froze twice, but only on 500km zoom.

moser, 2020-05-20:
> @moser thanks a lot! So this indeed looks like a lima driver issue :( Last time I saw logs resembling this, the problem was solved by uprgrading mesa to the newest version. It was 2 months ago (https://gitlab.freedesktop.org/lima/linux/-/issues/28). We can try this trick again, but I didn't see anything lima related in recent changelog for mesa. Anyways, this narrows down the number of targets to attack quite a bit.

I quickly tried updating mesa to 20.0.7 but that did not fix the issue. Maybe I messed up something because I don't fully understand bitbake/yocto and all the moving parts :-)
As far as I understand, mesa is the userland part while lima is part of the kernel. The log output says kernel and states something about `gp error irq state=4`. When I google that I get to an [old issue on lima/mesa](https://gitlab.freedesktop.org/lima/mesa/issues/36) which mentions `state=4` being `LIMA_GP_IRQ_PLBU_OUT_OF_MEM`. In the [commit log of the `drm/lima` folder](https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/log/drivers/gpu/drm/lima) in the kernel, there is a (relatively) [recent commit](https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/drivers/gpu/drm/lima?id=6707b755060563cb6b10d3c390fae10a600eb19d) that should improve the handling of that error. 
The kernel version OV uses at the moment is 5.2-rc2, right? The commit mentioned above is contained starting at version 5.7-rc1. 5.7 is at rc6 right now. Any chance of updating to that (rather bleeding edge) kernel version?

kedder, 2020-05-21:
@moser that's a great find! It is definitely worth upgrading to 5.7, because it also contains my [patch](https://github.com/torvalds/linux/commit/d718e53a48f7bcfa6cdd0c00d5ed1fb516595446) to support LVDS screens on A20. I'll work on this.

kedder, 2020-05-22:
Ok, I've built an image with the bleeding edge linux-5.7-rc6. It seems to boot well on my bare display-less cubieboard (but lacks ethernet for some reason, will figure this out later). I'll retrieve my OV from the glider tomorrow and see if it makes the difference. 

Will keep you posted.

moser, 2020-05-23:
Hi @kedder,
I built an image based on the 5.7 branch from your fork. After ~1h of running XCSoar (at 500km zoom) I got the display freeze again (once until now, so frequency is less than before FWIW).
The log message is different though:
```
May 23 20:53:28 openvario-7-CH070 kernel: [[0;1;31m[[0;1;39m[[0;1;31mlima 1c40000.gpu: pp task error 0 int_state=0 status=0[[0m
May 23 20:53:28 openvario-7-CH070 kernel: [[0;1;31m[[0;1;39m[[0;1;31mlima 1c40000.gpu: pp task error 1 int_state=0 status=0[[0m
May 23 20:53:28 openvario-7-CH070 kernel[157]: [[0;1;31m[[0;1;39m[[0;1;31m[ 1997.098997] lima 1c40000.gpu: pp task error 0 int_state=0 status=0[[0m
May 23 20:53:28 openvario-7-CH070 kernel[157]: [[0;1;31m[[0;1;39m[[0;1;31m[ 1997.105193] lima 1c40000.gpu: pp task error 1 int_state=0 status=0[[0m
```

kedder, 2020-05-23:
@moser, thanks! Have to confess, I was secretly hoping for an immediate great success with 5.7, but I guess it is not gonna be that simple. 

I brought my device home, so I'm going to play with it tomorrow too. I guess we'll have to escalate this to lima devs and get some instructions how to proceed from here.

kedder, 2020-05-24:
Ok, I played with this issue a bit too.

The simplest way to reproduce the original problem I have found is this : 

1. Using 20115 testing image, download and install [.xcsoar](https://www.dropbox.com/sh/83p53l0pgfh408v/AAD3dTRNZuUV7vOCopIqmyp6a?dl=0) dir from @bomilkar 
2. Start replay of `2020-05-18_09-30.nmea` file in XCSoar
3. Zoom the map out to 500 km scale
4. Wait

After less then a minute it consistently freezes with this messages in the system log:
```
lima 1c40000.gpu: gp error irq state=4 status=2b
lima 1c40000.gpu: gp task error int_state=0 status=aa
```
XCsoar itself doesn't hang, it reacts to keyboard input and I was even able to exit it by blindly navigating menus. Just graphics stop functioning. 20115 image runs kernel 5.2-rc2.

Now, with the newest kernel 5.7-rc6 I can no longer reproduce this. I was able to replay `2020-05-18_09-30.nmea` in full (about 2 hours), on 500 km map scale. The UI is not very responsive on 500 km scale, input takes ~2 seconds to prcess and CPU is fully saturated. But it does not hang and continues to refresh the map about once in 2 seconds.

@moser, I wasn't able to repro this `pp task error 0 int_state=0 status=0` error, unfortunately. But 5.7 makes a very visible and measurable difference, so I think it is worth upgrading anyways.

@bomilkar do you want to try out the image with 5.7 kernel to see if it fixes the issue for you? You have CH070 flavor, right?

kedder, 2020-05-24:
@bomilkar and anyone interested, please test this (unofficial) image, containg kernel 5.7-rc6:

* [20145 57-lvds](https://www.dropbox.com/s/dsjhea0vfysjo4w/OpenVario-linux-openvario-image-testing-glibc-ipk-20145-openvario-57-lvds.rootfs.img.gz?dl=0)
* [20145 7-PQ070](https://www.dropbox.com/s/s5euc9f38orpvza/OpenVario-linux-openvario-image-testing-glibc-ipk-20145-openvario-7-PQ070.rootfs.img.gz?dl=0)
* [20145 7-CH070](https://www.dropbox.com/s/ykpyns4ez8kwyk6/OpenVario-linux-openvario-image-testing-glibc-ipk-20145-openvario-7-CH070.rootfs.img.gz?dl=0)

bomilkar, 2020-05-25:
Very encouraging! I'm just running CH070.
My observations at zoom level of 500km:
- it is sluggish but it doesn't freeze anymore. I can still flip pages and zoom in again.
- it seem to use more than 1 CPU: in top I see up to 80% CPU usage for XCSoar. That's good!
- at reasonable zoom levels 38km or 50km the CPU usage for XCSoar is below 10%. (It used to be in the 20s.)

**Very well done, Andrey!** 

My configuration:
FLARM and e-vario connected to ttyS1 and ttyS2.
systemctrl disable variod sensord
.xcsoar stuff from there:
https://www.dropbox.com/sh/83p53l0pgfh408v/AAD3dTRNZuUV7vOCopIqmyp6a?dl=0

I have just started a replay of a 6 hour flight (NMEA log replay in XCSoar) at normal speed. I'll report in 6 hours. I'm very confident!

bomilkar, 2020-05-25:
After ~8 hours of nmea log replay I can say there are no more freezes.
However, at a zoom level of 500km (to some extent even at 250km) the system is loaded to the extent that keyboard and mouse inputs are unusable. Flipping between pages or moving the mouse cursor is next to impossible. For instance: at zoom level 500km I hit the up-arrow to zoom in, but no reaction for minutes. Map redraws still happen. CPU load is at ~50%.
**But no more freezes.** It's definitely a big step forward!

kedder, 2020-05-25:
Great! It seems like the final 5.7 is going to happen next week, so I suggest to wait for it before merging it to openvario. I'll submit a PR once it is out.

bomilkar, 2020-05-26:
The only remaining performance issue I see is related to mouse and keyboard responsiveness at high zoom levels: 150km and more. It makes it very difficult to point&click on the map.
That should become a new Issue if this here is done.

kedder, 2020-05-28:
So, @bomilkar reported that there still was a "freeze" during his flight with 5.7-rc6 kernel. The symptoms are slightly different than the original issue (happened during circling, map zoom was about 1.5km), so this one might have a different cause, but nevertheless.

There is no known way to reproduce this problem at the moment. Until we find one, we need to try to extract logs for the event. Currently logs are volatile on openvario, meaning they are written to the RAM-disk and discarded on each reboot. I've made them more permanent with #78 - they will be stored on sd card in `/var/log`. Once that PR is accepted, I'll make another set of images to test.

What we need to do is: with new image, if "freeze" happens, do not power cycle immediately, but keep flying with frozen openvario for couple of minutes (so that file buffers have a chance to flush to the sd card). After the flight, zip contents of `/var/log/journal` directory and attach here. Also record the approximate time of the event and time when you archived the logs (so we can find out the openvario-time of the freeze)

CptFrikadel, 2020-05-31:
This might be unrelated, but I get a very similar issue on the 17119 CH070 image. 

After a couple hours of flight the system freezes. Like @bomilkar reported some infoboxes and especially the vario gauge still works. So sensord and variod are both still working. 

I run SteFly hardware with the sensor board, and flarm on TTyS2. I have tried several new SD cards to no avail. I have had a look through the journal, but I could not find anything. Mostly because I didn't know what I was looking for.

hpmax, 2020-05-31:
I believe the slowness that @bomilkar reported goes away if you remove the airspace file.

kedder, 2020-05-31:
@CptFrikadel oh, this is interesting. Do you mind publishing your `~/.xcsoar` dir from your 17119 image?

17119 is a completely different system though, different kernel (3.x), different graphics stack implementation, so might not be related, unless it is some kind of bug in xcsoar.

I'm waiting for Linus to release new kernel (5.7) today or tomorrow. Once it is out, I'll make new images based on it for testing.

Can you try the new image and report back? Will the freeze still happen?

kedder, 2020-05-31:
@hpmax :exclamation: that's an awesome observation! Indeed (just checked), 500km zoom is no problem without airspaces rendered. Now I wonder if freezes would also disappear without airspaces (or with different, smaller, airspace file).

kedder, 2020-05-31:
I just reproed the freeze by replaying `2020-05-18_09-30.nmea` on 5.7-rc6. Airspaces were switched off, map was on 500km scale. The map was not literally "frozen", but repeatedly switched between two frames. Log showed these 3 lines (similar to what @moser observed):

```
May 31 17:21:12 openvario-57-lvds kernel: [drm:lima_sched_timedout_job [lima]] *ERROR* lima job timeout
May 31 17:21:12 openvario-57-lvds kernel: lima 1c40000.gpu: pp task error 0 int_state=0 status=0
May 31 17:21:12 openvario-57-lvds kernel: lima 1c40000.gpu: pp task error 1 int_state=0 status=0
```

kedder, 2020-05-31:
It looks like [mesa-20.1.0](https://www.mesa3d.org/relnotes/20.1.0.html) was released just few days ago with bunch of lima-related changes. I'll upgrade to it and make a new image. We have to test with latest mesa anyways before reaching out to lima devs.

bomilkar, 2020-06-01:
> 
> 
> I believe the slowness that @bomilkar reported goes away if you remove the airspace file.

Are you saying the airspace file is corrupt? Is there a way to check the integrity of an airspace file (other than loading it into XCSoar)?
I concatenate airspace files from de at ch cz fr it pl si into CE (stands for Central Europe).
I can use just one country's airspace file to see if it makes a difference. But that would only be a temporary solution to isolate an issue.

bomilkar, 2020-06-01:
> 
> 
> Now I wonder if freezes would also disappear without airspaces (or with different, smaller, airspace file).

A smaller airspace file attached. Just for Germany.
[Airspace_de_2020-05-26.txt](https://github.com/Openvario/meta-openvario/files/4710343/Airspace_de_2020-05-26.txt)

CptFrikadel, 2020-06-01:
@kedder  I'll see if I can find an old SD card with the image and share. I'll of course try the new image. However it won't be until a few weeks until my club can start flying again.

What's is more interesting is that I also run two airspace files and fly in complex (Dutch) airspace. In my case (with the old image) freezes usually (but not always) happened when switching waypoints.

hpmax, 2020-06-01:
@bomilkar No, I wasn't trying to suggest the airspace file is corrupt.  I had my airspace file loaded (entire US) which I believed to be correct and saw that it was hitting high CPU loads.  I then re-flashed the drive and it came back up with the European airspace file that is far away from me.  When I switched to 500 km range I did not see high CPU loads.  Hence my contention that the problem is airspace related.  I'm suggesting the problem may be related to how many objects it may be seeing in the airspace file within the zoom range.  This may be indicative of a bug or poor in XCSoar.  Has anyone tested to see if the same behavior is present in the legacy code?

kedder, 2020-06-01:
Here are new images with just released kernel 5.7, mesa-20.1.0 and persistent logs. Hot from the oven.

* [20153-openvario-7-CH070](https://www.dropbox.com/s/sig3uo39g83gu07/OpenVario-linux-openvario-image-testing-glibc-ipk-20153-openvario-7-CH070.rootfs.img.gz?dl=0)
* [20153-openvario-7-PQ070](https://www.dropbox.com/s/awj33y59xfv9u8r/OpenVario-linux-openvario-image-testing-glibc-ipk-20153-openvario-7-PQ070.rootfs.img.gz?dl=0)
* [20153-openvario-57-lvds](https://www.dropbox.com/s/7t8i3n425hlgicp/OpenVario-linux-openvario-image-testing-glibc-ipk-20153-openvario-57-lvds.rootfs.img.gz?dl=0)

Please try it out and report back! Repeating the debugging instructions for your convenience:

> with new image, if "freeze" happens, do not power cycle immediately, but keep flying with frozen openvario for couple of minutes (so that file buffers have a chance to flush to the sd card). After [...]

bomilkar, 2020-06-01:
Great, Andrey!
Can you give us instructions on how to get the build environment updated? I'd like to include my "scalable infobox title font" patch.

bomilkar, 2020-06-01:
> @bomilkar No, I wasn't trying to suggest the airspace file is corrupt. I had my airspace file loaded (entire US) which I believed to be correct and saw that it was hitting high CPU loads. I then [...]

I wish we had fewer objects in our airspace, not just to reduce CPU load. ;-)
In any case, the number of (airspace, waypoints, topography, ...) objects on the map area at the given zoom level determines the amount of work the system has to do. One other key element is the GPS source. Since consecutive fixes differ enough to trigger a redraw the CPU load is low as soon as you disconnect the GPS source. At 500km zoom w/o GPS signal there is no issues at all. So either an empty map, or no GPS fixes will make the system run well at 500km zoom. But that's useless.

About 25 pilots are using the big airspace files (together with map and waypoints) on various platforms since over a year now. Of course, these files have been updated several times, but the only platform showing issues is the new OV image. The old image 17119 had no issue with it whatsoever.

hpmax, 2020-06-01:
I agree... I think CPU utilization is higher than in the "legacy" SD card image, but hopefully the knowledge that excess airspace objects are driving the load up maybe that will help track down what is causing it.

kedder, 2020-06-01:
> Can you give us instructions on how to get the build environment updated

I used my branch from #83 to build these, no other changes to current `warrior` branch.

[14 more replies]

PR #88 (2020-06-27): Fix the XCSoar freezing problem
Work around the XCSoar random freezing problem, as discussed on [lima/linux#33](https://gitlab.freedesktop.org/lima/linux/-/issues/33). I'm going to submit the patch to to mainline kernel.

As a bonus, remove old unused kernel patches.

FIxes #71
closed 2020-06-27 by linuxianer99