Openvario/meta-openvario #74: How to recompile xcsoar

hpmax, 2020-05-24:
I am trying to modify xcsoar to debug issues I'm having with AHRS.

I modified LevilAHRS_G.cpp in:
/poky/build/tmp/work/armv7vet2hf-neon-ovlinux-linux-gnueabi/xcsoar-testing/git-r12/package/usr/src/debug/xcsoar-testing/git-r12/git/src/Device/Driver 4

However, I cannot for the life of me figure out how to rebuild the executable which I assume is in:
/poky/build/tmp/work/armv7vet2hf-neon-ovlinux-linux-gnueabi/xcsoar-testing/git-r12/package/opt/XCSoar

The times I've been successful at rebuilding using bitbake -f -c compile xcsoar-testing or something similar it seems to overwrite the source code directory and so no change is actually made.

Is there a trick to recompiling xcsoar?

kedder, 2020-05-24:
@hpmax, bitbake is not meant for software development. It is for building customized linux distributions for embedded devices. You can use the toolchain to compile xcsoar binary with your changes, but it is not going to be very convenient and you can lose your changes if you are not careful.

The way you can do that is start devshell, which will set up the build environment for cross-compilation, build, copy the binary to the device and test:

```
bitbake xcsoar-testing -c devshell
make -j$(nproc) DEBUG=n DEBUG_GLIBCXX=n ENABLE_MESA_KMS=y GEOTIFF=n
```

Next time you run `bitbake xcsoar-testing -c devshell` your working directory may (or may not) be reset.

It's much more convenient to build XCSoar for your own native platform (without bitbake) and test your changes on your pc instead of openvario.

hpmax, 2020-05-24:
Other than producing a 170 megabyte executable this worked great.  Unfortunately, all I learned was that it is executing the code that reads in the AHRS sentences... regards them as valid, but still doesn't seem to indicate it knows it has AHRS informtion.

kedder, 2020-05-24:
I suspect most of these 170 MB are debug information built into the executable (e.g. full source code). There should be a way to strip it if it is a problem.

> AHRS was working fine all along, I just hadn't configured XCSoar to display the horizon.

Well, on the positive side, you now know how to compile things in openembedded :)

hpmax, 2020-05-24:
I assumed it was either debug information or perhaps statically linking things.  That's a lot of debug information.  But yes, it was definitely helpful.

closed 2020-05-24 by hpmax