Openvario/meta-openvario #79: TTyS3 not working

iglesiasmarianod, 2020-05-29:
Hi guys. Yesterday I connected my GPS to TTyS3 and it did not connect at all. It works on TTyS1 and TTyS2.  TTyS3 works with Armstrong image.
Do you know if someone else is having the same issue?
Do these ports get enabled with cubieborard2-cubiescreen.fex file?

kedder, 2020-05-29:
Hm, I don't have ttyS3 on my Stefly device. 

I think it might be an issue with device tree (.dts) file. There are bunch of `&uart*` definitions in `openvario-common.dts` file, but there might be a mistake somewhere. We need to check the actual wiring for tty3. What pins on cubieboard is it actually wired to?

iglesiasmarianod, 2020-05-29:
These are the schematics for the adapterboard. I think TTyS3 is inside de DB connector in Stefan's Devices.

[image]

iglesiasmarianod, 2020-05-29:
[image]
[image]

And here's the .fex definition from the Armstrong build.

```
[uart_para0]
uart_used = 1
uart_port = 0
uart_type = 2
uart_tx = port:PB22<2><1><default><default>
uart_rx = port:PB23<2><1><default><default>

[uart_para1]
uart_used = 0
uart_port = 1
uart_type = 8
uart_tx = port:PA10<4><1><default><default>
uart_rx = port:PA11<4><1><default><default>
uart_rts = port:PA12<4><1><default><default>
uart_cts = port:PA13<4><1><default><default>
uart_dtr = port:PA14<4><1><default><default>
uart_dsr = port:PA15<4><1><default><default>
uart_dcd = port:PA16<4><1><default><default>
uart_ring = port:PA17<4><1><default><default>

[uart_para2]
uart_used = 0
uart_port = 2
uart_type = 4
uart_tx = port:PI18<3><1><default><default>
uart_rx = port:PI19<3><1><default><default>
uart_rts = port:PI16<3><1><default><default>
uart_cts = port:PI17<3><1><default><default>

[uart_para3]
uart_used = 1
uart_port = 3
uart_type = 2
uart_tx = port:PG06<4><1><default><default>
uart_rx = port:PG07<4><1><default><default>
uart_rts = port:PG08<4><1><default><default>
uart_cts = port:PG09<4><1><default><default>

[uart_para4]
uart_used = 1
uart_port = 4
uart_type = 2
uart_tx = port:PG10<4><1><default><default>
uart_rx = port:PG11<4><1><default><default>

[uart_para5]
uart_used = 1
uart_port = 5
uart_type = 2
uart_tx = port:PI10<3><1><default><default>
uart_rx = port:PH07<4><1><default><default>

```
I did not wanted to modify the .fex file because I'm not very familiar with yocto yet.

kedder, 2020-05-29:
Thanks! The .fex file is not used in newer kernels, it is replaced with "device tree" (.dts) files. For openvario it is `meta-openvario/recipes-kernel/linux/linux-mainline/openvario-common.dts` file.

I guess I'll need to solder couple of wires to my DB15 connector to pleay with it.

iglesiasmarianod, 2020-05-29:
Great! Thanks!
I'll keep reading Yocto manuals and try to learn something about it.

iglesiasmarianod, 2020-05-30:
Think I found something in `sun7i-a20.dtsi`

```
			/omit-if-no-ref/
			uart5_ph_pins: uart5-ph-pins {
				pins = "PH6", "PH7";
				function = "uart5";
			};

			/omit-if-no-ref/
			uart5_pi_pins: uart5-pi-pins {
				pins = "PI10", "PI11";
				function = "uart5";
			};
```

`uart5_ph_pins` are used in openvario-common.dts file  but we need PH6 and PI10.

kedder, 2020-05-30:
Hm, that's a good find. I think you can add new pin assignment in openvario-common.dts (or just override `uart5_ph_pins` to quickly try it out.

iglesiasmarianod, 2020-05-30:
@kedder . Did overwrite pin assignment inside &pio.
In my system it built and devices connect to TTyS3.
Don't know if it is the right way to change pin assignement. Never worked with .dts before
If there's a better way please point me in the right direction.

PR #80 (2020-05-31): Fixed TTyS3
Changed TTyS3 pinout to match schmatics. Fixes #79
commit 6017dd28 (2020-05-31): Fixed TTyS3

Changed TTyS3 pinout to match schematics. Fixes #79
closed 2020-05-31 by linuxianer99