Openvario/meta-openvario #81: Support for 2 WiFi adapters?

hpmax, 2020-05-31:
I'd like to be able to support the use of two WiFi networks simultaneously.  I am using a Stratux which provides AHRS, ADS-B In, and GPS through its own WiFi network (192.168.10.x) , and I'd also like to be able to access the Internet through my phone's hotspot (192.168.1.x and everything else).  When I put in two WiFi dongles, both are seen and identified as wlan0 and wlan1 on ifconfig, but I'm not sure how to access these individually through connman.

kedder, 2020-05-31:
Does `connman` support this kind of configuration? What does `connmanctl technologies` show with two wifi dongles connected?

hpmax, 2020-05-31:
I don't have it setup at the moment, but if I recall correct it shows nothing different from a single dongle.  It doesn't identify things as wlan0 vs wlan1, just wifi.

kedder, 2020-05-31:
Well, it boils down to connman support. There is not much you can do on openvario side, apart from replacing connman with something else. But that's something I'd like to avoid at this stage.

hpmax, 2020-05-31:
I was hoping you might have some idea how to do it in connman...  I am assuming at this point, that connman is the only real support for wifi on the Openvario platform?  So its either connman supports it or... wait until another option is available which is a low priority.

linuxianer99, 2022-05-30:
@hpmax  Is this issue still relevant ?

hpmax, 2022-05-30:
Still think it would be nice to support multiple adapters, but since that seems like a significant effort, might as well close this.

closed 2022-05-30 by hpmax