Openvario/meta-openvario #89: Problem with rotary encoder

Blaubart, 2020-06-18:
I installed the last self compiled image from yesterday where touchscreen is working. If I choose NMEA Device at menu, I can select device with up and down (rotary encoder at the right hand side), but I can't choose edit, monitor, reconnect etc. (rotary encoder at the left hand side). At other menu options right and left with this encoder works. With the old image everything works fine.

kedder, 2020-06-18:
This might be an XCSoar bug. What version of XCSoar are you running? Blue one (6.x) or Red one (7.x-preview)?

Blaubart, 2020-06-18:
7.0_preview15

Blaubart, 2020-06-18:
of course, I will try!

Blaubart, 2020-06-18:
I tried an image from today with XCSoat 6.8.x. It is the same

kedder, 2020-06-18:
I just started the old image out of curiosity,  and it seems like it behaving exactly the same way as on new image.

I don't have rotary encoder, but I'm assuming it is just a usb keyboard device, producing arrow keypresses when you rotate knobs. If it is so, on old image (17119) it is possible to select device with up/down keys, but left/right button just make cursor jump to first and last device (you still can't choose edit, monitor, reconnect etc.). Maybe your rotary encoder produces some other keypresses?

Blaubart, 2020-06-18:
rotary encoder just producing arrows, yes. Up and Down works perfectly. left/right button just make cursor jump to first and last device. It's exactly the same at my and your device.

kedder, 2020-06-18:
But that is the same on old image too.  I think this is XCSoar issue and it ptobably never worked as you are expecting it to work. I think this should be filed as a bug on XCSoar project. There is not much openvario can do about it.

Blaubart, 2020-06-18:
I tried 4 different images:
- 17119 but with a self compiled XCSoar 7 (everything works)
- a selbst compiled a couple weeks old with XCSoar 7.0_preview15. Sorry, don't know exactly how old it is (everything works)
- image from yesterday with XCSoar 7.0_preview15 (dosen't work)
- image from today with XCSoar 6.8.14 (dosen't work)

kedder, 2020-06-18:
Does it work for you with normal usb keyboard (not rotary encoder)?

Blaubart, 2020-06-18:
with a keyboard it is exactly the same.

kedder, 2020-06-18:
Weird, I can't make it work on old image with my keyboard

kedder, 2020-06-18:
@linuxianer99, @mihu-ov, are you aware of any special tricks to make keyboard arrow keys navigate "edit", "monitor", "reconnect" and other buttons in "devices" menu in xcsoar on old image?

mihu-ov, 2020-06-19:
Hi Andrey, 

this is a known issue and one of the reasons why the "mouse mode" on the stick remote is still required. Other menus are affected as well, see https://github.com/XCSoar/XCSoar/pull/330 and https://github.com/XCSoar/XCSoar/issues/394. As this is not OV specific I would suggest to close this here and continue in one of the mentioned XCSoar topics.

kedder, 2020-06-19:
Hm, I don't quite understand why / how it works for @Blaubart on old image then.

mihu-ov, 2020-06-19:
I won´t believe it really works before I actually see it ;-)
IIRC it works for the first and last device in the device list but not for any of the other devices. You can scroll through devices and menu options but can not select (highlight) a specific device and then select a menu option. The first or last device stays highlighted when the cursor changes from devices to menu options.

Blaubart, 2020-06-19:
I try to publish a video. It really works!!

kedder, 2020-06-19:
@Blaubart, do you have some special version of XCSoar or it works for you with the stock 17119 without modifications?

Blaubart, 2020-06-19:
I used a couple of patches:
nearestmapitem.patch
nearestmapitem_refresh.patch
LocationOnMapEnableGoto.patch
icon.patch
more_alternates.patch
WPlistAirport.patch
FreeVario.patch

The only one I don't use since couple days is LocationOnMapEnableGoto.patch, because it dosen't work anymore:
ERROR: xcsoar-testing-git-r13 do_patch: Command Error: 'quilt --quiltrc /OpenVario/poky/build/tmp/work/armv7vet2hf-neon-ovlinux-linux-gnueabi/xcsoar-testing/git-r13/recipe-sysroot-native/etc/quiltrc push' exited with 0  Output:
Applying patch LocationOnMapEnableGoto.patch
patching file src/Dialogs/MapItemListDialog.cpp
Hunk #5 succeeded at 183 with fuzz 2.
Hunk #6 succeeded at 254 with fuzz 2.
patching file src/Dialogs/Waypoint/WaypointCommandsWidget.cpp
Hunk #1 succeeded at 320 (offset 1 line).
patching file src/Dialogs/Waypoint/WaypointCommandsWidget.hpp
Hunk #1 succeeded at 42 with fuzz 1.
patching file src/Dialogs/Waypoint/WaypointDialogs.hpp
patching file src/Dialogs/Waypoint/dlgWaypointDetails.cpp
Hunk #3 FAILED at 691.
1 out of 3 hunks FAILED -- rejects in file src/Dialogs/Waypoint/dlgWaypointDetails.cpp
patching file src/MapWindow/Items/Builder.cpp
patching file src/MapWindow/Items/MapItem.hpp
Patch LocationOnMapEnableGoto.patch does not apply (enforce with -f)
ERROR: xcsoar-testing-git-r13 do_patch: 
ERROR: xcsoar-testing-git-r13 do_patch: Function failed: patch_do_patch
ERROR: Logfile of failure stored in: /OpenVario/poky/build/tmp/work/armv7vet2hf-neon-ovlinux-linux-gnueabi/xcsoar-testing/git-r13/temp/log.do_patch.651
ERROR: Task (/OpenVario/poky/meta-openvario/recipes-apps/xcsoar/xcsoar-testing_git.bb:do_patch) failed with exit code '1'

Blaubart, 2020-06-19:
This is the video:
https://vdrblaubart.dynvpn.de/s/7eLapiGfk6MyAZ9

mihu-ov, 2020-06-19:
That´s great, thank you for proving me wrong :-)
Can you check the info pages for the git references of the two working self compiled and the two not working XCSoar versions?

mihu-ov, 2020-06-19:
@Blaubart: Have you manually included the patch mentioned in https://github.com/XCSoar/XCSoar/pull/330 in your self compiled XCSoar versions?

Blaubart, 2020-06-19:
> @Blaubart: Have you manually included the patch mentioned in XCSoar/XCSoar#330 in your self compiled XCSoar versions?
I don't know which patch you mean.  

> Can you check the info pages for the git references of the two working self compiled and the two not working XCSoar versions?
Yes, I will check

kedder, 2020-06-19:
@Blaubart it would also be interesting to know if you can navigate with arrow keys on raw 17119 image from https://openvario.org/doku.php?id=downloads, without any modifications to xcsoar or its data files.

The problem is pretty annoying to me too, I'd really prefer to be able to use arrows to navigate everywhere, so I'm interested to understand what makes it work in your case.

Blaubart, 2020-06-19:
Ok, I'll check this evening. Can I see info pages for the git references only when XCSoar starts, or can I find at the menu, too?

kedder, 2020-06-19:
I believe you can see the exact version in "Credits" page in 3rd "Info" menu.

[image]

Blaubart, 2020-06-19:
[image]
[image]
[image]
[image]
Nr 1 and 2 are working, number 3 and 4 don't

Blaubart, 2020-06-19:
image 17119 for Pixel Qi works. I've changed nothing! Just copied to SD card
[image]

Blaubart, 2020-06-22:
any ideas??

mihu-ov, 2020-06-22:
Re-build 7.0 preview 15 from May 2nd and test just to make sure it really works without additional patches and then bisect with 7.0 preview 15 from June 16th? This all seems a bit strange ...

Blaubart, 2020-06-22:
How can I rebuild the image from May 2nd?

kedder, 2020-06-22:
@Blaubart, I loaded stock 17119. When I go to devices, select the device C, press "left" on a keyboard, cursor just goes to the device A. It just doesn't work the way you describe it for me :( Am I missing anything obvious?

Blaubart, 2020-06-22:
No, I made it the same way!

kedder, 2020-06-22:
Why does it work differently then?

Blaubart, 2020-06-22:
Sorry, no Idea!! I'm not a software engineer :-D

iglesiasmarianod, 2020-06-30:
Hi guys, had a similar behaviour some time ago with Armstrong image.
I made my own encoder code. It seemed to work OK but was a bit jittery in behavior.
I debounced with resistors, capacitors and a timer.
With this code I was able to access the devices page side menu regardless the device I had highlited.
Because my code was so fuzzy and inaccurate I implemented a state machine to count encoder pulses.
From that time on I wasn't able to go back to selecting those options. I wasn't able to select them via keyboard either. 
I think the fact @Blaubart can select those options has something to do with the encoder code rather than XCSoar.

Blaubart, 2020-06-30:
But I used always the same encoder code.

iglesiasmarianod, 2020-07-01:
Hi @Blaubart!
I always thought that getting to those options with an encoder, in the old version of OpenVario image, might have been a "lucky bug" of my encoder code. I wasn't able to get there with a keyboard so I concluded my encoder code was faulty.
Let me load 17119 and test.
I did not fully understand if you tested it with a pc keyboard or not.

Blaubart, 2020-08-23:
I like to compile XCSoar 7.0 preview15 commit 9d7767596bdcb24c8ff2d9ea282703fa4040cf9b again, to check if rotary encoder works again. How do I checkout this commit? I tried 
git clone git://github.com/XCSoar/XCSoar.git;protocol=git;commit=9d7767596bdcb24c8ff2d9ea282703fa4040cf9b

but it seems I get the newest version. How I could be sure getting the right version?

kedder, 2020-08-23:
Did you try specifying [SRCREV](https://www.yoctoproject.org/docs/latest/ref-manual/ref-manual.html#var-SRCREV) variable?

Blaubart, 2020-08-23:
No. What do I have to use??

kedder, 2020-08-23:
Oh, I was assuming you are using bitbake recipe to compile it. If not, `git checkout` is your friend.

Blaubart, 2020-08-23:
Thanks!! I got it 2 minutes ago ;-)
How Ould I verify that I really have the right commit?

Blaubart, 2020-08-23:
But you are right, in future it is easier to use it with bitbake. 

in xcsoar-testing_git.bb is written:
git://github.com/XCSoar/XCSoar.git;protocol=git;branch=master \

What do I have to change there, if I like to have a specific commit??

kedder, 2020-08-23:
I don't know, but there's `SRCREV_pn-xcsoar-testing` variable in that file. Try putting the commit id there.

Blaubart, 2020-08-23:
SRCREV_pn-xcsoar-testing = "9d7767596bdcb24c8ff2d9ea282703fa4040cf9b" 

Like this?

kedder, 2020-08-24:
That looks good to me.

Blaubart, 2020-08-25:
It works, thanks

Blaubart, 2020-08-26:
that's crazy. I tried to recompile XCSoar 7.0 preview 16 git 9d7767596b. Last time rotary encoder works with this version, but today they don't. Last image from openvario website (17119) works.

Blaubart, 2021-06-03:
Could this solve the problem?
https://github.com/XCSoar/XCSoar/issues/625

kedder, 2021-06-03:
Yes, it might!

linuxianer99, 2021-12-07:
@Blaubart , @kedder : Could we close this issue ??
Seems to be XCSoar bug, not Openvario specific ...

Blaubart, 2021-12-07:
I'm not sure it's really not an OpenVario problem. OpenVario is obviously recognizable as a keyboard-only device for XCSoar. Wouldn't something have to be adjusted in OpenVario so that XCSoar recognizes it not as a keyboard-only device?

Blaubart, 2022-01-07:
The problem seems to be that XCSoar recognizes that the image contains libinput. As a result, XCSoar assumes that it is not a keyboard-only device. But by no means all users of the OpenVario use a touchscreen! All the users will then not be able to operate the OpenVario properly. Wouldn't it make sense to build XCSoar for the OpenVario in such a way that this option is not queried and the OpenVario is always seen as a keyboard-only device?

Blaubart, 2022-01-07:
sorry, I accidentally closed !!

linuxianer99, 2022-01-20:
@Blaubart Think this can be closed ??

Blaubart, 2022-01-20:
yes, sorry, I forgot!!

closed 2022-01-20 by Blaubart