Openvario/meta-openvario #90: XCSoar profile lost after abrupt power off

kedder, 2020-06-29:
According to several reports, it is pretty easy to lose the active xcsoar profile by switching off the power supply. Repruducing is pretty simple:

1. Run XCSoar
2. Change something in the settings
3. Exit XCSoar
4 Turn the power off (without using Power Off menu)
5. Turn power back on

After the system boots, XCSoar will start unconfigured, with all-white UI without any map. This happens because `~/.xcsoar/openvario.prf` became of 0 bytes length.

CazYokoyama, 2020-06-29:
The solution is "run sync" on "3. Exit XCSoar", correct?

kedder, 2020-06-29:
Yep. I'll post a PR after I get a confirmation from users (with issue fixed in [20181-openvario-7-CH070](https://www.dropbox.com/s/ujyto5i2tud343e/OpenVario-linux-openvario-image-testing-glibc-ipk-20181-openvario-7-CH070.rootfs.img.gz?dl=0)).

hpmax, 2020-06-30:
I have generally had a lot of problems with the system getting corrupt after an errant power off.  Given the nature of the system, I think this really needs to be addressed.  Having to go through a sequence to safely shut down seems unreasonable.  My system is wired directly through my main power switch, so if someone flips the power switch on to test the radio or anything else, that forces you to go through the entire boot sequence and then shut down.  Or if there is a bad battery or bad battery connection, or the breaker trips in flight, you could end up rebooting and have no flight computer.

PR #91 (2020-07-01): Flush filesystem write-cache after key write operations
Force write-cache flushing to avoid losing data when user turns off
power without shutting down the device gracefully.

Fixes #90.
closed 2020-07-01 by linuxianer99