Openvario/meta-openvario #94: Release new preview image

kedder, 2020-07-01:
@linuxianer99, @mihu-ov, how about releasing a new preview image (after merging #93)? There were a bunch of good improvements since the last release.

mihu-ov, 2020-07-01:
Hi Andrey, you are much more up to date on current status than me, if you think the time is right go ahead.

linuxianer99, 2020-07-02:
Yes! good idea! i will have a look on this the next days

linuxianer99, 2020-07-11:
Build are on the way ...

kedder, 2020-07-11:
Thanks @linuxianer99! @mihu-ov, would it be any trouble for you to write an announcement?

linuxianer99, 2020-07-11:
maybe wait 1 day. there is some error in the build process. I have to debug first.

kedder, 2020-07-11:
sounds good!

mihu-ov, 2020-07-12:
@kedder: I am happy to make an announcement once the binaries are availabe.
Would it be possible to include the XCSoar start screen rotation fix even if it is not implemented in XCSoar yet? What about Openvario Shell and Competition Manager? Could we include them as ipk for those who want to switch over?

kedder, 2020-07-12:
@mihu-ov, I can post a openvario PR with rotation patch. Even though XCSoar PR is there (https://github.com/XCSoar/XCSoar/pull/453), you never know when it gets reviewed / merged.

For openvario-shell and competition manager, these are available as ipk packages, just in my own PPA (personal package archive) - [installation instructions](https://github.com/kedder/openvario-shell#installation) should be pretty easy to follow. I can't just post few .ipk files to download, because they also require some dependencies. I'd prefer to keep them in the PPA until that software matures, because it allows for quicker release cycle - it is much easier for me to release a new version or a fix and doesn't involve posting pull requests and involve other people.

mihu-ov, 2020-07-12:
@kedder : If you consider the rotation patch good to go for Openvario I suggest merging it even with XCSoar review pending.
@linuxianer99 : When do you expect the "testing" images will be available?

kedder, 2020-07-12:
here you go: https://github.com/Openvario/meta-openvario/pull/96

linuxianer99, 2020-07-12:
@mihu-ov 
@kedder 
Uploads are on the way. 
I have no change to flash the image for testing ... Maybe one of you can make a quick check ?

closed 2020-07-19 by linuxianer99