Openvario/meta-openvario #98: airspace warning - sound dosen't work

Blaubart, 2020-07-24:
I use a self compiled new image with XCSoar 7.0_preview 15. I disabled variod because of testing vario sound of my FreeVario.

Sound for airspace warning dosen't work. I only have a strange sound. 
[Warnings.mp3.zip](https://github.com/Openvario/meta-openvario/files/4971576/Warnings.mp3.zip)

I like to test with stable version, but there is no menu Audio-Vario or Audio in general.

Blaubart, 2020-07-29:
nobody else has this problem??

kedder, 2020-07-30:
I couldn't check airspace warnings specifically yet, but I remember other sounds (e.g. take-off) do work on my device. I'll check the how airspace warning sounds like when I get to my testing device (probably next week).

Blaubart, 2020-07-30:
Other sounds wok, but airspace warning not

linuxianer99, 2020-07-30:
But then it is XCSoar problem, right ?
As audio subsystem is working fine in Openvario ...

Maybe you can address this issue in XCSoar forum ?

Blaubart, 2020-07-31:
I can't evaluate if it is a problem of XCSoar or OpenVario. I already ask about this problem in XCSoar forum, but didn't get any answer.

kedder, 2020-08-03:
Okay, I'm finally back and first thing I did is I tested the airspace warning sound. TLDR: it sounds normal (like one on android mobile) in my case.

Here's how I tested:

1. I've flashed the latest released image (`*-testing-glibc-ipk-20193-*`)
2. Started it with no changes to xcsoar profile. Went to xcsoar system config and picked the only available airspace file, shipped with the image.
3. Since my testing unit is not connected to real GPS, nor I can fly it into the restricted airspace, I exit the XCSoar and start simulator, by running `/opt/XCSoar/bin/xcsoar -simulator` from the console and then used altitude infobox to get into the nearest vertical airspace.

The airspace warning sounded just like one on android phone. 

Now, maybe it is a problem with sound mixing (variod sound + xcsoar own sounds)? XCSoar simulator doesn't open real devices, so variod doesn't emit sound when xcsoar is running in that mode. To trick variod into thinking it is connected to real xcsoar, I started xcsoar on my laptop, and used ssh port forwarding to make it appear xcsoar opened port on openvario:
```
ssh -R 4352:localhost:4352 openvario-7-CH070.local
```
Now I can hear the variod sound and run xcsoar simulator at the same time.

But no, I still cannot reproduce this problem - warnings sound just right, even when mixed with vario sounds.

@Blaubart are you testing with the stock xcsoar configuration from the image, or you have uploaded your own config?

Blaubart, 2020-08-03:
I uploaded my own config. I will test it the same way you did.

Blaubart, 2020-08-03:
airspace warnings in simulator works with my image, too. Tomorrow I will test warings with Condor again.

kedder, 2020-08-04:
@Blaubart can you share your config (`~/.xcsoar` directory) and a short NMEA log file where you would approach the restricted zone (so the warning is emitted)? I'd like to try to replay it on my unit.

Blaubart, 2020-08-04:
Yes, I will send you an email. Tomorrow I go to my holidays, so I'm not sure if I could log the NMEA stream until tomorrow. If not I send it next week.

Blaubart, 2020-08-04:
I don't know what I did wrong, but today it works perfectly with Condor! Sorry, it seems to be a user problem.

closed 2020-08-04 by Blaubart