Openvario/meta-openvario #103: Turn off Backlight on Shutdown

iglesiasmarianod, 2020-08-08:
Hi guys, Is there a way to turn off the backligth before the cubieboard shuts down?
My image shows a strange fading screen on shutdown. If the backlight was off it would not show.
It is something similar to what happened at satrtup before @kedder fixed the logo sequence.

Thanks!

kedder, 2020-08-08:
It used to work on kernel 5.2 (backlight turned off on shutdown), but after upgrading to 5.7 it is no longer happening (in fact, it backlight turns of sometimes, but not always)

iglesiasmarianod, 2020-08-10:
Thanks again @kedder!
Do you think adding these lines to the .dts file might help?

```
	backlight: lcd {
		compatible = "pwm-backlight";
		pinctrl-names = "default";
		pwms = <&pwm 0 5000000 0>;
		brightness-levels = <0 8 16 32 64 128 255 384 512 768 1024>;
		default-brightness-level = <10>;
		post-pwm-on-delay-ms = <10>;
		pwm-off-delay-ms = <10>;
	};

```

kedder, 2020-08-10:
Hm, I believe `backlight: lcd` is already on the .dts, except `post-pwm-on-delay-ms` and `pwm-off-delay-ms`. TBH, I have no idea whether they might help or not. Do you want me to try? Did _you_ try?

iglesiasmarianod, 2020-08-10:
Just asked perhaps you knew before trying.
I just added both lines 

```
post-pwm-on-delay-ms = <10>;
pwm-off-delay-ms = <10>;
```
I'm building the image. No errors so far.
I'll let you know if it works when I load it to the OpenVario.

iglesiasmarianod, 2020-08-10:
It did not help.
I'll keep looking in case I find something useful.

kedder, 2020-08-26:
I've discovered something funny in relation to this issue:

The backlight power can be controlled by writing `0` or `4` to `/sys/devices/platform/lcd/backlight/lcd/bl_power` file. `0` means "power on", `4` means "power off" (according to https://www.kernel.org/doc/Documentation/ABI/stable/sysfs-class-backlight).

So, if you do `echo 4 > /sys/devices/platform/lcd/backlight/lcd/bl_power` the screen is supposed to go dark, and with `echo 0 > ...` screen should go back bright.

However, this is not what happens on my openvario. The backlight turns off only _sometimes_. And the probability of it turning off _seems_ to correlate with screen brightness itself: the higher the brightness, the less often screen goes dark when I write `4` to `bl_power`. When I set the screen to full brightness, it never goes dar. On 50% of the brightness it goes dark about half of the times. Interestingly, when screen does not go dark, it goes to full brightness.

I assume, when system shuts down, it effectively does the same as writing `4` to `bl_power`, so screen does not turn dark.

This all feels like a bug, just not clear where exactly.

kedder, 2020-08-26:
It feels like the line, controlled by pwm stays in whatever state it was the moment backlight was turned down - either full 1 or 0. Even though the pwm backlight driver explicitly asks to set the duty cycle to 0:

https://github.com/torvalds/linux/blob/master/drivers/video/backlight/pwm_bl.c#L81-L83

It might be that `pwm-sun4i` module does not execute that instruction very precisely. Sigh, where can I even discuss that? :(

iglesiasmarianod, 2020-08-26:
Great Andrey!
What do you think about writing 0 brightness with the shutdown script and
writing previous brightness at menu startup?
Would that impact Uboot backlight settings?
I know it is not the best solution. It would only be a patch.

kedder, 2020-08-26:
I'm not a huge fan of that, because one day the screen will simply not turn on (for whatever reason). I'd rather live with the screen that does not turn off on shutdown. But I still think this can be fixed or worked around somehow.

iglesiasmarianod, 2020-08-26:
You are right.  I didn't think about that.

kedder, 2020-08-27:
Submitted the bug report to linux devs: https://www.spinics.net/lists/linux-pwm/msg13689.html

kedder, 2020-09-02:
It looks like we have a fix for this: http://lists.infradead.org/pipermail/linux-arm-kernel/2020-September/597482.html, going to submit a PR soon

PR #105 (2020-09-03): Turn off the display backlight on shutdown
This fixes the problem of screen backlight staying on after system
is shut down.

Fixes #103
closed 2020-09-03 by linuxianer99