Openvario/meta-openvario #107: Sensor dosen't send climb rate and pressure value

Blaubart, 2020-10-27:
I disabled variod because I like to disable vario sound from OpenVario and like to use vario sound from FreeVario. At October the 21th I compiled the image new (git d6368f6102), but sometimes FreeVario dosen't get any climb rate. Sometimes it take a couple seconds and OpenVario sends climb rate to the Freevario. I adjust TCP port 4353 at NMEA Devices.

kedder, 2020-10-28:
Did you compile `-testing` variant or stable? What version of sensord did you get?

Blaubart, 2020-10-28:
I compiled testing variant. How do I figure out which version of sensor is installed?

kedder, 2020-10-28:
The version of sendord is displayed in "system info" page in the menu.

We did some changes to the sensord recently, but they are only applied to the -testing version of the image. Can you also build the stable version and check if you are still experience the problem?

How hard is it to reproduce the problem?

Blaubart, 2020-10-28:
Version of sensord is git-r9. I will test stable, too. I think it is easy to reproduce. Download the image from www.FreeVario.de and you should have the same error

Blaubart, 2020-10-28:
I don't have the problem with the stable version

kedder, 2020-10-29:
Ok, I'll try to repro myself. If not, I'll give you couple of binaries to try.

kedder, 2020-11-02:
It looks like this is a regression after https://github.com/Openvario/sensord/pull/25

hpmax, 2021-01-01:
Please see [https://github.com/Openvario/sensord/issues/28](url)

closed 2021-01-13 by Blaubart