Openvario/meta-openvario #109: Menu proposal

iglesiasmarianod, 2020-11-10:
Hi guys, It's been a while since I last wrote.
I've been thinking about upgrading OpenVario menu for some time now.
I've been playing with LVGL and made the example work with OpenVario.
https://lvgl.io/
It will give us a C interface with many widgets already functional.
I used Linux framebuffer to run the example in the Linux terminal. 
I'm now working on the drivers to achieve screen rotation.
LVGL uses MIT license, don't know if this would be an issue.
What do you think about this? Is it worth it to go this way?
How would you integrate it with the repo structure?
I'm asking this to branch my fork in a way that can be merged later here.

Cheers.

kedder, 2020-11-10:
I was considering GUI for the menu, but decided to go with simple text interface instead, since it is easier to develop and maintain. So I ended up with implementing [openvario-shell](https://github.com/kedder/openvario-shell) in python. It is pretty usable already, but lacks any fancy graphics, obviously.

iglesiasmarianod, 2020-11-10:
Hi @kedder. Thanks for your thoughts. Your menu looks great.
One of the reasons I'm thinking about this is to include some way of controling the .conf files without a text editor.
If we can change values that way we can have a vario gain setting you can modify without editing variod.conf with a text editor.
I wasn't able to find a way of doing this using dialog.

kedder, 2020-11-10:
That should be pretty easily doable in python in openvario-shell. That's on my todo list, but now I'm busy implementing tools to set up wifi connection.

bomilkar, 2020-11-11:
I agree with @kedder :  we don't need fancy graphics. And I personally find there is nothing wrong with a text editor to change configurations as long as the S/W which uses this configuration file is robust enough to deal (and flag) miss-configurations.
There are more pressing issues: WiFi connect, multi-pilot configuration, porting to other platform(s), etc.
I'm very happy with kedder's openvario-shell. Thanks Andrey!

phorward, 2021-07-08:
@kedder are there plans to include openvario-shell into the standard openvario distribution?

kedder, 2021-07-08:
@phorward yes, but I need to make few adjustments to make it more usable by a wider population. Apparently the current way of moving files to/from openvario is not very clear to most people: I have to explain it every time someone tries to upload a waypoint file to XCSoar. 

Otherwise I have a bitbake recipe to build an [image with openvario-shell](https://github.com/kedder/openvario-ppa/blob/master/recipes-ppa/images/openvario-image-ovshell.bb) built-in in my [ppa repo](https://github.com/kedder/openvario-ppa).

closed 2026-03-18 by linuxianer99