Openvario/meta-openvario #120: Voltage indicator

glidero, 2021-02-09:
Hello,
the battery voltage indicator is not working in "stable" image version. But it does work fine in the testing one.
The voltage_config parameter was set the same in sensord.conf file. Tried on recent images 21038 ( compiled on 06.02.2021).

hpmax, 2021-02-19:
Can you try to determine what the issue here is?

If the voltage isn't working it's likely the voltage_config in /opt/conf/sensord.conf is wrong, please see the following:
#voltage_config 736.0          # Use this for ADS1100
voltage_config 1248.6 0.645 # Use this for ADS1110 

The # line is commented out.
The first number is a gain, the second number is an offset.  Older versions of sensord.conf file only used 736 which will often cause no voltage to be displayed if you are using the ADS1110 (which is the case for most newer OVs).

Older versions of sensord do not support an offset (and will effectively always treat offset as 0, which will cause some inaccuracy).  I got the offset term empirically looking at my unit, and it's pretty accurate.  The ADS1100 likely should have an offset too, but since I don't have an ADS1100 based sensor, I haven't tested it.

glidero, 2021-02-21:
Thanks Dan,
I haven't mentioned that in first post, but I've tried both settings, for ADS1100 and ADS1110 (by commenting the line). 
But this is only on testing, red, image. On normal, blue one, there is only 1 option in sensord - old one, without offset. 
On my devices, voltage_config values varies from 1160 to 1200.
Last time I noticed another odd thing. On testing image, where the voltage indicator was working (ADS1100 line active) I've changed the language to german in shell menu and... the indicator stopped. No readout. Now I'm confused even more.
I will make more tests and report it here. Perhaps I'm just missing something.
Daniel

hpmax, 2021-02-21:
Apparently half the OV users are named Dan or Daniel.... This is getting confusing.

I added the offset term because it's impossible to get an accurate reading without it.  It won't be dramatically off, but it will be off by a couple hundred mV.  I just looked at it, and noticed also that I'm missing the the battery voltage, but what's weird is that there seems to be a properly constructed voltage sentence: "$POV,V,+11.97*14" (checksum seems correct when I manually check it) when I monitor the port.  Can you confirm that as well?  The weird thing is I haven't changed anything about that in a long time.  

Only thing I can think is the issue is in XCSoar, or the string is improperly terminated, or perhaps you aren't supposed to have a + in there?  However, it should be terminated the same way as all the other strings and I am seeing vario data...  I'm not seeing baro data though.  It claims no QNH, which may be why it won't display baro altitude, but I don't understand why it doesn't have QNH or how it gets it...

DanD222, 2021-02-21:
Regarding QNH - you need to set it manually in the Flight Setup menu (or by clicking on the Baro altitude infobox)

You can also check the sentences XCSoar is receiving from a particular device under Devices / Monitor

hpmax, 2021-02-21:
@DanD222 The weird thing is in the past I've seen it come up automatically.  I don't know where it was getting the information though.  And yes, I did check the sentences, I was getting baro pressure, IAS, voltage, and rate info and they all looked good as described above.

hpmax, 2021-02-21:
I did some playing with it.  The problem is NOT the '+' sign.  When I manually set my QNH it worked, I got baro altitude back.  I did recently reload the image, and switched from XCSoar 7 to XCSoar 6.8.  I'm wondering if both the QNH and battery voltage are related to that, or if perhaps the default config between the two images was different.  I can't find anything wrong with sensord or variod and I'm definitely getting valid voltage sentences out of variod into xcsoar.

kedder, 2021-03-26:
Should get fixed after upgrade to newly released 7.0 in #127

PR #127 (2021-04-06): Upgrade stable xcsoar to 7.2
XCSoar 7.0 was just released. Upgrade stable image to 7.0.

Fixes #39 
Fixes #120
closed 2021-04-06 by linuxianer99