Openvario/meta-openvario #123: LXNAV S10 communication issues

Scumi, 2021-02-22:
I have some issues with the connection of my LXNAV S10 variometer and XCSoar on the Openvario. 

1. After booting both devices, the S10 does not send its $PLXVx-sentences, which means some data (OAT and air data for example) doesn't get transferred. After hitting "reconnect" in the Devices-menu OR connecting the S10 in parallel to an Android smartphone with XCSoar running, it starts working.
2. I cannot in any way declare a task to the S10. It always fails. There is no workaround like the one for the first issue above.

As said above, this is specific to the OV+S10 combination. I tried the RJ45 ports (ttyS1-ttyS3) and even ttyUSB0 (with a converter cable).

If I hook-up the LXNAV S10 via Bluetooth to an Android version of XCSoar or to my Laptop (ttyUSB0) for the Linux version, the full functionality is there - no issues.

Any ideas?

kedder, 2021-02-22:
What version of XCsoar are you using on OV and on your Android? Does the issue 1 depend on what device boots first?

Scumi, 2021-02-22:
Testing version (although from various commits).
- Android: preview14 (#1ae852f937)
- Linux Laptop: master
- OV: various, currently #d50b9e442f

No difference for booting order of OV/S10.

kedder, 2021-02-22:
So, you are not receiving any $PLXV* sentences on OV initially? Do you receive any NMEA traffic from S10 at all?

Scumi, 2021-02-22:
Yes, exactly.
GPS and $LXWP* NMEA sentences are coming through.

kedder, 2021-02-22:
TBH I can't come up with recipe here. Try to figure out if the NMEA sentences are actually emitted by S10 over the serial port. I don't believe XCSoar would filter some particular sentences out. Most likely the signal isn't being emitted from S10.

Scumi, 2021-03-07:
When I intercept the data stream in Linux without XCSoar running, I do not see the "missing" sentences. They start to be emitted once XCSoar correctly connects to the device, which is somehow not happening by default on the Openvario image. After re-connecting, $PLXV* are transmitted, but uploading a task still fails (so its probably a separate issue).
Generally, issues with the LXNAV S8x/S10x devices are known for a while: https://forum.xcsoar.org/viewtopic.php?f=30&t=3195