Openvario/meta-openvario #130: Release new official image

kedder, 2021-04-18:
It's been a while since we released the official image. I suggest cutting a new release, after new XCSoar 7 was released.

linuxianer99, 2021-04-20:
@kedder : Will try to manage this in the next few days ...

hpmax, 2021-04-27:
How is this coming?  Is a CH070 image available yet?

linuxianer99, 2021-04-27:
Created some scripts for automation the last few days ...
Hopefully tomorrow i can start the build ...

linuxianer99, 2021-04-28:
Builds are running ...
Think we don't need -DS2 images at the moment ?? It will double the build time for release ...

Developer66, 2021-04-28:
Is there any fix for issue  #97 in the new version? 

We have two Stelfy with 201903 in different planes with the same vario behaviour. Several Variopeeks with +-5m/s while flying.

linuxianer99, 2021-04-28:
@kedder : variod and sensord have no new release right now. 
So the stable image will have the already released versions, but the testing image has the bleeding edge version.

Think this is intentional, right ??

kedder, 2021-04-28:
well, yes, I think those bleeding edge versions were never actually tested, so I'd expect them in the testing version. Hopefully more people will try them out.

hpmax, 2021-04-28:
@Developer66 I honestly don't remember exactly what was going on with that, I know that running Version 6.x of xcsoar had issues.  I believe the latest versions of variod and sensord are working as well as they are going to before the new hardware is introduced.

Developer66, 2021-04-28:
@hpmax I also dont know whats going on/wrong. We had these Vario Peeks while flying straight or in thermal. The Vario jumps to +-5m/s and goes back to a normal value. Tested on two Stefly on differnt planes with same software.
So there must be some Software issue with that. ;(

hpmax, 2021-04-29:
@Developer66 Honestly, I've lost interest in the old hardware.  I'm just starting development of software for the new software.  Let's try out the new release and if it's working call it a day, otherwise I'll try to hack together a fix.

@linuxianer99 it looks like the new release is out, but no images... will they be coming shortly?

linuxianer99, 2021-04-29:
@hpmax : Please be patient ... Yesterday, kernel.org download have been failed ... Server not reachable ..

linuxianer99, 2021-04-29:
@kedder : Built the openvario-7-PQ070 machine an hour ago. Both images have only 7,5MB ... openvario-7-CH070 has about 80MB. Content seems to be 1,2GB in both cases .... 
But looks strange to me ... any idea ??

hpmax, 2021-04-29:
BTW, is there any reason why the file space is so small?  SD cards are pretty cheap these days.  In fact, I don't think cards are typically less than 16GB.  Doesn't seem like an unreasonable requirement to put on it.

kedder, 2021-04-29:
@linuxianer99 hm, I'm not sure, but I noticed that when I build an image without any changes on the same day, I have this almost empty image. Not sure if that's the case here too, but you can try rebuilding `openvario-7-CH070`. If you get the 7.5MB image, then you're hitting the same thing.

@hpmax there is no requirement, there's just nothing to put on the image. The normal image size should be around 80MB.

linuxianer99, 2021-04-30:
@kedder : I build all machines in seperate build folders. Just to avoid such problems ... Hmmm ....
Will delete the folder and try again ...

hpmax, 2021-05-04:
Not to obnoxious or pushy, but has there been any progress?

linuxianer99, 2021-05-05:
Still have the problem with the small image file size...
Tried a lot of things now ...

Nevertheless, is published the release and added the artifacts, which are looking good ... Rest will follow ..

Developer66, 2021-05-27:
Anything new about the openvario-57-lvds release?
Thanks

linuxianer99, 2021-05-30:
@Developer66 : I tried again ... But the image is still only 6MB of size.
Nevertheless, can you please try flashing it ?? But be prepared to remove the SDCard ;-)

Thanks,
Timo
[OpenVario-linux-openvario-image-glibc-ipk-21118-openvario-57-lvds.rootfs.img.gz](https://github.com/Openvario/meta-openvario/files/6566552/OpenVario-linux-openvario-image-glibc-ipk-21118-openvario-57-lvds.rootfs.img.gz)

Developer66, 2021-06-07:
Unfortunately the file did not work :(
@linuxianer99

saxthi, 2021-11-22:
Anything new about the openvario-57-lvds release?
Thanks

linuxianer99, 2021-11-24:
Hey All!
I talked to a friend of mine, who is expert in Openembedded and bitbake. He gave me some tipps for this issue. I will try to check these in the next few days ... Hopefully this solves the issue ...

linuxianer99, 2021-11-27:
ok! So the openvario-image is uploaded for all platforms.
Testing relase is not building because of XCSoar. i will have a look on this.
@Developer66, @saxthi : Can you please test the release ??

saxthi, 2021-11-27:
Hi all, I just loaded it without any problems. I also loaded the right map but did not do any further checking…

Thanks!!

linuxianer99, 2021-11-28:
Great to hear !!
So how should we proceed with the testing version ?
Purpose was to have the latest XCSoar there ...

Maybe we can do a release with the latest buildable XCSoar within this base version of Openembedded ??
@kedder , @Developer66 : Do you know, what was the last version building with the used compiler ?

linuxianer99, 2021-11-28:
ok!
so i compiled XCSoar 7.15 for the testing release.
Feel free to check :-)

Think i close this ticket ;-)

closed 2021-11-28 by linuxianer99

Developer66, 2021-11-28:
Thank for your work!

I will give this a try in the next weeks.