Openvario/meta-openvario #132: Openvario on Cubietruck (dri device error)

palomba01, 2021-04-30:
I'm testing the openvario on a cubieboard3 (Cubietruck), but after the openvario background on boot it appears the following error:
"Exiting... 
Using DRI device /dev/dri/card0 (use environment variable DRI_DEVICE to override"
Any tips for fixing this error?

hor63, 2021-08-30:
Check if /dev/dri/card1 existists. If not you do not have a working Lima driver compiled in your kernel or loaded.
Check journalctl and dmesg for "lima", if messages exist, and if there are any errors upon loading.
If it exists try:
`DRI_DEVICE=/dev/dri/card1 -xcsoar`

linuxianer99, 2022-01-20:
Is this still a valid issue ??
Think Cubietruck is not a supported platform ....

hor63, 2022-01-27:
Cubietruck still uses the A20. Thus this is merely a question of packaging that the actual software, particularly when the issue deals with the Mali which is independent from the board where the A20 is soldered to.

But since there is also no feedback to my answer since August there seems not to be any more interest.

closed 2022-01-28 by linuxianer99