Openvario/meta-openvario #134: ttyUSB0 missing

hpmax, 2021-05-23:
I have a photo of my OV running in an older release and connected to /dev/ttyUSB0 for my FLARM (it's a SoftRF).  However when I run 21118 release on the exact same hardware I do not see /dev/ttyUSB0.  I could have sworn there was a previous discussion (which I can't find) in which this had to be turned on in the configuration (i.e. it's compiled into the kernel).

kedder, 2021-05-23:
Hm, can you take a look what happens in logs when you connect your device?

E.g. run `journalctl -f` and plug the device into the usb port. What gets printed?

hpmax, 2021-05-23:
Unfortunately, I'm using my backup/development hardware, and the SoftRF/FLARM on /dev/ttyUSB0 is in my glider.  I can't access it until next weekend.  Running journalctl -f only seems to show i2c errors related to sensord and variod (note that I do not have a sensorboard plugged in, so this is kind of expected).

As a side note...  I configured everything on my development board and then took the uSD card out and put it in the one installed in my glider.  The keyboard and mouse which I used fine on the development board did not appear to be recognized on the flight hardware.  I'm wondering what the issue with that could have been.

hpmax, 2021-06-01:
https://pastebin.com/hkyzsfLT

I saw no mention of ttyUSB0.  Like I said, I could have sworn that someone (probably you) had to do something to the configuration of the kernel to enable it.  The device is a SoftRF (TTGO T-Beam)...  It's always possible something is wrong, it's under the seatpan so not easy for me to get to it, however...  it's powered by USB and it starts a webserver so it is presumably plugged in and working.

Because of the configuration, I ended up plugging and unplugging (all of my USB devices except my WiFi dongle).

hpmax, 2021-06-05:
Anything else I can get that would help?

hpmax, 2021-06-11:
I found this, which seems relevant:

https://forum.openwrt.org/t/solved-dev-ttyusb0-where-is-it/17652

kedder, 2021-06-11:
@hpmax The only way to debug it I know is to look at the logs when you insert the USB device. Run `journalctl -f` and see what it outputs. Look at the logs on your laptop (where your USB device works), then at the logs on your OV and compare. Usually there will be an error message that would lead to the reason why it doesn't work.

hpmax, 2021-06-11:
I posted the journalctl -f log as requested 10 days ago.  There does not appear to be any entry relevant to the device on the OV.  I can try plugging it into the laptop next time I go to the glider.  However, I believe the issue is a missing module as described in my post from 5 hours ago.  I'd like to point out, it worked fine on a previous CH070 release, so it is almost certainly a software issue.

kedder, 2021-06-11:
There is something called "Arduino Leonardo" - is it the device you miss?

```
May 31 20:54:56 openvario-7-CH070 kernel: usb 1-1.3: Product: Arduino Leonardo
May 31 20:54:56 openvario-7-CH070 kernel[142]: [  881.346243] usb 1-1.3: New USB device found, idVendor=2341, idProduct=8036, bcdDevice= 1.00
May 31 20:54:56 openvario-7-CH070 kernel[142]: [  881.396280] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
May 31 20:54:56 openvario-7-CH070 kernel[142]: [  881.440079] usb 1-1.3: Product: Arduino Leonardo
May 31 20:54:56 openvario-7-CH070 kernel: usb 1-1.3: Manufacturer: Arduino LLC
May 31 20:54:56 openvario-7-CH070 kernel: usb 1-1.3: SerialNumber: HIDFG
May 31 20:54:56 openvario-7-CH070 kernel[142]: [  881.467881] usb 1-1.3: Manufacturer: Arduino LLC
May 31 20:54:56 openvario-7-CH070 kernel[142]: [  881.495582] usb 1-1.3: SerialNumber: HIDFG
May 31 20:54:56 openvario-7-CH070 kernel: cdc_acm 1-1.3:1.0: ttyACM0: USB ACM device
```

hpmax, 2021-06-11:
Unfortunately, I don't know for sure, but I doubt it.  I have two USB hubs installed, one at the OV, and one under my seatpan.  The T-Beam/SoftRF is also under the seatpan and connected to that hub, along with a Bluetooth dongle.  Everything else (including the hub under the seatpan, a GPS, a WiFi dongle, the stick remote, and the rotary encoder) are plugged into the hub attached to OV or the OV itself.

When I tested it, I was unsure which cables were which, so I simply unplugged everything (except the WiFi dongle) from the OV, and then plugged it back in.  I doubt the Arduino Leonardo is the device in question.  That device in question is an ESP32, but more to the point, the Arduino seems to identify itself as a HIDFG.  HID (in the context of USB) typically refers to as a "Human Interface Device" (i.e. keyboard or mouse), so I assume those messages relate either to the stick remote or the rotary encoder both of which should be USB-HIDs and I think both have Arduinos in them to emulate the USB-HID.

Next time I go out to the airport (hopefully this weekend), I should be able to disconnect the hub and plug it into a Linux laptop and run journalctl -f there.  Alternately, I can buy another one and test it on my development machine -- they're only about $35.

kedder, 2021-06-11:
Are you hubs powered? Because device could not work because OV USB is not powerful enough to power all the devices you have attached.

hpmax, 2021-06-11:
The hub that the SoftRF and Bluetooth dongle are connected to is powered.   I have a USB power meter and can measure power consumption on my development setup.  The development setup has a similar (same chipset) WiFi dongle, a similar GPS (same chipset), keyboard, and mouse (as opposed to stick remote and rotary encoder).  According to my USB power consumption meter, I am pulling 170mA max out of the USB under this setup.  The USB spec requires the ability to source at least 500mA, so this has a lot of margin.

I'd like to reiterate, this exact hardware configuration worked fine under a previous build of the software -- and the WiFi server on the device seems to be working which implies it is properly being powered.

hpmax, 2021-06-15:
I didn't get out to the field but I ordered a newer version of it.  Hopefully should arrive within 1-2 weeks and will be able to more thoroughly test it.

hpmax, 2021-06-23:
I tried plugging the hub into my laptop and running journalctl -f, this is what I recorded:

[https://pastebin.com/p4BmqGEq](url)

I think this was the remote stick, a Bluetooth Dongle and the SoftRF device.  I didn't notice anything in the log that looked like the device, just the "keyboard."  Without pulling the seatpan there isn't much more I can do.  I have a slightly newer version on order and will be able to play around with that more easily.  I should have tried to log in and see if I could talk to it over WiFi.

kedder, 2021-06-23:
What does `lsusb` say when the device is inserted and when it is not?

hpmax, 2021-06-23:
Don't know...  The plane is back at the airport, about 70 minutes away from me.  I normally only go out on the weekends if the weather is good, might not be out again until next weekend.  Is there anything else you want me to try while I'm there?  It's a lot easier/faster for me to do everything all at once than one thing at a time.  Once I get the new version it, I'll have it on me and it will be easy to test here.

kedder, 2021-06-23:
> It's a lot easier/faster for me to do everything all at once than one thing at a time.

I can't really tell you what buttons to press. All I can do is to give you some ideas that would help you diagnose the problem.  It is an iterative process. You gradually narrow down the area where the problem could be until you can spot the problem. Set up an experiment, analyze the result, set up another experiment based on therse result, analyze the result and so on. If we can do only one experiment in a week, we are not going to move anywhere. 

I know almost nothing about your device, apart from it is not working and it has a serial interface. That's not enough to do anything. Maybe it is simply a broken hardware?

CazYokoyama, 2021-06-23:
Is USB serial device is enabled on Linux kernel config file, .config, especially cp2102? I may be suggesting irrelevant because I have not read all messages here. To find whether SoftRF (TTGO T-Beam) is detected by kernel, you may run the command, dmesg. When it is detected, dmesg has the message which tells you "it  is in USB XXXX and detected as /dev/ttyUSBn".

hpmax, 2021-06-23:
> Is USB serial device enabled on Linux kernel config file, .config, especially cp2102? I may be suggesting irrelevant because I have not read all messages here. To find whether SoftRF (TTGO T-Beam) is detected by kernel, you may run the command, dmesg. When it is detected, dmesg has the message which tells you "it is in USB XXXX and detected as /dev/ttyUSBn".

I suspect you are 100% dead-on here.  I do recall someone specifically having to change the kernel configuration to support /dev/ttyUSB0.

@kedder can you look into this?

CazYokoyama, 2021-06-24:
My hypothesis is irrelevant.

TTGO T-Beam's USB serial device is CP2104 as below.

[3741936.473355] usb 1-11: Product: CP2104 USB to UART Bridge Controller

Its device ID is
Bus 001 Device 050: ID 10c4:ea60 Silicon Labs CP210x UART Bridge

meta-openvario/recipes-kernel/linux/linux-mainline/arm/defconfig has
CONFIG_USB_SERIAL_CP210X=m

and its kernel source code,  build/tmp/work/openvario_7_CH070-ovlinux-linux-gnueabi/linux-mainline/5.7-r0/linux-5.7/drivers/usb/serial/cp210x.c has
	{ USB_DEVICE(0x10C4, 0xEA60) }, /* Silicon Labs factory default */
So the kernel detects TTGO T-Beam's USB serial device but no /dev/ttyUSB0.

@hpmax , how do you find there is no /dev/ttyUSB0? Device menu of XCSoar is known that it does not show /dev/ttyUSB0. Am I incorrect?

hpmax, 2021-06-24:
If I do a ls /dev/ttyUSB* from the command line,  nothing shows up.  Again, to reiterate, at some point in the past,  it was working on an identical hardware configuration (but not software).  I have a photo showing not just the the Device Config of FLARM on ttyUSB0, but that it had a GPS fix...

hpmax, 2021-06-24:
https://pasteboard.co/K81t8zn.jpg

CazYokoyama, 2021-06-24:
@hpmax , I built the latest code and found corresponding kernel module, cp210x.ko is built. Could you run the following commands from the command line?

ls -l /lib/modules/`uname -r`/kernel/drivers/usb/serial/cp*
lsmod | grep cp

If they have cp210x, kernel detects SoftRF.

hpmax, 2021-06-24:
When I do the ls, I get:
-rw-r--r--    1 root     root         35256 Apr 28 08:48 cp210x.ko

When I do the lsmod | grep cp, I get:
cpufreq_dt             16384  0

So file is there, module isn't loaded?

CazYokoyama, 2021-06-24:
@hpmax , correct. There are following several reasons why the module is not loaded.

1. USB of OpenVario is not active - probably NO. You can attach USB flash memory, correct?
2. disable loading cp210x module explicitly - Nobody wants.

Could you show your dmesg?

hpmax, 2021-06-24:
dmesg:

https://pastebin.com/aPN7mfji

I haven't tried using USB flash memory.  USB definitely works, as I am talking to the device via a USB WiFi dongle.

Is there a way to manually load the cp210x module?  I tried:
insmod /lib/modules/5.7.0/kernel/drivers/usb/serial/cp210x.ko 
insmod: ERROR: could not insert module /lib/modules/5.7.0/kernel/drivers/usb/serial/cp210x.ko: Unknown symbol in module...
dmesg then reports:
[  340.740831] cp210x: Unknown symbol usb_serial_generic_open (err -2)
[  340.747344] cp210x: Unknown symbol usb_serial_generic_close (err -2)
[  340.754072] cp210x: Unknown symbol usb_serial_deregister_drivers (err -2)
[  340.761010] cp210x: Unknown symbol usb_serial_register_drivers (err -2)

CazYokoyama, 2021-06-24:
Now its root cause is clear, i.e. Linux kernel has missing module for cp210x.ko, probably usbserial.ko or similar name.

hpmax, 2021-06-24:
@CazYokoyama Maybe... maybe, not?  

I tried: "modprobe cp210x" and it seemed to work?

"lsmod | grep cp" now reports:
cp210x                 24576  0
usbserial              28672  1 cp210x
cpufreq_dt             16384  0

I tried plugging a device in that I thought had a cp210x in, but it seemed to recognize it as a USB HID.  I will try again once I get the new SoftRF board I ordered.  Hopefully should be in the next couple days.  I doubt that I will be out at the airport this weekend.  If I do go out, I'll definitely give that a shot though and see if it's recognized.

CazYokoyama, 2021-06-24:
No, the root cause is not missing module, usbserial.ko. It is no auto kernel module load mechanism for cp210x.ko.

BTW, I tested with fairly new t-beam 1.1 which has sx1262. I don't expect any change with your will-arrive t-beam.

hpmax, 2021-06-24:
I ordered an SX1262 TBeam, hopefully should get it in the next couple days. 

So you think modprobe cp210x fixed the problem?  I just need to create an /etc/ modules files to tell it to load cp210x?

CazYokoyama, 2021-06-24:
Yes, run modprobe cp210x by some way. The best is autoload such as CentOS does.

hpmax, 2021-06-24:
@CazYokoyama Cool, once I get the new T-Beam in,  I'll play with it and see if I can replicate it and setup autoloading...

Have you found it to be an effective FLARM substitute.  I don't think we have many FLARM users in my club.   However,  I connected to my Stratux European Edition which sees ADSB as FLARM, and I think it warned of a collision with the tow plane while behind it.

hpmax, 2021-06-25:
I created a file called /etc/modules-load.d/cp210x.conf that simply contained: "cp210x"

This appears to automatically load the module on startup.  Once I get access to a T-Beam, I'll confirm it works and close the ticket.  I guess a question is, do we want to change the default configuration?  If it doesn't get changed by default, I'm quite sure this issue will come up again at some point.

hpmax, 2021-06-26:
I can now confirm that the /etc/modules-load.d/cp210x.conf file fixes the issue, as per my last post.

Does anyone have any comments about officially modifying the default distribution to avoid this problem rather than make users go in and add the file on a case by case basis? @kedder @linuxianer99

kedder, 2021-06-26:
This seems like a workaround rather than a fix. We should not force loading the module for users that have no such device.  Normally the module is loaded automatically when device is attached.

hpmax, 2021-06-27:
So, why do you believe it's not loading automatically?  How does it know to do that?

kedder, 2021-06-27:
I was hoping to get some clue from the logs that are being written when you insert the device into the USB socket. Usually the logs give an idea what's happening. 

Watch the logs, insert the device, look at what is being printed at that moment.

hpmax, 2021-06-27:
Here's the log:

https://pastebin.com/pp21iWeB

Unfortunately (or fortunately?), I can no longer replicate the problem.  I deleted cp210x.conf from /etc/modules-load.d, and rebooted.  I do: "lsmod | grep cp2" and get nothing.  Then I plug in the device and do "lsmod | grep cp2" and get:
"cp210x                 24576  0
usbserial              28672  1 cp210x"

So, it is now properly autoloading the module but was not prior to me running modprobe.  I don't know if the configuration file was what did it or the modprobe or either-or...  It's almost like it needs to be told to manually load it first, and then it knows it's okay to autoload...

kedder, 2021-06-27:
Yeah, the log seem to indicate everything worked as expected - device is detected and module is loaded. I don't think manual loading does anything to make the module "loadable". There was probably some other reason behind device not working. If you get this situation again, try looking at the logs the same way - there should be a reason written out.

hpmax, 2021-06-28:
@CazYokoyama Since you have a T-Beam, can you confirm the same issue I had?  CP210x not recognized until you modprobe it, and then after that it recognizes it every time?

CazYokoyama, 2021-06-28:
> Yeah, the log seem to indicate everything worked as expected - device is detected and module is loaded. I don't think manual loading does anything to make the module "loadable". 
----
Agree. Something suppresses auto-loading of  cp210x.ko.

CazYokoyama, 2021-06-28:
> @CazYokoyama Since you have a T-Beam, can you confirm the same issue I had? CP210x not recognized until you modprobe it, and then after that it recognizes it every time?

My OpenVario hardware is in attic, i.e. it is for development of software, not for my glider. Sorry,

Do you have another device on USB when you power-on?

hpmax, 2021-06-28:
@CazYokoyama I'm using development hardware too, however, in the plane I'd have the remote stick, the rotary encoder board, a GPS dongle and a BT dongle and a WiFi dongle.  At home I would always have WiFi.  I might have a keyboard, mouse, and GPS plugged in.

It's very weird that the problem seems to have permanently gone away after putting cp210x in the /etc/load-modules.d directory, even after removing it.  Although I also ran "modprobe", so I don't know which one of those did it.  But even after removing the config file and power cycling a couple times it now seems to correctly autoload.

linuxianer99, 2022-01-20:
@hpmax : Is this still valid ?? or can we close ?

hpmax, 2022-01-20:
@linuxianer99 I can no longer replicate the problem.  It's hard for me to remember exactly what was going on, it would be nice to fix this in the release itself, but maybe it's already fixed in the current release.

closed 2022-01-20 by hpmax