Openvario/meta-openvario #136: XCSoar sais: OpenVario on TCP port 4352 not connected

phorward, 2021-05-29:
Hello!

I've downloaded the latest 20193 release for my OpenVario with 5,7" display and now recognize a strange behavior: The OpenVario device occasionally is found in XCSoar and runs immediately and with vario sound out of the box, but in most times when I start the computer, it states "Not connected" as you can seen from the photo here:
[image]

It seems to be a race condition or some kind, and I'm curious that nobody else every had this problem before. 
Is there any way to fix it?

sfalmo, 2021-07-01:
What sensorboard do you use? I have seen this problem only once about two weeks ago with a SteFly 5.7'' running on 20193 stable. But I did not think much of it since it did not occur again in the numerous times I have used it.

This could be a pain to debug, but if you can reproduce it frequently, maybe take a look at some of the logs.
I personally don't think it is XCSoars fault, but you can still check `xcsoar.log`.
Instead, I suspect sensord to be the problem here; its error messages should hopefully appear in `journalctl -u sensord`.

phorward, 2021-07-08:
Hello @sfalmo, thanks for your reply.

Well, I think it's gonna be a huger problem with my hardware. Its not a SteFly, I ordered it used somewhere. Two weeks ago, when it was very warm outside, my OpenVario froze like shown below entirely for about 5 times during flight. I restarted it several times and it froze again occasionally after two minutes or so. I think there is something generall wrong with it, and I want to replace it now with a SteFly one.

[image]

I think this is not a problem with OpenVario. I'll check the logs and so some more tests when I replaced it with the new one soon.

phorward, 2021-07-08:
I'll gonna close this issue here for now, as I don't think it is a problem with XCSoar or the OpenVario software stack.
Is there any kind of forum or mailing list to discuss about OpenVario issues in general?

closed 2021-07-08 by phorward