Openvario/meta-openvario #144: bitbake xcsoar-image-testing

lordfolken, 2021-12-09:
With hardknott image-testing fails to compile sensord-testing.

xcsoar-image btibake target is fine.

August2111, 2021-12-09:
For this sensord-testing issue I did a PR on the sensord repository yesterday.
I hope this helps...
But after this you are running in a compiling error with xcsoar-testing, this has to do with the new code prepared for 7.21...
Maybe we should use the for this testing image the regular published xcsoar: The changed code here is not official published, and the published version is not far away from this (The big difference between this two setups between 2017 and beginning of this year is disappeared starting with XCSoar 7.1)

linuxianer99, 2021-12-09:
Fixed by openvario/sensord#35

closed 2021-12-09 by linuxianer99