Openvario/meta-openvario #145: voltage reported by openvario is too high

lordfolken, 2021-12-09:
The openvario reports 18.6V while the power supply reports 12,4 V.

https://user-images.githubusercontent.com/407371/145351044-5aa45aed-4df6-427d-9887-e363029df217.mp4

```bash
Build Configuration:
BB_VERSION           = "1.50.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "arm-ovlinux-linux-gnueabi"
MACHINE              = "openvario-57-lvds"
DISTRO               = "ovlinux"
DISTRO_VERSION       = "21343"
TUNE_FEATURES        = "arm armv7ve vfp thumb neon callconvention-hard"
TARGET_FPU           = "hard"
meta                 
meta-poky            
meta-yocto-bsp       = "HEAD:7edd2be25fdf72187738fc551235df0f24e918bb"
meta-sunxi           = "HEAD:612de90a00fbd89762e3d85d85c6846fdc1f9ef6"
meta-openvario       = "HEAD:7c349522ada91ccae88766593fe3da4af2b22b3e"
meta-oe              
meta-python          
meta-networking      = "HEAD:f72a73b42fa740130b388ba8555cdbefdee8d37d"
```

August2111, 2021-12-09:
In my case with machine AM070-DS2 the same issue - I assumed that has to do with my DS2 hardware... 
But because that also happens on machine '57-lvds' too I have to look deeper inside...

August2111, 2021-12-09:
BTW with higher voltage ithe display switched to '0.0' ;-(

August2111, 2021-12-10:
The voltage calculation in the sensord was missaligned with factor 1.5 (since around February).
I will make an PR!