Openvario/meta-openvario #146: openvario bootup animation isn't rotated

lordfolken, 2021-12-09:
The first screen is displayed correctly "Openvario: Starting"
Then the openvario bootup with progressbar is displayed for about 1 second.
After this follows a 10s of complete darkness. 
Thereafter the openvario menu is displayed all correctly rotated

```bash
Build Configuration:
BB_VERSION           = "1.50.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "arm-ovlinux-linux-gnueabi"
MACHINE              = "openvario-57-lvds"
DISTRO               = "ovlinux"
DISTRO_VERSION       = "21343"
TUNE_FEATURES        = "arm armv7ve vfp thumb neon callconvention-hard"
TARGET_FPU           = "hard"
meta                 
meta-poky            
meta-yocto-bsp       = "HEAD:7edd2be25fdf72187738fc551235df0f24e918bb"
meta-sunxi           = "HEAD:612de90a00fbd89762e3d85d85c6846fdc1f9ef6"
meta-openvario       = "HEAD:7c349522ada91ccae88766593fe3da4af2b22b3e"
meta-oe              
meta-python          
meta-networking      = "HEAD:f72a73b42fa740130b388ba8555cdbefdee8d37d"
```

August2111, 2021-12-09:
I saw this 2nd part of boot 'bootup animation' the 1st time here, up to now I don't know, when this displaying starts: In the u-boot or in the startup of linux...
I will look inside too! Maybe we can the long black time shorten or - better - complete switch off ;-)

lordfolken, 2022-01-26:
So with the recent builds, the "times of darkness" are reduced quite a bit.

For the splash screen rotation i have a solution in #201. 

Still the screen going dark after uboot, leaves a large uncanny valley. and its the first impression we give to the user.

closed 2022-02-04 by lordfolken

linuxianer99, 2022-02-04:
@lordfolken Should we make another ticket as enhancement:

Improve user experience during bootup sequence 

?

lordfolken, 2022-02-04:
yes, i would agree also that this is belongs in a new ticket.