Openvario/meta-openvario #148: Update concept of Openvario flight computer

linuxianer99, 2021-12-09:
The update of the software of openvario can be done in several ways:

1. Packet based
2. Image based

At the start of the project, both ways have been implemented and maintained, but as time goes by, the packet based approach seems not to be the best solution for the whole image.

**Suggestion:**
Don't use package based update no longer for OS packages, but for "special" packages like XCSoar or sensord.
OS Update is only done by USB Stick update via update stick.

kedder, 2021-12-09:
I myself find package based updates very convenient, especially when you have internet on Openvario. The only thing that we cannot really cleanly update with the packages is the kernel / u-boot, but I guess new kernels can come with new images.

I'm hosting my own (limited) package repo (https://github.com/kedder/openvario-ppa). It is limited in a sense that it only contains several apps I support and their dependencies. These dependencies, however, constitute pretty much half if not more of all the packages in the openvario "distro". 

So, I'm definitely for package-based updates for XCSoar and some other openvario-specific packages (like sensord and friends). I don't have super-strong opinion on system packages.

lordfolken, 2021-12-09:
Very good. This is the discussion i intended to have.

For me there is 0 or 100%. 
Personally i'm a big fan of package managers. They hide a lot of complexity, make upgrades easy consistent etc.
Everything openvario os and app related should be packaged.

I'd love to have list of package names, which i could download to the stick and upgrade. 
Say i'm interested in:
 - Alps region
 - CH
 - FR
 - IT
 - DE
 - AT

The deal is however to do it for everything. Airspace,flarmnet,waypoint details, maps etc.

**I'm just thinking.. it would actually be a good way to package things even for non-yocto platforms.**

lordfolken, 2021-12-10:
What would be a doable first step, would be to create an ipkg feed for the maps on https://download.xcsoar.org/

The job that builds the maps is here: 
https://github.com/XCSoar/xcsoar-data-maps/blob/master/.github/workflows/maps-build.yml

That job runs every 2 weeks, and upon requested changes in: 
https://github.com/XCSoar/xcsoar-data-maps/blob/master/data/maps.config.js

Maybe someone could give me some instructions how to build ipkgs of map files? I could implement it there. 

We have some plans to merge that repo and handle everything from https://github.com/XCSoar/xcsoar-data-content see
https://github.com/XCSoar/xcsoar-data-content/issues/180

August2111, 2021-12-10:
@lordfolken : As I said in a previous discussion to issue #139: In our case the predefined maps from XCSoar don't make so much sense for us: In our 3-country-corner we need the maps from DE, CZ and PL simultanously - and so we create a user defined map ourself with the very funny xcsoar tool :) - and for this I have no package, but I can it load very easy with the XCSoar menu itself. 
Personelly I prefere a solution inside the xcsoar map selection menu with a 'copy from' button (next to the button 'Download') to copy this file somewhere from the filesystem, especially from the USB stick...
I would rather increase the variety of functions of XCSoar than put a lot of effort into such side projects.
From my point of view there are a lot of other things to do first...

August2111, 2021-12-10:
An other thing to the 'concept update' I want to propose: @lordfolken: Can you support us with the possibility of bringing the xcsoar-ipk packages to the XCSoar download page. In my opinion, is that more important than various other packages such as the Cupieboard-Debian package?

linuxianer99, 2021-12-10:
> Can you support us with the possibility of bringing the xcsoar-ipk packages to the XCSoar download page

@August2111 : Think that is not a good idea ... We do some patching to the XCSoar code from Openvario side ... So this is not a "pure" XCSoar any longer, but a "openvario-specific" XCSoar. So we can offer the XCSoar IPK on the OV Download page.

August2111, 2021-12-10:
I don't like the patching of XCSoar code...
Therefore I would remove f.e. the patch 'FreeVario-Driver'. If we whant to have support from Max K. for our project we have to cleanup our code immediately. I have no problem to test such things in the 'openvario-image-testing' package - but not in our official release! I had spoken to Blaubart about this possibilty - and indicated the possibility of helping him to integrate his code into XCSoar

From Max I know he don't like every code patches and said that all this code has to inserted in his project - with code review (and maybe some compiler flags...)

linuxianer99, 2021-12-10:
ok. also fine for me ..
The FreeVario-Driver was just a compromise to help to FreeVario Team, as they had some problems integrating their driver to XCSoar. But you are completely right, the driver has to go to XCSoar Repo in the end!!

But nevertheless, does it make sense to publish a IPK for OV on XCSoar website ? This package can only be installed on our distribution ...

August2111, 2021-12-10:
> The FreeVario-Driver was just a compromise to help to FreeVario Team, as they had some problems integrating their driver to XCSoar. But you are completely right, the driver has to go to XCSoar Repo in the end!!

Ok, I know. but so for FreeVario all things are 'completed' - and nobody works on it yet... Because Blaurock makes every time his own images there is no need to support this in the future (maybe in the testing package it shoud be exist a while)

> But nevertheless, does it make sense to publish a IPK for OV on XCSoar website ? This package can only be installed on our distribution ...

... and the KOBO can only installed on Kobo, Raspie only on Raspie, Android only on Android...
We have a very fine device with hundreds of usern (f.e. ask Stefan Langner) - and so XCSoar would help a very large community!

linuxianer99, 2021-12-10:
ok ok ;-)
If Max will agree, let's do so ;-)

But for our own concpet, we should define, which update packages we will provide and which content is only available via Image-flash.

Can you make a detailed proposal from your point of view ??

lordfolken, 2021-12-10:
regardless of how you decide: what would be fantastic would be a prebuilt openvario-root-tgz that is generated automatically. This i would then intergrate into xcsoar's docker container. That way we could test PRs and commits by crosscompiling with TARGET=CUBIE CUBIE_ROOT=/opt/cubie/

linuxianer99, 2021-12-10:
> regardless of how you decide: what would be fantastic would be a prebuilt openvario-root-tgz that is generated automatically. This i would then intergrate into xcsoar's docker container. That way we could test PRs and commits by crosscompiling with TARGET=CUBIE CUBIE_ROOT=/opt/cubie/

@lordfolken :  https://www.yoctoproject.org/docs/1.6.1/adt-manual/adt-manual.html
Think this is the thing you are looking for ??

lordfolken, 2021-12-12:
No, I was hoping to find a prebuilt root tar.gz built by your toolchains. So I could compile xcsoar against it.

linuxianer99, 2021-12-12:
> No, I was hoping to find a prebuilt root tar.gz built by your toolchains. So I could compile xcsoar against it.

Let's make a seperate issue ... #150

lordfolken, 2021-12-12:
What i can offer is that we do a webhook trigger when a new release of xcsoar is done. That way openvario could trigger a new build.