Openvario/meta-openvario #150: Build environment for XCSoar

linuxianer99, 2021-12-12:
There should be a openvario based root.tar.gz for building XCsoar against.

linuxianer99, 2021-12-14:
@lordfolken: 
I had a look at this the last few days. Think there are two possibilites:

- Just extract the rootfs of the image. This is an easy task, just copy from the build dir ...
- Use the SDK feature of openembedded. This generates an installer package including the whole environment. In my mind, this would be the best way, as it is the same environment as bitbake itself is using. Maybe we can create an XCSoar target called OPENVARIO and omit all the build patches we are using at the moment ...

Here some documentation about this:
https://docs.yoctoproject.org/sdk-manual/working-projects.html

There is also a container for running the SDK (https://github.com/crops/extsdk-container)

What do you think ??

lordfolken, 2021-12-15:
Hey @linuxianer99 

Key Goals:
 * Make it easy for developers to build an openvario build of xcsoar
 * Test PRs and Commits against the openvario environment (as to not introduce incompatibilites) etc. 

What I would have envisioned:
* to download the the tar.gz of ftp.openvario.org/developers/cubie-root/cubie-root-latest.tar.gz somewhere, during build of the container
* to extract it to /opt/cubie/root/ inside the container
* to adapt the build script for openvairo with make -j$(nproc) TARGET=CUBIE  

What i don't want:
 * To Build the root in the automatic process that creates the docker container for build tests (this takes too long for github actions, and is massive 30+GB)

linuxianer99, 2021-12-15:
Hey @lordfolken 

Ok ... got your points!
To be honest, i was not aware of this docker build environment used by XCSoar... In this case, it make sense to enhance this of course !

I had a look at the links in the post before:

- The TARGET CUBIE is just for Openvario, right ?? or is it used anywhere else ?
- There is no installation of any rootfs.tar.gz up to now in the scripts, right ?? Or did i miss it somewhere ?

So jobs to do:

1. Extract rootfs from Openvario build
2. Create provision script for installing rootfs inside the XCSoar build container 
3. Modify XCSoar makefile for TARGET CUBIE to build correctly

Did i miss something ?

lordfolken, 2021-12-15:
> Hey @lordfolken
> 
> Ok ... got your points! To be honest, i was not aware of this docker build environment used by XCSoar... In this case, it make sense to enhance this of course !

Way ahead of you. ;-) https://github.com/XCSoar/XCSoar/tree/master/.github/workflows
Next up will be release automation. I would love to upload the artificats to the release page aswell.
 
> I had a look at the links in the post before:
> 
>     * The TARGET CUBIE is just for Openvario, right ?? or is it used anywhere else ?
To my knowledge the purpose of that target was to crosscompile for the cubie board. So mainly openvario.

>     * There is no installation of any rootfs.tar.gz up to now in the scripts, right ?? Or did i miss it somewhere ?
Yes that is currently not handled by XCSoar's build provisioning environment. I would push a new build-cubie.yml and this would run along with the other runs on prs/commits. 
I just would like to stress that this should not replace your xcsoar build process. its just for validation and development in case you have to test or debug something.

> So jobs to do:
> 
>     1. Extract rootfs from Openvario build
Yep.
>     2. Create provision script for installing rootfs inside the XCSoar build container
I can do that. We just need a place where we keep the current built rootfs around where its publicly accessible.
>     3. Modify XCSoar makefile for TARGET CUBIE to build correctly
I was hoping that here no changes would be needed.  ;)

Thank you for your efforts! :) Lets make openvario an excellent platform. ;)

linuxianer99, 2021-12-15:
> I can do that. We just need a place where we keep the current built rootfs around where its publicly accessible.

What about the Github Packages ??
Or just as an asset along with an release of Openvario ??

lordfolken, 2021-12-15:
> > Way ahead of you. ;-) https://github.com/XCSoar/XCSoar/tree/master/.github/workflows
> > Next up will be release automation. I would love to upload the artificats to the release page aswell.
> 
> I play around a long time with this automation, but the complete image build takes too long, so even with a self-hosted runner, there are some problem ...
>
Yes i can image that that is a problem with github actions. Here i would go in the direction of git and webhooks. Yocto seems to have something called toaster: https://docs.yoctoproject.org/toaster-manual/index.html

it has sort of an api: 
https://docs.yoctoproject.org/toaster-manual/reference.html#determining-status-of-a-specific-build

you can also run builds by calling a python script:
https://docs.yoctoproject.org/toaster-manual/reference.html#runbuilds

> > I can do that. We just need a place where we keep the current built rootfs around where its publicly accessible.
> 
> What about the Github Packages ?? Or just as an asset along with an release of Openvario ??
Fine with me.

linuxianer99, 2021-12-16:
@lordfolken 
Here is the rootfs for download:
https://github.com/Openvario/meta-openvario/releases/download/21118/OpenVario-linux-openvario-image-testing-glibc-ipk-21118-openvario-7-PQ070.rootfs.tar.gz

lordfolken, 2021-12-16:
Thats very good. Would it be possible to get a stable url? The release is kind of hard to encode in a script, unless you know a magic curl command?

August2111, 2021-12-17:
> > I had a look at the links in the post before:
> > ```
> > * The TARGET CUBIE is just for Openvario, right ?? or is it used anywhere else ?
> > ```
> 
> To my knowledge the purpose of that target was to crosscompile for the cubie board. So mainly openvario.

Yes and no - the result of this crosscompile target is a .deb file - and this can be used on any linux running on Cubieboard2. Unfortunately we have a complete splitting of our development work now: One group is going to an Ubuntu on Cubieboard (hor63, roeles) - our group is staying at Yocto... (I think the much better solution). But I think too, Kai (hor63) and Roel (roeles) needs the deb file for their development. Because I want to bring much more functionality from OpenVario to the XCSoar later, and so I cannot use the CUBIE target itself... But this is a story for the future.. :)

linuxianer99, 2022-01-13:
> Thats very good. Would it be possible to get a stable url? The release is kind of hard to encode in a script, unless you know a magic curl command?

The question is, if we need to update the rootfs for the XCSoar build, with every release of OV ?
I think so ... so it make sense, to store the rootfs file within the Release structure, right ?

I will have a look, if github has some magic url for the latest release or stuff like this ..