Openvario/meta-openvario #153: Kernel config not current

lordfolken, 2021-12-14:
The Kernel configuration isn't current for hardknott and a lot of drivers are missing:

https://github.com/Openvario/meta-openvario/blob/cb717925100ca2927ae626084b26710355f67ba7/recipes-kernel/linux/linux-mainline/arm/defconfig#L3

lordfolken, 2021-12-14:
```bash
CONFIG_RTL8192CE=m
CONFIG_RTL8192SE=m
CONFIG_RTL8192DE=m
CONFIG_RTL8723AE=m
CONFIG_RTL8723BE=m
CONFIG_RTL8188EE=m
CONFIG_RTL8192EE=m
CONFIG_RTL8821AE=m
CONFIG_RTL8192CU=m
```

August2111, 2021-12-17:
Hi @lordfolken, this is a very good point, your 1st comment shows only an old script entry in the defconfig, we had in the warrior branch a linux kernel 5.7 - in this kernel we have in the hardknott branch too without any changes. But in meta-sunxi there a a kernal 5.10.2 supported - and this version we should also use inside openvario. I propose to do it with the simular setting of recipe: Using only an append file `linux-mainline_%.bbappend`  with the special code from OV in addition ...
But for this I need a little bit time - but I think this is important for the further progress on our project :)

Regarding the usb wireless driver: I didn't found them in the defconfig from sunxi with kernel 5.10.2 From my point of view this is an other issue... 
BTW: If you mean the support of a WLAN stick on the OV? This works on my openvario-7-AM070-DS2 perfectly ;-)

lordfolken, 2021-12-17:
Your suspicion is correct. I was trying to get the wifi stick working. (BTW USB Ethernet adapters don't work either...) 

I was trying to run a TPLink usb wlan stick on the current hardknott OV, and it doesn't create a wlan0 device. (openvario-57-lvds). 
https://www.tp-link.com/us/home-networking/usb-adapter/tl-wn725n/

It sees the device appearing on USB, but then fails to load the regulatory.db and doesn't load the kernel rtl8188 kernel module. The chip seems to be an rtl8188eus but this module seems to be missing on openvario.

August2111, 2021-12-17:
Ok, but not so easy to support, because I don't have such a stick.
But normally it should work if a driver is in :)
I have a D-Link stick - this works fine, but with my AVM-FritzBox stick don't work. But here I read, that AVM don't support Linux, but this I cannot believe...

lordfolken, 2021-12-17:
The problem is that we are applying a 4.XX config to a 5.7+ kernel. The options not selected will not be built. So all these drivers will not make into the OV. I tried to update it with make oldconfig in the kernel source.. but it then tries to configure a x86 architecture.

lordfolken, 2021-12-17:
I'm sure the module is all that is missing.

August2111, 2021-12-17:
Together with the switch to the kernel 5.10.2 - supported from meta-sunxi - I' want to cleaning up the defconfig file. 
But that has to be done with a lot of fingertip sensitivity and attention...

August2111, 2021-12-17:
... uboot and ovmenu are running with new kernel 5.10.2 - only xcsoar chrashed up to now ;-(

lordfolken, 2021-12-17:
yikes, do you have an error message? or xcsoar.log?

August2111, 2021-12-17:
The xcsoar log has only one line '[$DATE, $TIME] Starting XCSoar XCSoar v7.21-Linux'...
The error message before disappers to fast - so I have insert a break point before the ovmenu ist starting! Maybe this tells me more.
But I think that have something to do with the display or other drivers - there I didn't checked up to now!
As I said: I need a little bit time for this :)

August2111, 2021-12-18:
With the old config file `defconfig`  it is running again with linux kernel 5.10.2!
So I/we have to adjust a new config file very carefully! Maybe @linuxianer99 is more experianced for this - but of course I also will try it too :)

linuxianer99, 2021-12-18:
I can have a look on this next week ...

August2111, 2021-12-18:
As I mentioned: I also will try it too - maybe I proposed a new file - and you review it?
I'm a really Windows user and developer with only a small Linux experience, but I would like to know more about the background to the system and there is no way around digging deeper into the matter.
I think I've made quite good progress so far :)

linuxianer99, 2021-12-18:
Yes!! You made very good progress!!!
Feel free to provide a defconfig for review !!

August2111, 2021-12-31:
Update: The installation of this drivers have a good progress - at my system all USB wifi dongles are running (ok, the AVM sticks have an additional problem, but this is also solvable :) ): D-Link and TP-Link (my cipset: rtl8192eu) works, USB to LAN adapter works, USB to Serial adapter (pl2303) works...
But unfortunately my cubieboard2 died and gave up the ghost :-( so I had a lot of things to do to get a running test system ....

lordfolken, 2021-12-31:
Yikes. Thank you for all the work.

lordfolken, 2022-01-01:
So i tried the current image and there is no change. Neither usb nor wifi adapter work. I cannot find the rtl8188 modules and it doesn't seem to appear in the config. 
At least comparing to my debian grepping 
wget https://raw.githubusercontent.com/kedder/openvario-shell/master/scripts/openvario-shell-install.sh -O - | sh

```bash
 $  grep -i rtl /boot/config-5.10.0-8-amd64 
CONFIG_BT_RTL=m
CONFIG_BT_HCIBTUSB_RTL=y
CONFIG_BT_HCIUART_RTL=y
CONFIG_USB_RTL8150=m
CONFIG_USB_RTL8152=m
CONFIG_RTL8180=m
CONFIG_RTL8187=m
CONFIG_RTL8187_LEDS=y
CONFIG_RTL_CARDS=m
CONFIG_RTL8192CE=m
CONFIG_RTL8192SE=m
CONFIG_RTL8192DE=m
CONFIG_RTL8723AE=m
CONFIG_RTL8723BE=m
CONFIG_RTL8188EE=m
CONFIG_RTL8192EE=m
CONFIG_RTL8821AE=m
CONFIG_RTL8192CU=m
CONFIG_RTLWIFI=m
CONFIG_RTLWIFI_PCI=m
CONFIG_RTLWIFI_USB=m
# CONFIG_RTLWIFI_DEBUG is not set
CONFIG_RTL8192C_COMMON=m
CONFIG_RTL8723_COMMON=m
CONFIG_RTLBTCOEXIST=m
CONFIG_RTL8XXXU=m
# CONFIG_RTL8XXXU_UNTESTED is not set
# CONFIG_SERIAL_UARTLITE is not set
CONFIG_DVB_USB_RTL28XXU=m
CONFIG_DVB_RTL2830=m
CONFIG_DVB_RTL2832=m
CONFIG_DVB_RTL2832_SDR=m
CONFIG_SND_SOC_SOF_ELKHARTLAKE_SUPPORT=y
CONFIG_SND_SOC_SOF_ELKHARTLAKE=m
CONFIG_RTL8192U=m
CONFIG_RTLLIB=m
CONFIG_RTLLIB_CRYPTO_CCMP=m
CONFIG_RTLLIB_CRYPTO_TKIP=m
CONFIG_RTLLIB_CRYPTO_WEP=m
CONFIG_RTL8192E=m
CONFIG_RTL8723BS=m
CONFIG_IBM_RTL=m
```
when grepping openvario:
```
 $  grep -i rtl defconfig 
CONFIG_BT_RTL=m
CONFIG_BT_HCIBTUSB_RTL=y
CONFIG_USB_RTL8150=y
CONFIG_USB_RTL8152=y
# CONFIG_RTL8187 is not set
CONFIG_RTL_CARDS=m
CONFIG_RTL8192CU=m
CONFIG_RTLWIFI=m
CONFIG_RTLWIFI_USB=m
CONFIG_RTLWIFI_DEBUG=y
CONFIG_RTL8192C_COMMON=m
CONFIG_RTL8XXXU=m
# CONFIG_RTL8XXXU_UNTESTED is not set
# CONFIG_SERIAL_UARTLITE is not set
```

August2111, 2022-01-01:
Hey @lordfolken , this was one step to fast... Up to now I havn't a PR with any change of this: From my point of view it is much more important to make a complete switch to the 'new' kernel 5.10.2 instead of 5.7! Your change of the defconfig is only a small part of this.

But because I haven't a running OV system during the last days I couldn't prove my last changes - I hope to do that up to end of next week (09/01/2022). Hopefully I will publishing a 1st proposal in the beginning of next week on my fork https://github.com/August2111/meta-openvario - and here you can find all drivers in the system like my OV!

lordfolken, 2022-01-01:
Sorry i wasn't precise enough. I tried your fork, with the change in the log, and that is what doesn't contain the rtl modules.

August2111, 2022-01-01:
Pls can you tell me which branch / commit to check your setup?

lordfolken, 2022-01-01:
i'm sorry to hear that your board is broken. 
commit a28385e1d997de52dce86065c27c5bf11c82c51d (HEAD, ov/hardknott, m/hardknott)

August2111, 2022-01-01:
Ok, from my point of view there is small change in the layer recipe  meta/linux-firmware/linux-firmware_20211027.bb necessary or - better - an additional setting in meta-openvario/linux-firmware/linux-firmware_%.bbappend, but this I have to check to minimize the side effects... And up to now this change isn't published in my repository ;-(, so it couldn't work :(

August2111, 2022-01-03:
With PR #157 I have again put together a package of changes that primarily aims to solve the annoying driver problem from this issue.
If any important driver isn't functional please:
* remove the hardware device from usb if inserted
* switch to shell mode
* put this problematic device in the USB slot
* call `dmesg` - and send me the last ~20 lines, than maybe I have a chance to solve without such a device!
@lordfolken You can also try my branch 'hardknott' https://github.com/August2111/meta-openvario/commit/33b177617ee690293b02300ae461161d31c85975  with some cleaning variod and sensord - and the usage of the sensord_0.3.5

In this sensord version 0.3.5 it is a preliminary better calibration of the battery voltage - all other investigations on this topic 37 of sensord (https://github.com/Openvario/sensord/issues/37) is for the cat :) in the upcoming sensord-testing it is much better designed and calibrated!

linuxianer99, 2022-01-20:
@lordfolken  Think this can be closed ??
Max did a lot of work in #186 #188 #192

lordfolken, 2022-01-21:
@linuxianer99 i'll check the current build.

linuxianer99, 2022-03-08:
@lordfolken Is this issue still valid ??