Openvario/meta-openvario #156: Home folder in Openvario

lordfolken, 2022-01-01:
Hello, 

The current image seems to use /root/.xcsoar for xcsoar's data dir. 
The backup scripts etc point to /home/root/.
https://github.com/Openvario/meta-openvario/blob/cb717925100ca2927ae626084b26710355f67ba7/recipes-apps/ovmenu-ng-skripts/files/download-all.sh#L4

Could we use the $HOME variable?

August2111, 2022-01-01:
For this data directory I have an other proposal, but this have to be discussed here: Of course you should use the $HOME variable itself - my proposal is to use for this the free area of sdcard as 3rd partition (/dev/sda3/XCSoarData) with FAT32 as standard folder, this is big enough for all files - and you can access to it if you have the sdcard outside of the OV at a Linux or Windows PC!

But of course the DOWNLOAD_PATH and the xcsoar's data dir have to be aligned to each other!

lordfolken, 2022-01-01:
I would not. Having yet another way to access the xcsoar data folder that works only half the time isn't helping. 

There has to be a clear designed official way. 

Besides:
* fat32 is corruptable, not journaling, not easly fscked etc. 
* fumbeling with micro-sdcards in the cockpit is just unpractical. (hidden slots, things can fall into cracks, the ejection mechasims break, easly lost, adapters, different standards...) ( i know the big ones do it, but at least in my club we are mounting usb ports everywhere now..)
* USB Sticks are much more practical for this

However I would:
 * seperate the userhome from the os:
   * replace the os without wiping the user data
   * should run as a dedicated user, not root
   * filing the data partition won't affect the OS

kedder, 2022-01-04:
I'm with @lordfolken on this. I'd wish we could utilize a separate data partition on sd card that would use the whole size of an sd card. In a similar fashion ubuntu live-disk works. However I'm not aware of any avarege-user-accessible tools that could create such a filesystem on a flash card.

lordfolken, 2022-01-04:
RPI do it simillarly. They create a small second paritition, and then resize it at boot time to the whole device. Such a thing could be achieved with sfdisk.

August2111, 2022-01-04:
Hi kedder, I had done some experimental work on this topic to create a separate partitition on the 4, 8, 16, 32 GB sized sd card: a lot of room for many data. But why lordfolken decline the fat32? From my point of view I don't need the journaling and fsck on my user data, most of the sd cards and usb sticks are formatted with fat32 - and it is easy readable and editable as well on Linux as Windows.
Ok, during one season I don't like to fumbling with this micro card in the cockpit - during a season the device should work as it is, maybe an update on the fly with WLAN or USB stick without any deleting/moving of this data or something else.
Of course everybody can do what he want (like opensource :) ) but the standard user  should consider the sd card to be permanently installed!

The USB stick isn't a good location for user data, only for backups, data transports or updates: It is easy to manage - but also easy to forgot too, or the wrong stick is inserted and so on!

And so at least this points are the my target:

- dynamically allocated 3rd partition on the large free memory of the sd card ( I would very much like to investigate this point soon)
- Which fs should be used has to be decided, but consider a large number of Windows user flying with OV
- A user 'ovuser' should be created and the owner of the device (and the data folder)

August2111, 2022-01-04:
> RPI do it simillarly. They create a small second paritition in the dd image, and then resize it at boot time to the whole device. Such a thing could be achieved with sfdisk.

Thanks @lordfolken: This is what I mean and wish!

lordfolken, 2022-01-04:
FAT32 is corruptible if you turn off the power or loose the power during operation. Improper shutdown or similar. And all that is very likely in a battery operated environment. Usually if that happens you need to manually check and repair the filesystem.
Ext4 will survive pretty much anything, and is able to fsck itself at boot.

The usb sticks are usually mounted synchronosly for that reason. But mounting something synchronosly will

kedder, 2022-01-04:
Yes, some users seem to expect the device to be turned off by simply flipping the power switch. I remember I added a bunch of `sync` to OV menus to avoid data loss in that case. I can imagine fat32 would not withstand such an abuse. I can see how fat32 is more accessible to regular windows users, but they might not fully appreciate that if the partition with the flight logs becomes not-writable or corrupted after an accidental power cycle.

kedder, 2022-01-04:
Dynamic resizing of a filesystem on startup / first boot sounds like a neat idea! We can probably steal it from RPI.

linuxianer99, 2022-01-05:
So i support @lordfolken proposal:

- Having a small data partition with a journaling filesystem (ext4)
- Resize partition on the first boot
- Use the directories already used for the data (homefolder of the user XCSoar is running)

- USB Stick can be used of course for transfer the files to PC ...

linuxianer99, 2022-01-05:
@August2111 @lordfolken @kedder
I enabled the Discussion function in Github. Maybe we can have basic discussions (like this) in the discussion tab in the future ??
What do you think ?

August2111, 2022-01-05:
I'll try one more time anyway: What do you think about NTFS? It has journaling, there are fsck available - and it is read- and writable on all systems Linux, Mac, Windows...

Because of the problems with switching off, I would like to suggest another variant: Another, separate partition for logging - this would mean that a hard switch-off when writing Flash would not affect the other, important content ...

August2111, 2022-01-05:
@linuxianer99 :Do we want to end this point here or do you want to switch to the 'Discussion function'?

linuxianer99, 2022-01-05:
> @linuxianer99 :Do we want to end this point here or do you want to switch to the 'Discussion function'?

No, not for this topic ..
I want just announce the new feature for future discussions ..

linuxianer99, 2022-01-05:
> I'll try one more time anyway: What do you think about NTFS? It has journaling, there are fsck available - and it is read- and writable on all systems Linux, Mac, Windows...
>
> Because of the problems [...]

NTFS is a bit tricky in Linux distributions ... 
So what is the big disadvantage in your mind by using EXT4 ??

August2111, 2022-01-05:
EXT4 is much more tricky on Windows systems as NTFS on Linux.
In my club all 6 OV owner are Windows user, they burn a complete image on a sd card (with Win32DiskImager) - and that's it. They have a very small linux knowledge - and they want also edit the content of the data directory, not every day - but maybe in the winter season.

MaxKellermann, 2022-01-13:
UDF? This file system is available on all operating systems, and doesn't suffer from many of FAT's limitations.

August2111, 2022-01-13:
UDF sounds good 👍 
I will check and test it!