Openvario/meta-openvario #171: Airspace not showing correctly

iglesiasmarianod, 2022-01-10:
Hi guys, I'm having a problem with airspace files.
Airspace arc sectors are not shown on the map.
Airspace file works OK in my smartphone.
Anyone else has had this kind of problem?
Here's the airspace file I'm trying to use.
[ESPACIO_AEREO_SROSA.txt](https://github.com/Openvario/meta-openvario/files/7839756/ESPACIO_AEREO_SROSA.txt)

Do you think it is a XCSoar problem or OpenVario problem?

lordfolken, 2022-01-23:
This should be discussed in
https://github.com/XCSoar/XCSoar/issues

For general format consideration:
http://winpilot.com/usersguide/userairspace.asp

The sectors seem to be correct, after comparing. If these don't show, its a bug.

lordfolken, 2022-01-23:
So at least on current head version of xcsoar (7.22+) the airspaces are displayed. 
INFO->INFO->INFO->Airspaces will show you the airspaces.

iglesiasmarianod, 2022-01-23:
Hi @lordfolken thanks for the reply.
The file contains 4 airspaces total. 2 circles and 2 partial sectors.
Circular airspaces actually show, only the first two (partial sectors) are missing.
The same file shows all the airspaces using android version.
It happened to all the OpenVario users that tried to use the above file.

If I define the partial sectors in this form:
```
* OSA
AC P
AN PROHIBIDO OSA
AL SFC
AH FL200
V D=+
V X=36:35.031 S 64:16.370 W
DB 36:32.45 S 64:14.15 W, 36:38.033 S 64:16.370 W
DB 36:50.05 S 64:16.370 W, 36:22.016 S 64:7.0833 W

* Planeadores
AC R
AN Desde 6000 FT
AL FL060
AH FL200
V D=+
V X=36:35.031 S 64:16.370 W
DB 36:32.45 S 64:14.15 W, 36:38.033 S 64:16.370 W
V D=-
DB 36:50.05 S 64:16.370 W, 36:22.016 S 64:7.0833 W

```
They show.

lordfolken, 2022-01-23:
I assume you are using the current ov linux image on the homepage? 
Please wait a few days, then the new image will come with a much more current xcsoar version, that is in sync with android.

iglesiasmarianod, 2022-01-23:
Great, I'm in the process of installing Debian 11 to build images.
I'll wait till I can test the most current version.
Thanks!

linuxianer99, 2022-03-08:
@iglesiasmarianod Is this issue still valid ??

iglesiasmarianod, 2022-03-13:
Hi @linuxianer99. I'm away from home this week. I'll check asap and close it if it is solved.

iglesiasmarianod, 2022-04-20:
Hi @linuxianer99 airspaces show OK in the latest testing version. Thanks.
I'll close the issue.

closed 2022-04-20 by iglesiasmarianod