Openvario/meta-openvario #175: Use systemd-networkd instead of connman

MaxKellermann, 2022-01-13:
connman is a heavy dependency, and systemd-networkd is installed anyway, and can do the same (with a tiny footprint).
For Wifi, only wpa_supplicant is needed.

kedder, 2022-01-13:
connman is used by [ov-shell](https://github.com/kedder/openvario-shell) to manage network connections. Throwing connman out will make that part useless, so I suggest to leave it.

MaxKellermann, 2022-01-13:
systemd-networkd has a DBus interface as well.
Removing connman saves 10 MB in the OpenVario image - 124 MB -> 114 MB!
Compare that to systemd-networkd, which weighs just 0.9 MB.

kedder, 2022-01-13:
That would still mean a near-complete rewrite of the most complicated part of the ov-shell. I can see the benefit, but I want to keep ovshell working. So this effort needs to be synched somehow.

MaxKellermann, 2022-01-13:
It doesn't make sense to keep the connman dependency at the image level.
If one day ov-shell gets integrated, connman should be a dependency of the ov-shell recipe, but not of the image. The image doesn't need connman, only ov-shell does. Until this integration happens, connman is useless.
Whenever ov-shell gets integrated, there will need to be the Python3 dependency. I just tried: adding Python adds "only" 40 MB (plus a megabyte for urwid); 114 MB -> 155 MB. That's +36% image size (and +44% if you also consider the overhead for connman).

condoratberlin, 2023-01-23:
Is there any solution for peaple using connman? Is there an option to build OpenVario yourself with connman? Or even install connman later?